espressif/esp-idf #4957: I2S communication format unclear

callalilychen, 2020-03-18:
According to the [reference manual of esp32](https://www.openhacks.com/uploadsproductos/esp32_technical_reference_manual_en_qg.pdf), for I2s, 3 Audio Standards are supported, but I am not sure, how to configure the three standards using the esp-idf. Because in the i2s_types.h, 6 communication format is defined:

https://github.com/espressif/esp-idf/blob/0ae960f2fe92de1ee9c7c624b7115d06faca119e/components/soc/include/hal/i2s_types.h#L64-L71

I can not find further information about these enums entries, so I read the code. But I can not understand, what is the difference between these configs:

```
.communication_format = I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB, 
```
or 
```
.communication_format = I2S_COMM_FORMAT_I2S, 
```

koobest, 2020-03-19:
Hi, @callalilychen 
Thank you for reporting this issue. If MSB-aligned communication format is required, it should be configured as "I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_LSB" and we will improve the documentation.

thanks !!

Alvin1Zhang, 2020-04-14:
@callalilychen Thanks for reporting, we have a fix which is now under internal reviewing, will update once it is available on GitHub. Thanks.

closed 2020-06-12 by igrr