espressif/esp-idf #4958: Update MbedTLS version to 2.18 for use with Apple's open source HomeKit ADK

bpietsch, 2020-03-19:
**Is your feature request related to a problem? Please describe.**

Apple has released an open source HomeKit development kit for hobbyists / makers. Despite this, Espressif will not also release the ESP32 HomeKit SDK, so I'd like to port the open source ADK from Apple. Unfortunately, the ADK from Apple requires MbedTLS 2.18 as the minimum version.

**Describe the solution you'd like**

Update the current MbedTLS version (2.16) to 2.18, or make it a configuration option to choose between 2.16 and 2.18 if it is desirable to continue to make 2.16 an option.

**Describe alternatives you've considered**

I've tried cherry-picking the missing functionality, but this is a hack at best (and I still haven't successfully pulled it off)

shahpiyushv, 2020-05-15:
@bpietsch , we have made available a port for HomeKit ADK here: https://github.com/espressif/esp-apple-homekit-adk. The required patches from mbedtls 2.18 have been applied on 2.16.6 on this mbedtls branch: https://github.com/espressif/mbedtls/tree/mbedtls-2.16.6-adk

Alvin1Zhang, 2020-05-18:
@shahpiyushv Thanks for the support. @bpietsch Thanks for supporting, will close the ticket now.

closed 2020-05-18 by Alvin1Zhang

AxelLin, 2020-05-18:
Hi @shahpiyushv 
Since you already have a mbedtls-2.16.6-adk branch.
You can also create a mbedtls-2.16.6-idf branch for ESP-IDF users
which includes fix for CVE-2020-10932.
https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2020-10932

shahpiyushv, 2020-05-18:
@AxelLin , it will be available soon. mbedtls-2.16.6-adk required testing of only ADK, whereas mbedtls-2.16.6-idf requires much more testing as it can affect many components. It is on its way though :)