espressif/esp-idf #4959: Crash when RMT mem_block_num > 1

cskilbeck, 2020-03-19:
When `rmt_config` is called with `rmt_config_t` containing `mem_block_num` value > 1, then a hard crash occurs in rmt translation function.

```
    void IRAM_ATTR rmt_translate(const void *src, rmt_item32_t *dest, size_t src_size, size_t wanted_num, size_t *translated_size, size_t *item_num)
    {
        size_t size = 0;
        size_t num = 0;

        constexpr rmt_item32_t const on = { { { T1H, 0, T1L, 1 } } };    // these are inverted because the level shifter inverts them back
        constexpr rmt_item32_t const off = { { { T0H, 0, T0L, 1 } } };
        constexpr rmt_item32_t const on_off[2] = { off, on };

        if(src != null && dest != null) {
            byte const *psrc = (byte const *)src;
            while(num < wanted_num) {
                byte const data = *psrc++;
                for(int bit = 7; bit >= 0; bit -= 1) {
                    *dest++ = on_off[(data >> bit) & 1];
                }
                num += 8;
                if(++size >= src_size) {
                    dest[-1].duration1 = 2000;    // reset ~50uS
                    break;
                }
            }
        }
        *translated_size = size;
        *item_num = num;
    }
```

cskilbeck, 2020-03-19:
Init function:
```
void neopixel_init()
{
    rmt_config_t config;
    config.rmt_mode = RMT_MODE_TX;
    config.channel = LED_RMT_TX_CHANNEL;
    config.gpio_num = LED_RMT_TX_GPIO;
    config.mem_block_num = 1;
    config.tx_config.loop_en = false;
    config.tx_config.carrier_en = false;
    config.tx_config.idle_output_en = true;
    config.tx_config.idle_level = RMT_IDLE_LEVEL_HIGH;    // high because level shifter inverts it
    config.clk_div = 2;

    ESP_ERROR_CHECK(rmt_config(&config));
    ESP_ERROR_CHECK(rmt_driver_install(config.channel, 0, ESP_INTR_FLAG_IRAM));
    ESP_ERROR_CHECK(rmt_translator_init(config.channel, rmt_translate));
}
```

write function:
```
    ESP_ERROR_CHECK(rmt_wait_tx_done(LED_RMT_TX_CHANNEL, portMAX_DELAY));
    ESP_ERROR_CHECK(rmt_write_sample(LED_RMT_TX_CHANNEL, rgb, num_leds * 3, false));
```

Alvin1Zhang, 2020-03-19:
@cskilbeck Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

koobest, 2020-03-19:
Hi, @cskilbeck 
 Do you want to  drive an WS2812 LED strip with RMT?  you can refer to https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt/led_strip
thanks !!

cskilbeck, 2020-03-19:
I'll see if I can make a minimal repro, hang on

cskilbeck, 2020-03-19:
OK, the repro is as follows:
- start with the rmt/led_strip sample
- add the following at line 83 `config.mem_block_num = 8;`
- build and run
result is a crash

cskilbeck, 2020-03-19:
From looking at the debug log, it seems to be having a problem in `rmt_ll_write_memory`, called from line 519 in `driver/rmt.c`

```cpp
static void IRAM_ATTR rmt_fill_memory(rmt_channel_t channel, const rmt_item32_t *item,
                                      uint16_t item_num, uint16_t mem_offset)
{
    RMT_ENTER_CRITICAL();
    rmt_ll_set_mem_owner(p_rmt_obj[channel]->hal.regs, channel, RMT_MEM_OWNER_SW);
    rmt_ll_write_memory(p_rmt_obj[channel]->hal.mem, channel, item, item_num, mem_offset);
    rmt_ll_set_mem_owner(p_rmt_obj[channel]->hal.regs, channel, RMT_MEM_OWNER_HW);
    RMT_EXIT_CRITICAL();
}
```

cskilbeck, 2020-03-19:
Also worth mentioning (although this is really a separate bug) that when `mem_block_num` == 1 and # of items is large (1200 bytes in this case), occasional flickering occurs, _possibly_ because the ISR is not servicing the RMT peripheral adequately. This is the reason for using 8 mem_blocks, which seemed to fix the flickering.

koobest, 2020-03-20:
Hi,@cskilbeck
Thanks for reporting this issue, this is a issue introduced by driver refactoring. We will fix it soon.

thansk !!

cskilbeck, 2020-03-20:
Great, thanks for looking into it, let me know if you want anything tested.

atwoz, 2020-04-05:
Having the same issue. I think I'm using version 4.0 but I could be wrong. I would happily apply a patch to the driver so as to not wait for a full version release (in case you already have it).

cskilbeck, 2020-04-05:
I haven't looked into it, sorry, just waiting for a new version.

Alvin1Zhang, 2020-04-07:
@cskilbeck @atwoz Thanks for reporting and follow-up, we have a fix which is now under internal reviewing, will update once it is available on GitHub. Thanks.

koobest, 2020-04-07:
Hi, @atwoz
release 4.0 do not have this breaking change, Can you check your IDF version?

koobest, 2020-04-07:
Hi, @cskilbeck 

the fix of this issue will remove the line 

```length = (off + length) > SOC_RMT_CHANNEL_MEM_WORDS ? (SOC_RMT_CHANNEL_MEM_WORDS - off) : length;```

in 

```static inline void rmt_ll_write_memory(rmt_mem_t *mem, uint32_t channel, const rmt_item32_t *data, uint32_t length, uint32_t off)```

Can you help check if this fix is OK?

@atwoz Can you provide you IDF version?

thanks

cskilbeck, 2020-04-07:
I'll give it a go and report back.

3miliano, 2020-05-05:
@koobest I can confirm that the crash goes away when removing that line. Tested on v4.2.

closed 2020-06-12 by igrr