espressif/esp-idf #4960: Cannot exit OTA when disconnect with wifi hostpot

binhla, 2020-03-19:
When tried to FOTA my ESP32 device via a HTTP server, i tested to turn off my wifi hostpot (from my pc). And it seem like ESP32 stucked in this error

> E (53821) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
> E (53831) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
> E (53831) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
> E (53841) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
> E (53851) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
> E (53861) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
> E (53861) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55

any idea, and how to fix this. I dont want to have reset the ESP32 by reset button or turn off the device

Alvin1Zhang, 2020-03-19:
@binhla Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

binhla, 2020-03-19:
> @binhla Thanks for reporting. Would you please help provide more details as suggested in the [issue [...]

binhla, 2020-03-19:
> @binhla Thanks for reporting. Would you please help provide more details as suggested in the [issue [...]

Dears, thanks for replying, i added some information, sorry that i cannot share my code

Alvin1Zhang, 2020-03-30:
@binhla Thanks for reporting, we have a fix internally under reviewing now, will update once it is available on GitHub. Thanks.

closed 2020-04-28 by espressif-bot