espressif/esp-idf #4961: re-open issue RX IR signal from remote (IDFGH-2866) #4923

varunsavita, 2020-03-19:
@NatashaLi-ESPRESSIF @suda-morris  @
I am not able to receive data from AC remote
Please help guys

suda-morris, 2020-03-19:
the example only support NEC and RC5 protocol, if your AC remote control uses other protocols, you need to implement the decoder yourself.

varunsavita, 2020-03-19:
I have Mitsubishi AC remote and want to decode so what is the approach.
how to implement it?

negativekelvin, 2020-03-19:
Search GitHub for esp32 IR projects.  Ask on the forum.  If you are building a commercial device ask for support through your sales rep.  Otherwise it is unreasonable to ask for this type of support here.

varunsavita, 2020-03-20:
Thank you for your reply.
I have to make an IR receiver so it is not necessary to use other protocols.
Now I am searching remote which has NEC protocol.
Actually I send hardcore date from an IR LED and received something in IR_procotol examples but not which I send it may be modulation is different.
I share it with you then you give me some clarity of what is going on.

I (39980) example: Send command 0x32 to address 0x10
I (40020) example: Scan Code  --- addr: 0x912c cmd: 0xa4ea
I (42090) example: Send command 0x33 to address 0x10
I (42130) example: Scan Code  --- addr: 0xff00 cmd: 0xf807
I (44200) example: Send command 0x34 to address 0x10
I (46310) example: Send command 0x35 to address 0x10
I (48240) example: Scan Code  --- addr: 0x912c cmd: 0xa4ea
I (48420) example: Send command 0x36 to address 0x10
I (50350) example: Scan Code  --- addr: 0xff00 cmd: 0xf807
I (50530) example: Send command 0x37 to address 0x10
I (52640) example: Send command 0x38 to address 0x10
I (54750) example: Send command 0x39 to address 0x10
I (56460) example: Scan Code  --- addr: 0x912c cmd: 0xa4ea
I (56860) example: Send command 0x3a to address 0x10
I (58570) example: Scan Code  --- addr: 0xff00 cmd: 0xf807
I (58970) example: Send command 0x3b to address 0x10
I (61080) example: Send command 0x3c to address 0x10
I (63190) example: Send command 0x3d to address 0x10
I (64670) example: Scan Code  --- addr: 0x912c cmd: 0xa4ea
I (65300) example: Send command 0x3e to address 0x10
I (66780) example: Scan Code  --- addr: 0xff00 cmd: 0xf807

suda-morris, 2020-04-10:
In some ways, RMT just works like a transceiver, it doesn't have much to to with the IR protocol you choose. In IDF, we have supported two protocols: NEC and RC5, but there still be hundreds of other IR protocols on the market, it's impossible to include them all in IDF.
I'm afraid you need to implemented the protocol yourself if they're not NEC or RC5.
First of all, you should find the protocol spec, and maybe you also need to capture some waves sending from your AC remote control. When you got the data, you can take the existing the example as the reference to write your decoder. (the most hard part could be the representation of  bit zero and one I think.)

Alvin1Zhang, 2020-04-10:
@varunsavita Thanks for reporting, will close for now. Thanks.

closed 2020-04-10 by Alvin1Zhang