espressif/esp-idf #4962: HTTP_CLIENT: Error parse url esp32 when spaces are present in URL esp_http_client

jonhunterking66, 2020-03-19:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
    v4.1-dev-2140-g2976f5dae
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: VS Code
- Power Supply:         USB

## Problem Description

When sending an HTTP request and setting the url in the client configs, if I have a space in the url I get a parse url error. The reason my url has a space was because on of the sections of my database has a header with a space, for example "Data Status"

### Expected Behavior

No error, url is parsed correct and the HTTP request is performed. 

### Steps to reproduce

1. Perform a HTTP request with a URL that has a space in it

Not to much to this bug report. I have got around it by changing the way my database headers are labelled (no spaces!) but I think it should be fixed.

Alvin1Zhang, 2020-03-20:
@jonhunterking66 Thanks for reporting, we will look into.

jonhunterking66, 2020-03-20:
@Alvin1Zhang thanks!

shubhamkulkarni97, 2020-04-27:
Hi @jonhunterking66 

As defined by [RFC 3986](https://tools.ietf.org/html/rfc3986#section-2), space is not a URL safe character. Space must either be encoded using `percent encoding` or different encoding which uses URL safe characters.
You can replace `space` with `+` or `%20`

jonhunterking66, 2020-04-27:
@shubhamkulkarni97 ahhh, that is good to know. I appreciate the feedback! I think we can close this.

closed 2020-04-27 by jonhunterking66