espressif/esp-idf #4963: BLE mesh - cmds with ACK: Unable to identify which command is acknowledged

AlpinProgrammer, 2020-03-19:
## Environment

- Kit version : v3
- Module or chip used:  ESP32-WROVER-B
- IDF version: v4.0
- Build System:         Make
- Compiler version: (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

Hi,

I am currently facing a problem to identify which command the ACK received refers to. 

With an onoff client node, when sending an onoff command with ACK on the BLE mesh network, the ACK is received thru the _esp_ble_mesh_generic_cb_ (set by esp_ble_mesh_register_generic_client_callback). However, I do not see any way to identify which command is acknowledged.

I was expecting to find information such as the transaction ID.

### Expected Behavior

An ACK is received after sending a BLE mesh command. A way is provided to identify which command it refers to.

### Actual Behavior

An ACK is received after sending a BLE mesh command but it does not seem possible to find an id that would allow us to identify the command acknowledged.

### Steps to reproduce

1. Provision the onoff client example and bind the key.
2. Provision any onoff server and bind the key.
3. Send acknowledged onoff commands from the client to the server.

### Code to reproduce this issue

See _esp-idf-v4.0/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_client_

## Debug Logs

> I (323747) ble_mesh_client: esp_ble_mesh_generic_cb: event is 1, error code is 0, opcode is 0x8202
> I (323747) ble_mesh_client: ESP_BLE_MESH_GENERIC_CLIENT_SET_STATE_EVT
> I (323747) ble_mesh_client: ESP_BLE_MESH_MODEL_OP_GEN_ONOFF_SET: onoff 1

Thanks !

Alvin1Zhang, 2020-03-20:
@AlpinProgrammer Thanks for reporting, we will look into.

Campou, 2020-03-23:
Hi @AlpinProgrammer 

On the Generic OnOff Client model side, the Generic OnOff Status message (defined by SIG officially) received by the Generic OnOff Client model only contains the following fields, no TID field included.
[image]
TID is only included in the Geneirc OnOff Set/Set Unack message.

Thanks.

Alvin1Zhang, 2020-08-18:
@AlpinProgrammer Thanks for reporting, would you please help share if any updates for this issue? Thanks.

AlpinProgrammer, 2020-08-18:
@Alvin1Zhang Thank you for the assistance. This issue is actually solved because as long as a BLE mesh command hasn't been acknowledged or declared in timeout it is not possible to send another one.

closed 2020-08-18 by AlpinProgrammer