espressif/esp-idf #4964: Best JTAG Debug Environment For ESP32 (under Win10) - Eclipse or VS Code?

jwestmoreland, 2020-03-20:
Hello All,

I've been doing things via the command line in win10 thus far; but want to be able to debug via JTAG and a decent GUI - I've set up Eclipse and VS Code and have run into the various issues with various degrees of success - but, in the opinions of those here - which is the best environment for effective JTAG debugging - Eclipse or VS Code?

Also, I just picked up the ESP-Prog units - I do have a Segger - which one is better?

TIA,
John W.

sch0bert, 2020-03-20:
I think the best you can get is Eclipse because all the tools provided officially from Espressif are from it.
You should take a look at this demo:

https://www.youtube.com/watch?v=DqnZy4d03_s

jwestmoreland, 2020-03-20:
Already have.

I think this one is better (IMHO):
https://www.youtube.com/watch?v=KRyvly_SYS8
And this:
https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension
sdkconfig gui actually works too with that.

But, I wanted to see what the opinions were on all of this out there.  Once I get into any issue that doesn't follow what happens in the demos - they both seem to unravel rather quickly.  I didn't count on having to become an expert in .json programming in all of this.

Regards,
John W.

jwestmoreland, 2020-03-22:
Hello All,

Just an FYI - I'm using eclipse under Win10 - and I have one of the ESP-Prog units currently attached to my target ESP32 board - and it's working pretty good.

When VS Code supports this (via the plug in that is)  - I'll try that again.

[image]
 
If anyone is curious - I'm running the latest eclipse at this time: 

Eclipse IDE for C/C++ Developers
Version: 2020-03 (4.15.0)
Build id: 20200313-1211
v4.0 of the IDF

And, the sdkconfig tool does appear to work as well, which is nice.

Also, thanks to those that responded to me on and off-list - I appreciate all of the great suggestions.

Best Regards,
John W.

tom-borcin, 2021-05-10:
Hello John @jwestmoreland,

thank you for opening this issue and sharing your experience with our userbase. We think the issue is resolved and there is nothing left to do from our side, therefore we are closing this issue to keep our backlog small and manageable. If you think we should keep it open, please let me know.

Best Regards,

Tomas Borcin

closed 2021-05-10 by tom-borcin