espressif/esp-idf #4966: Bluetooth: BTC task name mismatch

hb-kaertech, 2020-03-20:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v4
- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.2-dev-404-gd6e672ed5
- Build System: idf.py
- Compiler version: (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description
The BT task has two names.

In `components/bt/host/bluedroid/stack/btu/btu_init.c`:
`btcT` at https://github.com/espressif/esp-idf/blob/fb3d648e07033b952578f187b4fcbf7fd8aced3b/components/bt/common/btc/core/btc_task.c#L71 (currently unused)
and
`BTC_TASK` at https://github.com/espressif/esp-idf/blob/fb3d648e07033b952578f187b4fcbf7fd8aced3b/components/bt/common/btc/core/btc_task.c#L336

I believe it was introduced in 9691fe9daefe0242c7f899f0f26bb0b26d4f16db.

## Proposed Fix
- Use `BTC_TASK_NAME` in `osi_thread_create`
- Decide which `BTC_TASK_NAME` you want to keep: either `btcT` or `BTC_TASK`

Wth-Esp, 2020-09-02:
Hi,

We have already changed the task name, please try to pull the latest IDF.
And please close this issue if you have confirmed the changes.

Thanks
Weitianhua

hb-kaertech, 2020-09-02:
Hello @Wth-Esp,

I still see the issue on the current master branch. `BTC_TASK_NAME` is unused.

Wth-Esp, 2020-09-02:
Hi @hb-kaertech 

The code should sync to Github after several days.

Thanks
Wei Tianhua

closed 2020-09-16 by espressif-bot