espressif/esp-idf #4967: HFP AG example: outgoing callback not triggered

pavestru, 2020-03-20:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version v4
- Module or chip used:  ESP32-WROVER-B
- IDF version: v4.2-dev-781-g47253a827
- Build System: make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     macOS
- Using an IDE?: No
- Power Supply:      USB

## Problem Description

ESP32 connects to a BT headset, receives audio from BT headset (CVSD codec), however, it is not sending audio to the same BT headset. Callback `bt_app_hf_outgoing_cb` (see [here](https://github.com/espressif/esp-idf/blob/47253a827a80cb78f22db8e4736c668804c16750/examples/bluetooth/bluedroid/classic_bt/hfp_ag/main/bt_app_hf.c#L120-L133)) almost never gets called. If it ever does, it is followed by a crash exactly like described in #4794 . 

### Expected Behavior

Callback `bt_app_hf_outgoing_cb` is expected to get called to play the [sine wave](https://github.com/espressif/esp-idf/blob/47253a827a80cb78f22db8e4736c668804c16750/examples/bluetooth/bluedroid/classic_bt/hfp_ag/main/bt_app_hf.c#L120-L133) in the headphones.

### Actual Behavior

Callback `bt_app_hf_outgoing_cb` is not run at all except very rarely and randomly, followed by crashes exactly like described in #4794 .

### Steps to reproduce

1. Open [HFP AG](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/hfp_ag example)
1. put a `ESP_LOGI("...");` statement to both [`bt_app_hf_outgoing_cb`](https://github.com/espressif/esp-idf/blob/47253a827a80cb78f22db8e4736c668804c16750/examples/bluetooth/bluedroid/classic_bt/hfp_ag/main/bt_app_hf.c#L120-L133) and [`bt_app_hf_incoming_cb`](https://github.com/espressif/esp-idf/blob/47253a827a80cb78f22db8e4736c668804c16750/examples/bluetooth/bluedroid/classic_bt/hfp_ag/main/bt_app_hf.c#L135-L140)
1. Follow [README](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/hfp_ag) in hfp_ag example
1. Uncomment [this line](https://github.com/espressif/esp-idf/blob/47253a827a80cb78f22db8e4736c668804c16750/examples/bluetooth/bluedroid/classic_bt/hfp_ag/main/app_hf_msg_set.c#L16) and put mac address of your BT headset there
1. Set HCI data path in menuconfig
1. leave `Bluedroid Options --> Wide Band Speech` off
1. set `BR/EDR Sync(SCO/eSCO) Max Connections` to 1 or more (default is 0)
1. `make flash monitor` (runs HFP AG example)
1. `hf con;` (if not connected yet)
1. `hf cona;`

### Code to reproduce this issue

Example code for callbacks in bt_app_hf.c:

```cpp 
static uint32_t bt_app_hf_outgoing_cb(uint8_t *p_buf, uint32_t sz)
{
    ESP_LOGE(BT_HF_TAG, "bt_app_hf_outgoing_cb");
    
    int sine_phase = esp_random();

    for (int i = 0; i < TABLE_SIZE_CVSD; i++) {
        p_buf[i * 2]     = sine_int16[sine_phase];
        p_buf[i * 2 + 1] = sine_int16[sine_phase];
        ++sine_phase;
        if (sine_phase >= TABLE_SIZE_CVSD) {
            sine_phase -= TABLE_SIZE_CVSD;
        }
    }
    return sz;
}
 
static void bt_app_hf_incoming_cb(const uint8_t *buf, uint32_t sz)
{
    ESP_LOGE(BT_HF_TAG, "bt_app_hf_incoming_cb, buf size: %d", sz);
    // direct to i2s
    esp_hf_outgoing_data_ready();
}
```

## Note

I am using Sony SBH20 and Sony WH-1000XM3 headsets, both have the same behaviour.

Thank you.

Alvin1Zhang, 2020-03-23:
@pavestru Thanks for reporting, we will look into.

KollarRichard, 2020-04-04:
Hi, is there any news on this issue?

ypiotrp, 2020-06-16:
Hi. I got the same issue on the newest master. What is the status here? Is it a work in progress?

howroyd, 2020-07-03:
Same issue here.  The outgoing callback hasn't been called once, but the incoming is firing as expected.

howroyd, 2020-07-24:
Just tested on latest pull from `release v4.2` and still got the issue.

My log is just being flooded with the following.  Incoming audio (from HFP client) seems ok, but a little glitchy.  Certainly nowhere near as stable as A2DP.

```
W (22802) BT_APPL: bta_ag_sco_read_cback: status(2)
```

glennhuber, 2020-09-23:
Just tested on 4.3 and still not working. Is this issue being looked into at Espressif?

ypiotrp, 2020-10-12:
Hi @Alvin1Zhang! Have made any progress on this issue?

Alvin1Zhang, 2020-10-13:
Thanks for the updates, and sorry for slow turnaround, we are still debugging this issue, will share updates once available. Thanks.

boblane1, 2020-10-13:
Hi, @pavestru and other users!
I offer a patch to fix this problem, you can have a try. Please let me know if it still has problems. By the way, it is not recommended to print too much information in the callback function.

[0001-fix-ag-example-outgoing-callback-not-triggered.zip](https://github.com/espressif/esp-idf/files/5369340/0001-fix-ag-example-outgoing-callback-not-triggered.zip)

ypiotrp, 2020-10-13:
Hi @boblane1!
I applied the patch you shared to release v4.1 ESP-IDF and this is the result that I got - https://www.youtube.com/watch?v=peONupbGCGs. I don't know if this is what should be expected but it is definitely a step up. 

Thank you very much! 

For everyone else. To reproduce this effect checkout the ESP-IDF at `v4.1`, apply the patch from the @boblane1's post, change the SCO path to HCI, disable the Wide Band Speech and change the BR/EDR Sync(SCO/eSCO) Max Connections to 1. Compile, flash, voila!

closed 2020-11-03 by espressif-bot