espressif/esp-idf #4969: esp_http_client_example doesn't enable ESP_HTTP_CLIENT_ENABLE_BASIC_AUTH

KaeLL, 2020-03-20:
Title. [basic_auth_redirect](https://github.com/espressif/esp-idf/blob/47253a827a80cb78f22db8e4736c668804c16750/examples/protocols/esp_http_client/main/esp_http_client_example.c#L270) fails due to it.

Alvin1Zhang, 2020-03-23:
@KaeLL Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

KaeLL, 2020-03-23:
The [example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/esp_http_client) assumes basic auth is enabled. Since [it isn't by default](https://github.com/espressif/esp-idf/blob/6330b3345e87eb4401e7be7c8b6fea2870c35d9f/components/esp_http_client/Kconfig#L10), and there's not supplementary Kconfig parameter enabling it the example, this [client_perform](https://github.com/espressif/esp-idf/blob/6330b3345e87eb4401e7be7c8b6fea2870c35d9f/examples/protocols/esp_http_client/main/esp_http_client_example.c#L277) fails with != ESP_OK.
Given that it takes a lot more work to provide that to you than you just building the example, running, and looking at the output, or just working out the explanation I just gave, and that the entire issue is something so small that it's definitely not worth the effort of providing all that, I won't help you.

Alvin1Zhang, 2020-03-23:
@KaeLL Thanks for providing the additional information. We will look into. Thanks.

shubhamkulkarni97, 2020-03-24:
@KaeLL I'm attaching a patch which should fix your issue. Please try the patch and feel free to post if you have any issue regarding the same.
[http_client_fix_basic_auth.patch.zip](https://github.com/espressif/esp-idf/files/4373361/http_client_fix_basic_auth.patch.zip)

closed 2020-03-25 by espressif-bot