espressif/esp-idf #4970: WiFi - Need an API for detecting WiFi NVS configuration existence/validness

huming2207, 2020-03-20:
**Is your feature request related to a problem? Please describe.**

Yes, it's related to a problem. See here for more details: https://www.esp32.com/viewtopic.php?f=13&t=14720

**Describe the solution you'd like**

I would like to know is there an API for detecting the validness or existence of WiFi configuration stored in NVS. I might have misread the documentation, please point it out if I am wrong. But if there is no such API, I would like to suggest we should add one for that.

For example, the API can be something like:

```c
bool esp_wifi_has_config(wifi_interface_t interface);
```
It returns true if a valid configuration is set in NVS for the related WiFi interface, or false if it is not valid or doesn't exist.

Or alternatively, `esp_wifi_get_config()` should return `ESP_ERR_NVS_NOT_FOUND` instead of `ESP_OK` if the configuration is invalid.

**Describe alternatives you've considered**

My current workaround is either detecting the return config struct from `esp_wifi_get_config()` (as I've mentioned in the forum thread), or alternatively add a flag in a separated NVS namespace to achieve that. But the first option may have flaws (mentioned in the forum thread) and the second option causes storage overheads in NVS.

**Additional context**

N/A

Alvin1Zhang, 2020-03-23:
@huming2207 Thanks for your detailed explanation on the feature request, we will look into and evaluate.

huming2207, 2020-03-23:
@Alvin1Zhang Yep thanks. This could be very helpful in my opinion. It won't be too hard to implement, just a magic NVS value/flag is fine.

liuzfesp, 2020-03-23:
Hi @huming2207, the current implementation of WiFi NVS is: always initialize WiFi NVS to a valid and default value if the customer doesn't configure the WiFi NVS. 

From WiFi internal perspective, it's not as easy as you thought for us to support the API `esp_wifi_get_config()`. Anyway, we are now considering to implement it, we hope we can support it on late April or early May. Hope it's OK for you.

huming2207, 2020-03-23:
@liuzfesp Ah ok thanks for pointing that out. I thought the NVS is just for WiFi configurations.

But yea I'm really looking forward to seeing a standalone API for detecting the validness. This would save a bit of storage overhead and code size. Also, it would help to make the code looks a bit neater.

esp-HarveyRong, 2020-06-03:
Hi @huming2207,
We are sorry to say that we have fully discussed this issue internally, and WiFi will not add a similar API. Because this API will introduce some OTA issue, it will bring many challenges to the subsequent maintenance of WiFi.

huming2207, 2020-06-03:
@esp-HarveyRong Okay but I'm not quite understood why adding such a detection API will cause OTA issue... But anyway I will do it externally instead. Thanks.

closed 2020-06-03 by huming2207