espressif/esp-idf #4972: esp_log prints out of order

chegewara, 2020-03-22:
How is it possible running on esp32 with only 1 core (on AWSFreeRTOS) to have printed logs with wrong order or with wrong timestamp (ticks count):
```
I (15099) SHADOW: debugPrintDeviceState:(save_state_to_nvs) checkSum => 119, shadowVersionMajor => 1, shadowVersionMinor => 2
I (15109) SHADOW: debugPrintDeviceState:(save_state_to_nvs) ValveType => 0, ValveState => 0, ValveStateReq => 0, valveStateCounter => 1
I (15119) SHADOW: debugPrintDeviceState:(save_state_to_nvs) adcNumSamp => 32, mtrDelayTime => 250
I (15129) SHADOW: debugPrintDeviceState:(save_state_to_nvs) mtrMaxI => 80, mtrMaxT => 3, mtrBackT => 500
I (15139) SHADOW: debugPrintDeviceState:(save_state_to_nvs) connect_not_flowing => 30
I (15149) SHADOW: debugPrintDeviceState:(save_state_to_nvs) connect_sleep_multiplier => 4, provisioned => 1
I (15159) SHADOW: compareNVM: checkSum
I (14939) OTA: vRunOTAUpdateDemo: State: RequestingJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
I (15179) SHADOW: save_state_to_nvs: nvs_commit succeeded
I (15179) SHADOW: 77 01 02 00 00 20 fa 00 50 00 03 00 f4 01 00 00 
I (15189) SHADOW: 01 00 1e 00 04 00 01 00 
I (15189) SHADOW: updateShadowTask: _sendShadowUpdates successful
I (15199) SHADOW: updateShadowTask: posting/releasing shadowSyncSemaphore
I (15209) SHADOW: updateShadowTask: posting/releasing shadowSemaphore
I (15209) DEVICE: device_state_task: Processing WiFi/PWR LED batt_stat(0)
```
This one is definitely not right:
`I (14939) OTA: vRunOTAUpdateDemo:`

esp-idf version 3.3 (awsfreertos) 
`boot: ESP-IDF v3.3-163-g601a03e 2nd stage bootloader`

mahavirj, 2020-03-23:
@chegewara Are these logs mapped to `ESP_LOGx` or they are using `logging` component from https://github.com/aws/amazon-freertos/tree/master/libraries/c_sdk/standard/common/logging?

chegewara, 2020-03-23:
ESP_LOGx, all of them, even that one out of order.
Yes, i know that AWS logging system is using special task which is delayed.

mahavirj, 2020-03-23:
@chegewara Seems like they are mapped to `logging` component, see https://github.com/aws/amazon-freertos/blob/master/demos/ota/aws_iot_ota_update_demo.c#L261 (`configPRINTF` -> `vLoggingPrintf`)

chegewara, 2020-03-23:
As far as i understand AFR internal logging like `IotLogInfo` is using task with Queue system to print logs:
```
xLoggingTaskInitialize(mainLOGGING_TASK_STACK_SIZE,
                           tskIDLE_PRIORITY + 5,
                           mainLOGGING_MESSAGE_QUEUE_LENGTH);
```
ESP_LOGx is printed in place, just like in esp-idf.

This single "incident" is not a big issue to me and i posted it here just in case that may be some bug in esp-idf. 
If you think its more related to AFR logging system feel free to close that issue and it can be opened in future if someone find similar problem, especially it is very rare and i believe its not possible to reproduce.

mahavirj, 2020-03-27:
@chegewara Will go ahead and close this. If you could come up with test code then please feel free to re-open.

closed 2020-03-27 by mahavirj