espressif/esp-idf #4973: cannot find python file or directory

baptistepattyn, 2020-03-22:
## Environment

- Development Kit:      [ESP32-DevKitC|]
- Kit version (DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    v4.2-dev-792-g6330b3345
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    1.22.0-80-g6c4433a5
- Operating System:     [Windows]
- (Windows only) environment type: [MSYS2 mingw32].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

When trying to execute make menuconfig in a project I am getting the following error:

make: python: No such file or directory
make: *** No rule to make target 'check_python_dependencies', needed by 'menuconfig'.  Stop.

### Expected Behavior
To launch the config to set the settings for the project
### Actual Behavior
Gives error and gets stuck
### Steps to reproduce

1. install esp-idf and toolchain
2. copy the hello_world project
3. run make menuconfig 

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue
No code needed
## Debug Logs

```
no debug logs
```

## Other items if possible
no other items

bitmakk, 2020-03-22:
Add the python.exe to your PATH

baptistepattyn, 2020-03-22:
[image]
you mean like this? :)

dobairoland, 2020-03-22:
Hi @baptistepattyn. Please don't use the globally installed "windows" python in MSYS. Use the one installed with package `mingw-w64-i686-python`.

The pre-packaged MSYS environment for IDF >v4.0 can be downloaded here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started-legacy/windows-setup.html#toolchain-setup

But personally I strongly encourage you to use the build system from Windows command line if you are using IDF >v4.0: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#esp-idf-tools-installer

baptistepattyn, 2020-03-22:
The installer you reccommend gives me an error too... I installed the pre-packaged environment but where is the python folder then located that i need to make a path to?

dobairoland, 2020-03-22:
> The installer you reccommend gives me an error too... 

Please take a look at the related open issues and report your issue (with the full error log file) if you encountered something which wasn't reported yet.

> I installed the pre-packaged environment but where is the python folder then located that i need to make a path to?

Everything is already setup for the pre-packaged ZIP-ed environment and no further setup is required. If you run  `python --version` then you should see the printed version of the Python interpreter.

baptistepattyn, 2020-03-22:
Does it matter if I have other instances of Python installed?

baptistepattyn, 2020-03-22:
thanks guys I managed to make it work

closed 2020-03-22 by baptistepattyn

Cloret20, 2022-06-09:
@baptistepattyn Is it possible you can share the solution how to fix the "make: python: No such file or directory"? I. just enter the same issue and have no clue.