espressif/esp-idf #4974: ESP websocket client, send large payloads results incorrect packets

jessekerkhoven, 2020-03-22:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-WROVER
- IDF version: V4.0
- Build System:  Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:  macOS
- Using an IDE?: No
- Power Supply: USB

## Problem Description

When sending data via `esp_websocket_client_send_text` with payload larger than the default buffer size, the payload is split in packets that fit the buffer size. But as a result of the underline code all packets are send with FIN bit and OPCODE set. Which results incorrect behaviour at the receiving end. 

### Expected Behavior

According [RFC6455 section 5.4](https://tools.ietf.org/html/rfc6455#section-5.4) fragmentation can be used to split the payload accordingly and set the FIN bit and OPCODE correctly for each part of the split payload.

### Actual Behavior

When using [esp_websocket_client_send_text](https://github.com/espressif/esp-idf/blob/463a9d8b7f9af8205222b80707f9bdbba7c530e1/components/esp_websocket_client/esp_websocket_client.c#L564), the underlying [esp_websocket_client_send_with_opcode](https://github.com/espressif/esp-idf/blob/463a9d8b7f9af8205222b80707f9bdbba7c530e1/components/esp_websocket_client/esp_websocket_client.c#L603-L616) split it in multiple parts if that is required and uses [esp_transport_ws_send_raw](https://github.com/espressif/esp-idf/blob/463a9d8b7f9af8205222b80707f9bdbba7c530e1/components/tcp_transport/transport_ws.c#L249) to write it to the transport layer where the FIN bit and OPCODE is always set. 

### Steps to reproduce

Use the example in the project folder [Websocket example](https://github.com/espressif/esp-idf/tree/v4.0/examples/protocols/websocket) and try to send data larger than the default (1024) buffer size. 

PS. Thanks for the great work for ESP32 and ESP-IDF.

Alvin1Zhang, 2020-04-14:
@jessekerkhoven Thanks for reporting, we have a fix which is now under internal reviewing, will update once it is available on GitHub. Thanks.

jessekerkhoven, 2020-04-14:
@Alvin1Zhang Thanks for the update, looking forward to the update.

closed 2020-04-24 by espressif-bot

jessekerkhoven, 2020-05-21:
@Alvin1Zhang Can I ask what the reason is to not include this bug fix in the v4.0.1 release? And based on the release branches it is only included in v4.2 release branch. It would be great to get it backported to 4.0 or 4.1.

Alvin1Zhang, 2020-05-22:
@jessekerkhoven Thanks for the updates. The fix will be back ported to other release branches, including IDF v4.0, and IDF v4.0.1 has been released at https://github.com/espressif/esp-idf/releases/tag/v4.0.1 on May 20th, 2020.

The fix on v4.0 is still under internal review and in our merge queue, it will be included in the next IDF v4.0 bug fix release. Thanks for your patience waiting for the fix, will update once the fix on IDF v4.0 is available on GitHub. Thanks.

jessekerkhoven, 2020-05-22:
@Alvin1Zhang Thanks for the quick reply. Great to hear that the fix will be backported to v4.0.