espressif/esp-idf #4975: smartconfig: SC version: V2.8.0  ESPTOUCH配网失败

shenyong-iot01, 2020-03-23:
您好，
    TP-Link TL-WDR7660千兆版  smartconfig: SC version: V2.8.0  ESPTOUCH 安卓9以上系统无法配网成功，请帮忙确认，iphone8测试也配网失败，暂时未测试过多版本！
设备打开log选项也无多余信息输出，
  esp_err = esp_smartconfig_start(smartconfig_callback, 1);

I (2350) wifi: mode : sta (24:6f:28:50:a8:c4)
I (2400) smartconfig: SC version: V2.8.0
smart config task
unsol packet:
03 00 91 F3 00 

unsol packet:
07 00 91 F1 05 00 00 01 46 

ble gw sw ver:T5.00.M00.B001_r06
I (6500) wifi: ic_enable_sniffer
[INFO] SC_STATUS_FINDING_CHANNEL

1、使用vivo X6D（android 5.1）、荣耀6X（android8.0）测试，可以配网成功

2、使用Mate10（android9）、nova5 Pro（android9、android10）、oppo A7x（android9）均无法配网成功

Alvin1Zhang, 2020-03-23:
@shenyong-iot01 Thanks for your report. We will look into. Thanks.

zhangyanjiaoesp, 2020-03-23:
@shenyong-iot01 
Which IDF version do you use? What's the commit ID? Your version is too old, currently the smartconfig version is 2.9.0, please update your IDF code and try again, thanks.

shenyong-iot01, 2020-03-23:
@zhangyanjiaoesp 我这边使用的是3.2.2SDK   经验证3.3.1SDK内配网库版本同样是SC version: V2.8.0 
3.3.1是个长期支持的版本，为什么没有更新配网库呢？smartconfig version 2.9.0是在 ESP-IDF v4.0上么，此版本更改较大直接升级到这个SDK会严重影响项目进度。
Documentation for release v4.0 is available at https://docs.espressif.com/projects/esp-idf/en/v4.0/

ESP-IDF v4.0 is a major update for ESP-IDF v3.x. Release v4.0 is mostly compatible with apps written for ESP-IDF v3.x, but there are some breaking changes (please consult the list below) and removal of deprecated functionality which will require code changes when updating projects

ESP-IDF v4.0.x will be supported for 18 months until October 2021, according to the ESP-IDF Support Policy. The current Long Term Support release is v3.3.1, v3.3.x will be supported until March 2022.

zhangyanjiaoesp, 2020-03-23:
@shenyong-iot01 
I just confirm that the smartconfig version is v2.8.0. Since the version number is different from v4.0, but for important changes, we will merge into all release version. So the smartconfig is updating all the time. Don't worry about it.
Have you tested with the 3.3.1 sdk? Does the problem still exist?

shenyong-iot01, 2020-03-23:
经验证，使用我司APP android 配网库0.3.7.0无法配网（同时发送广播和组播），使用乐鑫最新配网app demo 配网库0.3.7.2广播可正常配网，组播无法配网。固件SDK3.2.2和3.3.1现象一样。后续将我司app更新下配网库，再验证下！

shenyong-iot01, 2020-03-24:
修正下，我司APP使用的是组播配网；android9.0以上手机TP-Link TL-WDR7660千兆版无法配网；使用官方APP demo结果一样。今天又测试华为p9android8.0组播无法配网。更换路由器配网成功。请帮忙分析下原因，TP-Link TL-WDR7660千兆版这款路由器销量比较大，谢谢！

shenyong-iot01, 2020-03-24:
@zhangyanjiaoesp 修正下，我司APP使用的是组播配网；android9.0以上手机TP-Link TL-WDR7660千兆版无法配网；使用官方APP demo结果一样。今天又测试华为p9android8.0组播无法配网。更换路由器配网成功。请帮忙分析下原因，TP-Link TL-WDR7660千兆版这款路由器销量比较大，谢谢！

zhangyanjiaoesp, 2020-03-24:
@shenyong-iot01 
I will try to use the "smart_config" example to reproduce this problem. Did you use BLE at the same time, I see this info `ble gw sw ver:T5.00.M00.B001_r06` in your logs.

shenyong-iot01, 2020-03-24:
@zhangyanjiaoesp ESP32只使用WiFi，ble gw 是另外一个模块

zhangyanjiaoesp, 2020-03-24:
hi @shenyong-iot01 ,
I use Mate10（android9）for local testing, when I set the TP-Link TL-WDR7660 to `bgn` mode, the ESP-TOUCH failed. But when I set the TP-Link TL-WDR7660 to `bg` mode, the ESP-TOUCH succeed.

And I have capture packets. When the router in `bgn` mode, the beacon frame contain VHT info, so we doubt that the router and phone sending UDP multicast with 11ac rate. So we suggest you set the router to `bg` mode.
[image]

shenyong-iot01, 2020-03-25:
@zhangyanjiaoesp 
经验证，bg模式可以正常配网！更改设置只能作为一个临时措施，终端用户一般很少去设置这些配置。问题您这边已经复现，是否有后续修复计划及修复版本发布请及时通知！谢谢！

zhangyanjiaoesp, 2020-03-25:
> @zhangyanjiaoesp
> 经验证，bg模式可以正常配网！更改设置只能作为一个临时措施，终端用户一般很少去设置这些配置。问题您这边已经复现，是否有后续修复计划及修复版本发布请及时通知！谢谢！

@shenyong-iot01 The TL_WDR7660 us AC rate to send data, and ESP32 doesn't support 802.11 ac currently, so ESP-TOUCH failed in `bgn` mode with TL-WDR7660. But with other routers ,the ESP-TOUCH can succeed in 'bgn' mode. 

We will support 802.11 ac, but it will take a long time, thanks.

shenyong-iot01, 2020-03-25:
@zhangyanjiaoesp 
1. 若是11ac问题，我们会将bug挂起！
2. bgn模式发送广播为什么会走11ac不是太明白，手机和模块都连接的2.4GWiFiSSID并未连接5GwifiSSID；另外使用vivo X6D（android 5.1）、荣耀6X（android8.0）测试，可以配网成功，和android版本还有关系，请一起评估，谢谢！

zhangyanjiaoesp, 2020-03-25:
OK, we will do more test to find the root cause.

zhangyanjiaoesp, 2020-07-01:
@shenyong-iot01 
I have captured the UDP packets from the  Mate10（android9）when TP-Link TL-WDR7660 to bgn mode, as the following:

[image]

The `Spatial Stream=2`, and the ESP32 will discard this MIMO packets, at the same time, the router won't transmit the  Multicast packets, so the ESPTOUCH fail.

And I have tested on Honor 7x（Android 7.0 ), the UDP packets is as the following:

[image]

The Spatial Stream=1, and the ESP32 can handle this packets correctly, even though the router won't transmit the Multicast packets, the ESPTOUCH can succeed.

Alvin1Zhang, 2020-07-09:
Thanks for reporting, feel free to reopen if you still have the issue, and please help file a new ticket for new issue. Thanks.

closed 2020-07-09 by Alvin1Zhang