espressif/esp-idf #4976: 有时MQTT连接异常 HOST DNS无法退出

shenyong-iot01, 2020-03-23:
您好，
    使用3.2.2IDF SDK， MQTT直接使用3.3.1SDK内版本（直接调用接口，未做代码更改），有时重启设备出现mqtt连接失败
[12:10:13.629] E (59362) MQTT_CLIENT: Error transport connect
[12:10:17.606] [APP MQTT] MQTT_EVENT_DISCONNECTED
等待大概7min后连接成功，或重启设备后连接成功 wireshark抓包如附件，过滤器
eth.addr == 3c:71:bf:6e:19:44  开始时间2020-03-20  14：42：48  请帮忙确认。
[联网.zip](https://github.com/espressif/esp-idf/files/4367112/default.zip)

david-cermak, 2020-03-25:
Hi @shenyong-iot01 

Thanks for reporting this issue. It really looks like the connection gets stuck in DNS lookup. Not clear to me what the reason could be.
Could you please share the exact version of IDF you use?

AxelLin, 2020-03-26:
Might be related to https://github.com/espressif/esp-idf/issues/4510 since mqtt uses transport_tcp which uses gethostbyname().

shenyong-iot01, 2020-03-26:
@AxelLin 使用3.2.2IDF SDK， MQTT直接使用3.3.1SDK内版本 。代码定位应该走的是esp_tcp_connect（）  resolve_host_name（） getaddrinfo（） 谢谢！

david-cermak, 2020-03-31:
@shenyong-iot01  
> esp_tcp_connect（） resolve_host_name（） getaddrinfo（）

Can you confirm that this is the actual order of calls in your application? Seems like the `test.usa.goodtime-iot.com` is treated as a plain tcp by default. This could be overridden in user code forcing the ssl_transport to connect (and using `getaddrinfo()` for dns lookup) -- via `esp-tls` in non-tls mode -- but a not default setup.

@AxelLin Thanks for referencing the issue!

shenyong-iot01, 2020-04-01:
@david-cermak 谢谢你的回复！函数确定走的上面esp_tcp_connect（） resolve_host_name（） getaddrinfo（）来获取DNS，我们在测试时发现可能是路由器端配置问题（当前模块连接的路由WiFI经有线桥接到上级路由，上级路由器来dhcp ip地址，去掉桥接路由直接使用上级路由器释放热点，此问题消失），此问题可挂起，谢谢！

Alvin1Zhang, 2020-04-01:
@shenyong-iot01 Thanks for your updates and glad the issue is resolved. Will close the issue for now, feel free to reopen the issue if it happens again. Thanks.

closed 2020-04-01 by Alvin1Zhang