espressif/esp-idf #4977: The idf.py of ESP-IDF setup2.3 doesn't work because it raises UnicodeDecodeError: 'cp932' codec can't decode.

lovyan03, 2020-03-23:
The default code page for Japanese python is cp932, and if you don't specify the encoding when loading the file, you may get an error.

[image]

This can be solved by adding an encoding specification when the file is opened.

```diff
--- /esp-idf/tools/kconfig_new/kconfiglib.py
+++ /esp-idf/tools/kconfig_new/kconfiglib.py
@@ -1125,7 +1125,7 @@
         was set when the configuration was loaded.
         """
         try:
-            return open(filename)
+            return open(filename, encoding="utf-8")
         except IOError as e:
             if not os.path.isabs(filename) and self.srctree is not None:
                 filename = os.path.join(self.srctree, filename)
```

dobairoland, 2020-03-23:
Hi @lovyan03. Thank you for reporting this issue. 

Please give us your IDF version (`git describe --tags`) and your Python version (`python --version`).

I'm guessing that your IDF version is v4.0 or older because this seems to been fixed in the upstream kconfiglib library which is shipped with IDF v4.1 and newer. 

Would it be possible that you check the latest IDF master branch to verify that you have no issue there?

lovyan03, 2020-03-23:
Hello !
thank you for reply.

I used the installer esp-idf-tools-setup-2.3.exe, and in Download ESP-IDF I chose v4.0.

git describe --tags
v4.0

python --version
Python 3.7.3

I tried to synchronize ESP-IDF, but I didn't know how to use it and couldn't build it.

dobairoland, 2020-03-23:
Thank you. Please use your workaround until we fix this.

lovyan03, 2020-03-23:
Thank you !
I used the installer again and tried to select develop master when choosing ESP-IDF.
You're right, the problem has been fixed and I can now build it.

dobairoland, 2020-03-23:
That is great!

I keep this issue open until this is fixed for IDF <=v4.0.

closed 2020-04-02 by dobairoland