espressif/esp-idf #4978: ble app reg数量的问题

fzboffice, 2020-03-23:
在menuconfig中把 BT/BLE MAX ACL CONNECTIONS设置为7，但是当我使用时，我需要同时运行gattc和gatts，但是我使用esp_ble_gattc_app_register和esp_ble_gatts_app_register最多只能注册5个gattc的profile和1个gatts的profile，我想知道为什么这个数字小于7

fzboffice, 2020-03-23:
我使用的是release v4.0

WCCWCC, 2020-03-24:
Hi @fzboffice ,
*  Can you provide the error log, or the complete project for me,
*  I need to reproduce this problem first
* I will help you solve the problem as soon as possible

fzboffice, 2020-03-25:
> 嗨@fzboffice，
> 
> * 您能否提供错误日志或完整的项目给我，
> * 我需要先重现此问题
> * 我会帮助您尽快解决问题

我使用example里面的两个例子，一个是gatts一个是gattc multi connect ，合并后同时打开gatts和gattc

WCCWCC, 2020-03-26:
Hi, @fzboffice 
     我已经找到原因了，目前最多可以支持注册8 个 profile，不区分gatts 与 gattc。 协议栈 GATT 和 GAP 内部各使用了一个，所有只能注册 6 个profile。  
     你可以将 /components/bt/host/bluedroid/common/include/common/bt_target.h 中 GATT_MAX_APPS 增大。

[image]

补充：当时 GATT_MAX_PHY_CHANNEL 表示多连接时支持的 gattc profile 的个数。

fzboffice, 2020-03-27:
> Hi, @fzboffice
> 
> * 已经找到原因了，目前最多可以支持注册8 个 profile，不区分gatts 与 gattc。 协议栈 GATT 和 GAP 内部各使用了一个，所有只能注册 6 个profile。
> * 你可以将 /components/bt/host/bluedroid/common/include/common/bt_target.h 中 GATT_MAX_APPS 增大。
> * [image]
> 
> 补充：当时 GATT_MAX_PHY_CHANNEL 表示多连接时支持的 gattc profile 的个数。

thank you

closed 2020-03-27 by fzboffice