espressif/esp-idf #4979: CMake Error at CMakeLists.txt:85

Tiagodsmanin, 2020-03-23:
Everytime that I close the terminal and open another tab, this problem poup up: 

Executing action: all (aliases: build)
Running cmake in directory /home/tiago/Reagin/esp/esp-idf/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 /home/tiago/Reagin/esp/esp-idf"...
Warn about uninitialized values.
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:85 (idf_build_set_property):
  Unknown CMake command "idf_build_set_property".

-- Configuring incomplete, errors occurred!
See also "/home/tiago/Reagin/esp/esp-idf/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

Alvin1Zhang, 2020-03-24:
@Tiagodsmanin Thanks for reporting, we will look into.

renzbagaporo, 2020-04-07:
@Tiagodsmanin It seems you are building on the `$IDF_PATH` root? Please `cd` into an example directory such as `$IDF_PATH/examples/get-started/hello_world`.

closed 2020-04-30 by renzbagaporo