espressif/esp-idf #4980: Error in I2S example i2s_example_main.c

i-franzki, 2020-03-23:
Not really a bug, but something that noticed when looking at the the I2S example under 
https://github.com/espressif/esp-idf/blob/master/examples/peripherals/i2s/main/i2s_example_main.c

Function setup_triangle_sine_waves() is supposed to calculate one full cycle of the sine and triangle wave in the desired wave frequency.  However, it calculates more or less than a full cycle, dependent on sample rate and desired wave frequency.

```
static void setup_triangle_sine_waves(int bits)
{
    int *samples_data = malloc(((bits+8)/16)*SAMPLE_PER_CYCLE*4);
    unsigned int i, sample_val;
    double sin_float, triangle_float, triangle_step = (double) pow(2, bits) / SAMPLE_PER_CYCLE;
    size_t i2s_bytes_write = 0;

    printf("\r\nTest bits=%d free mem=%d, written data=%d\n", bits, esp_get_free_heap_size(), ((bits+8)/16)*SAMPLE_PER_CYCLE*4);

    triangle_float = -(pow(2, bits)/2 - 1);

    for(i = 0; i < SAMPLE_PER_CYCLE; i++) {
        sin_float = sin(i * PI / 180.0);
```
The last line shown here should rather be

```
sin_float = sin(i * 2 * PI / SAMPLE_PER_CYCLE);
```

to calculate one full cycle. For a full cycle the sin() function should get values from 0 to 2 PI. Since i runs from 0 to SAMPLE_PER_CYCLE, we need to use i * 2 * PI / SAMPLE_PER_CYCLE to get one full cycle.

With `sin_float = sin(i * PI / 180.0);` i is treated as an angel from 0 to 360 °, however, it runs up to more or less than 360 dependent on sample rate and desired wave frequency. For a sample rate of 44100 and a wave frequency of 100 it runs up to 441, for example. For the sample rate used by the example of 36000 and a wave frequency of 100 it does run to 360, but only for exactly these values.

Feel free to fix this, or leave it as is.

Alvin1Zhang, 2020-03-24:
@i-franzki Thanks for attention to the details, we will look into. Thanks.

Alvin1Zhang, 2020-03-30:
@i-franzki Thanks for reporting, we have a fix internally under reviewing. Hope it will soon be available on GitHub. Thanks.

closed 2020-03-31 by espressif-bot