espressif/esp-idf #4982: gattc 和gatts同时使用的一个事件问题

fzboffice, 2020-03-24:
当我的gattc和gatts都打开时，如果一个外部gatts被我的gattc打开时，会有一个connect事件，但是这个事件不仅触发了我的gattc的connect事件，还触发了我的gatts的connect事件
idf v4.0

WCCWCC, 2020-08-18:
Hi @fzboffice 

目前我们的实现是这样的，当调用 esp_ble_gattc_open  和对端设备建立连接后， 所有的 gatts 与 gattc 的回调函数都会收到 CONNECT_EVT.

fzboffice, 2020-08-19:
但是这样会混淆，能否做一个区分 gatts的open在gatts的回调出现 gattc的open在gattc的回调中出现

xiewenxiang, 2020-08-28:
We will add a `link_role` param in gatt connection event, this parameter can be used to determine the initiator of the connection

xiewenxiang, 2020-09-16:
@fzboffice Please update the IDF to the latest Mater branch, the link_role parameter has been added, you can use it to differentiate the origin of the connection.

fzboffice, 2020-09-17:
> @fzboffice请将IDF更新到最新的Mater分支，已添加link_role参数，您可以使用它来区分连接的来源。

感谢您的更新

closed 2020-09-17 by fzboffice