espressif/esp-idf #4983: Memory (heap) out after +100 TLS request with custom x509 root certificate

hassance3, 2020-03-24:
**What to do to produce the error:** 

1. Use ESP-IDF Example [https_mbedtls](https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_mbedtls)
2. Get root certificate of your server (self signed in my case) by using OpenSSL command: openssl s_client -showcerts -servername *.mydomain.com -connect MyServerIP:443 </dev/null 2>/dev/null|openssl x509 -outform PEM >/home/me/Desktop/server_root_cert.pem
3. In menuconfig add custom certificate to the custom bundle (/home/me/Desktop/server_root_cert.pem)
4. Build the firmware and flash..

It's will work great, but on every request will allocate around 2000 bytes on handshake stage, so after around 120 requests we get a out of memory warning.

Note: I get free heap size by using `ESP_LOGW(TAG, "Free Heap: %d", esp_get_free_heap_size());`

**Monitor log**

`--- idf_monitor on /dev/ttyUSB1 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7224
load:0x40078000,len:13092
load:0x40080400,len:3936
0x40080400: _init at ??:?
                                                                                                                             
entry 0x40080688
I (28) boot: ESP-IDF v4.2-dev-701-g0ae960f2f-dirty 2nd stage bootloader
I (29) boot: compile time 15:53:08
I (30) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2e1c4 (188868) map
I (185) esp_image: segment 1: paddr=0x0003e1ec vaddr=0x3ffb0000 size=0x01e2c (  7724) load
I (189) esp_image: segment 2: paddr=0x00040020 vaddr=0x400d0020 size=0x85194 (545172) map
0x400d0020: _stext at ??:?
                                                                                                                             
I (399) esp_image: segment 3: paddr=0x000c51bc vaddr=0x3ffb1e2c size=0x01b18 (  6936) load
I (402) esp_image: segment 4: paddr=0x000c6cdc vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/me/Desktop/MY_FIRMWARE/SDK/ESP_IDF/components/freertos/xtensa/xtensa_vectors.S:1817
                                                                                                                             
I (406) esp_image: segment 5: paddr=0x000c70e8 vaddr=0x40080404 size=0x14be4 ( 84964) load
I (463) boot: Loaded app from partition at offset 0x10000
I (463) boot: Disabling RNG early entropy source...
I (464) cpu_start: Pro cpu up.
I (467) cpu_start: Application information:
I (472) cpu_start: Project name:     https-mbedtls
I (478) cpu_start: App version:      1
I (482) cpu_start: Compile time:     Mar 23 2020 15:52:59
I (488) cpu_start: ELF file SHA256:  3e4cef9cc8901077...
I (494) cpu_start: ESP-IDF:          v4.2-dev-701-g0ae960f2f-dirty
I (501) cpu_start: Starting app cpu, entry point is 0x40081280
0x40081280: call_start_cpu1 at /home/me/Desktop/MY_FIRMWARE/SDK/ESP_IDF/components/esp32/cpu_start.c:273
                                                                                                                             
I (0) cpu_start: App cpu up.
I (511) heap_init: Initializing. RAM available for dynamic allocation:
I (518) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (524) heap_init: At 3FFB9530 len 00026AD0 (154 KiB): DRAM
I (531) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[... 215 lines cut ...]
```
W (2715) example: Free Heap: 206168
W (14855) example: Free Heap: 204212
W (26995) example: Free Heap: 202984
W (39195) example: Free Heap: 201760
W (51375) example: Free Heap: 200532
W (63595) example: Free Heap: 199308
```

The example work fine without custom certificate, this problem is only when adding custom x509 root certificate by menuconfig

```
#
# Certificate Bundle
#
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=y
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_FULL=y
# CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN is not set
# CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_NONE is not set
CONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE=y
CONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE_PATH="/home/me/Desktop/server_root_cert.pem"
# end of Certificate Bundle
```

Thank you,

hassance3, 2020-03-24:
After debugging line by line, I found that mbedtls_net_free() do nothing! 

**Request 1**
`mbedtls_net_connect()` --> Connect, and allocate memory.
`mbedtls_net_free()` --> Successful execution, but allocation in memory still exist.

**Request 2**
`mbedtls_net_connect()` --> Connect, and allocate memory in new area (of course).
`mbedtls_net_free()` --> Successful execution, but allocation in memory still exist!.

Can someone explain why?

igrr, 2020-03-24:
You might be seeing the TCP active close behavior? https://github.com/espressif/esp-idf/issues/2134#issuecomment-403336387

Please check if the heap returns back to normal after a few minutes.

hassance3, 2020-03-24:
That's it, the socket standard is keep it open for a while, I will test it by reducing the maximum opened socket and see if it's work.

Thank you Grokhotkov for your help.

hassance3, 2020-03-25:
Unfortunately didn't work, I set those settings : 

```
# LWIP
CONFIG_LWIP_MAX_SOCKETS=1
# TCP
CONFIG_LWIP_MAX_ACTIVE_TCP=1
CONFIG_LWIP_MAX_LISTENING_TCP=1
```

I sent some requests yesterday, and others this morning, it's seem heap memory keep shrinking!

hassance3, 2020-03-25:
Note: The only change I made in the default mbedTLS example is : 

- Disable Mozilla default trusted CA & Add my custom certificate
```
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=y
# CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_FULL is not set
# CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN is not set
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_NONE=y
CONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE=y
CONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE_PATH="/home/me/Desktop/my_root_cert.pem"
```

- Add sample Log to show available heap
```
...
if ((ret = mbedtls_ssl_setup(&ssl, &conf)) != 0)
{
   ESP_LOGE(TAG, "mbedtls_ssl_setup returned -0x%x\n\n", -ret);
   goto exit;
}
while(1){
   ESP_LOGE(TAG, "Heap: %d", esp_get_free_heap_size());
   ...
}

```

ESP-Marius, 2020-03-26:
@hassance3 Thanks for reporting this. You are right, there is a leak here. I'll push a merge request for it.

I'll attach a patch in case you want a fix asap.
[crt_mem_leak.txt](https://github.com/espressif/esp-idf/files/4384846/crt_mem_leak.txt)

hassance3, 2020-03-26:
@ESP-Marius I confirm that the **issue is fixed**, Thank you Marius.

Can you please give me the merge request ? I want keep it for reference.
Thank you,

hassance3, 2020-03-27:
@igrr Please can you give me the merge request ? this solve the memory leak issue.

igrr, 2020-03-27:
@hassance3 The merge request in review in our internal repository and will be merged after it passes the tests. This issue will be automatically closed when the fix lands on Github. This should happen in the next few work days.
In the meantime, please use the patch provided by @ESP-Marius locally as a fix, if possible.

hassance3, 2020-03-27:
I already use the patch provided by @ESP-Marius and it fix the issue, I would like just sign that the path it's worked for me.

Thank you for you support,
Sincerely.

X-Ryl669, 2020-03-27:
While you're reviewing it, could you also import the changes in mbedtls dev branch that prevents copying certificates in the heap (see [here](https://git.trustedfirmware.org/tls/mbed-tls.git/tree/library/x509_crt.c?h=development#n1366))

Certificate tends to be large, and idf is quite dumb with handling them. The automated build system should store certificate in DER format (and not PEM since it's 33% larger being base64 encoded) when making the asm sections. Also the mbedtls version used in esp-idf/component copies the given certificate in the heap because it "might need it later", which, as far as I know, is a useless copy when the certificate is in flash (it'll stay here).

If your certificate chain is large (mine is 24kB) that does make a big difference in the end (16kB instead of ~48kB used in heap, and 16kB instead of 24kB in flash).

Alvin1Zhang, 2020-04-09:
@X-Ryl669 Thanks for reporting and follow up. The fix in our internal repo. has been merged, the issue will be automatically closed when the fix lands on Github. For your information, thanks.

closed 2020-04-09 by espressif-bot