espressif/esp-idf #4984: Write main SPI memory (OTA) impacting UART, now generate fatal error in UART pattern det and invalid OTA

neorevx, 2020-03-24:
## Environment

- Development Kit: Custom
- Kit version: Custom
- Module or chip used:  ESP32-WROOM-32U
- IDF version: v4.2-dev-792-g6330b3345 (least today)
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: VS Code
- Power Supply: Custom 3.3v

## Problem Description + Debug Logs

The use of OTA has a negative impact on UART performance. Maybe it's just excessive CPU usage when writing SPI, or there may be a problem with silicon, I don't know.

**Context:**
In my project I use all UART peripherals: 0-console, 1-GPS module, 2-GSM modem (sim800L).
The GPS and the GSM modem use the hardware pattern det feature.
The system also has an OTA update, normally running when the GSM modem is connected in PPP mode (without pattern det).
When the update starts (OTA), the UART driver is unable to read all content received by the UART and the system generates "hw fifo full".

Okay, this is a problem, but of lesser impact. Until some previous versions of ESP-IDF (example 3.2), only few messages was displayed and the system recovered soon (when the OTA normalizes).
However, in version 4.2dev, the pattern det feature generates a critical failure to read an invalid memory position. Look:

```
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).

Core  0 register dump:
PC      : 0x40008150  PS      : 0x00060034  A0      : 0x8008758b  A1      : 0x3ffb1710
A2      : 0x3f420e08  A3      : 0x00000000  A4      : 0x00000004  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3ffb51f4  A9      : 0x3ffb1760
A10     : 0x000000ad  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x3ffcc92e
A14     : 0x00000a0d  A15     : 0x36372a38  SAR     : 0x00000000  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace:0x4000814d:0x3ffb1710 0x40087588:0x3ffb17a0 0x40087942:0x3ffb17c0 0x40082c61:0x3ffb1810 0x400855e8:0x3ffbf410 0x400824fd:0x3ffbf430 0x4009a995:0x3ffbf450
0x40087588: uart_pattern_enqueue at D:/OneDrive/ESP32-Master/components/driver/uart.c:347 (discriminator 1)

0x40087942: uart_rx_intr_handler_default at D:/OneDrive/ESP32-Master/components/driver/uart.c:866

0x40082c61: _xt_lowint1 at D:/OneDrive/ESP32-Master/components/freertos/xtensa/xtensa_vectors.S:1105

0x400855e8: spi_flash_op_block_func at D:/OneDrive/ESP32-Master/components/spi_flash/cache_utils.c:103 (discriminator 1)

0x400824fd: ipc_task at D:/OneDrive/ESP32-Master/components/esp_common/src/ipc.c:62

0x4009a995: vPortTaskWrapper at D:/OneDrive/ESP32-Master/components/freertos/xtensa/port.c:143

Core  1 register dump:
PC      : 0x4000c056  PS      : 0x00060134  A0      : 0x80008547  A1      : 0x3ffe32a0
A2      : 0x8ae742f9  A3      : 0x3ffe3270  A4      : 0x00000001  A5      : 0x3ffe32b0
A6      : 0x00000003  A7      : 0x00000000  A8      : 0x00000000  A9      : 0xfc000000
A10     : 0xb0000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3ffe3290
A14     : 0x00000002  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace:0x4000c053:0x3ffe32a0 0x40008544:0x3ffe32b0 0x40086562:0x3ffe32d0 0x4008a37e:0x3ffe32f0 0x4008a5d5:0x3ffe3330 0x40086207:0x3ffe3350 0x400f113b:0x3ffe3390 0x400f0a0b:0x3ffe33b0 0x400e9e44:0x3ffe33d0 0x400ea001:0x3ffe33f0 0x400e4741:0x3ffe3420 0x4009a995:0x3ffe34a0
0x40086562: delay_ms at D:/OneDrive/ESP32-Master/components/spi_flash/spi_flash_os_func_app.c:88

0x4008a37e: spi_flash_chip_generic_wait_idle at D:/OneDrive/ESP32-Master/components/spi_flash/spi_flash_chip_generic.c:273

0x4008a5d5: spi_flash_chip_generic_erase_block at D:/OneDrive/ESP32-Master/components/spi_flash/spi_flash_chip_generic.c:134

0x40086207: esp_flash_erase_region at D:/OneDrive/ESP32-Master/components/spi_flash/esp_flash_api.c:349

0x400f113b: esp_partition_erase_range at D:/OneDrive/ESP32-Master/components/spi_flash/partition.c:422

0x400f0a0b: esp_ota_begin at D:/OneDrive/ESP32-Master/components/app_update/esp_ota_ops.c:158

0x400e9e44: esp_https_ota_perform at D:/OneDrive/ESP32-Master/components/esp_https_ota/src/esp_https_ota.c:274

0x400ea001: esp_https_ota at D:/OneDrive/ESP32-Master/components/esp_https_ota/src/esp_https_ota.c:399

0x400e4741: do_firmware_update at d:\onedrive\esp32-projects\vcar\build/../main/ota/vupdate.c:31

0x4009a995: vPortTaskWrapper at D:/OneDrive/ESP32-Master/components/freertos/xtensa/port.c:143
```

It's may related to issue https://github.com/espressif/esp-idf/issues/4412. The error is log line:
https://github.com/espressif/esp-idf/blob/6330b3345e87eb4401e7be7c8b6fea2870c35d9f/components/driver/uart.c#L347

Since this log use ets_printf, it should not be an problem.

**For some even worse reason**, OTA is able to write a few bytes to the OTA partition and at the next boot (automatic restart after failure) the system starts on the "partial" OTA partition. And it doesn't start anymore.
Until then, in my tests, the OTA validation was correct. But now it seems that it is no longer, it starts a partition with invalid checksum. Failed to start:

```
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).

Core  0 register dump:
PC      : 0x4009df09  PS      : 0x00060d34  A0      : 0x800d46c8  A1      : 0x3ffe3bc0
0x4009df09: xt_ints_on at D:/OneDrive/ESP32-Master/components/freertos/xtensa/xtensa_intr_asm.S:164

A2      : 0x0300000c  A3      : 0x0100000c  A4      : 0x00000019  A5      : 0x00000000
A6      : 0x00000016  A7      : 0x00000000  A8      : 0x15122500  A9      : 0x00060d20
A10     : 0x3ffb5994  A11     : 0x00008000  A12     : 0x00000001  A13     : 0x00000001
A14     : 0x00000018  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace:0x4009df06:0x3ffe3bc0 0x400d46c5:0x3ffe3bd0 0x40081ab9:0x3ffe3bf0 0x40081dc5:0x3ffe3c40 0x4007922e:0x3ffe3c80 |<-CORRUPTED
0x4009df06: xt_ints_on at D:/OneDrive/ESP32-Master/components/freertos/xtensa/xtensa_intr_asm.S:163

0x400d46c5: esp_cache_err_int_init at D:/OneDrive/ESP32-Master/components/esp32/cache_err_int.c:70

0x40081ab9: start_cpu0_default at D:/OneDrive/ESP32-Master/components/esp32/cpu_start.c:390

0x40081dc5: call_start_cpu0 at D:/OneDrive/ESP32-Master/components/esp32/cpu_start.c:261
```
I'm not even sure if there is a valid binary when the OTA updater starts. That is, it could be that http sent error content in html format and was saved as a binary app.

**I believe the most important here are:**
- UART pattern det fails when the queue is full;
- Bootloader starting OTA with invalid checksum.

### Expected Behavior

- UART works normally when write SPI;
- UART don't fail do enqueue det pattern;
- Partial OTA can't be booted.

### Steps to reproduce

Use NMEA with OTA examples together. UART impact may occur without ppp.

Alvin1Zhang, 2020-03-25:
@neorevx Thanks for detailed reports, we will look into.

neorevx, 2020-03-25:
There are apparently no problems with the firmware in the SPI memory. After manual reboot (pin EN) it get working.
Checking the logs again, he started the program from the factory partition. Only now that I noticed that it is the same failure!

**(Cache disabled but cached memory region accessed)**

I don't understand how the system manages SPI memory when writing the OTA partition. Certainly there must be techniques to manage the reading of the code of the running program and the writing of the new binary.
There may be a problem with the system cache and reading the SPI during this step. This also explains why other tasks get slower.
System cache is disabled...
My UART interruption is in IRAM! The `uart_pattern_enqueue `function too! It may be related to IRAM.

More one thing: `0x400855e8: spi_flash_op_block_func at D:/OneDrive/ESP32-Master/components/spi_flash/cache_utils.c:103 (discriminator 1)
`
The stack for UART related interrupt come from spi_flash component.

neorevx, 2020-03-25:
Another thing not related direct to this issue: pattern det don't work if you compile optimized for performance. Maybe are missing some "volatile" keyword for some vars in UART driver, I don't know.

neorevx, 2020-03-26:
I read some docs over the internet and concluded:
- Cache is disabled for SPI write;
- When cache is disabled and IRAM intr is triggered, only IRAM code can be executed. I think ets_printf is not in IRAM, then call this function generates an exception. ets_printf should be put in IRAM or logs should be removed from interrupt. Or remove intr from IRAM;
- When I disabled "place SPI ... in IRAM" I still getting this exception;
- When I disabled "place UART ... in IRAM" (but SPI in IRAM), this exception does not happens anymore. I still getting some "fifo full", but it's no problem at moment.

koobest, 2020-03-27:
Flash writing operation will disable the cache,  the code and data in cache Inaccessible. So, `ESP_EARLY_LOGW(UART_TAG, "Fail to enqueue pattern position, pattern queue is full."); ` may cause a crash. We will think of a solution to this issue.
thanks!!.

zheyuejiang, 2021-04-11:
any update here?

negativekelvin, 2021-04-11:
Here's a hint https://github.com/espressif/esp-idf/commit/a0e66fef0880dba2caf93e9cb9c65cff7dc9b880

DBMTC, 2022-06-15:
I might have hit this or a similar error as my stacktrace is:
`
E (117340) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (117340) task_wdt:  - IDLE (CPU 0)
E (117340) task_wdt: Tasks currently running:
E (117340) task_wdt: CPU 0: ipc0
E (117340) task_wdt: CPU 1: IDLE
E (117340) task_wdt: Aborting.

abort() was called at PC 0x40129cbc on core 0
0x40129cbc: task_wdt_isr at /home/dbmtc/esp/esp-idf/components/esp_common/src/task_wdt.c:182 (discriminator 1)

Backtrace:0x40081882:0x3ffb0980 0x400892b9:0x3ffb09a0 0x4008f8f2:0x3ffb09c0 0x40129cbc:0x3ffb0a30 0x400829c5:0x3ffb0a50 0x40084768:0x3ffaf8a0 0x400834f6:0x3ffaf8c0 0x40083448:0x3ffaf8e0
0x40081882: panic_abort at /home/dbmtc/esp/esp-idf/components/esp_system/panic.c:404

0x400892b9: esp_system_abort at /home/dbmtc/esp/esp-idf/components/esp_system/system_api.c:112

0x4008f8f2: abort at /home/dbmtc/esp/esp-idf/components/newlib/abort.c:46

0x40129cbc: task_wdt_isr at /home/dbmtc/esp/esp-idf/components/esp_common/src/task_wdt.c:182 (discriminator 1)

0x400829c5: _xt_lowint1 at /home/dbmtc/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x40084768: xt_int_enable_mask at /home/dbmtc/esp/esp-idf/components/xtensa/include/xtensa/xtensa_api.h:170
 (inlined by) intr_cntrl_ll_enable_int_mask at /home/dbmtc/esp/esp-idf/components/hal/esp32/include/hal/interrupt_controller_ll.h:100
 (inlined by) interrupt_controller_hal_enable_int_mask at /home/dbmtc/esp/esp-idf/components/hal/include/hal/interrupt_controller_hal.h:192
 (inlined by) esp_intr_noniram_enable at /home/dbmtc/esp/esp-idf/components/esp_system/intr_alloc.c:815

0x400834f6: spi_flash_op_block_func at /home/dbmtc/esp/esp-idf/components/spi_flash/cache_utils.c:124

0x40083448: ipc_task at /home/dbmtc/esp/esp-idf/components/esp_ipc/ipc.c:74
`

my wild guess: as the task ipc0 is very high priority it might take a long time to execute without ever resetting the watchdog as in cache_utils.c this code can be spot:

`
        while (!s_flash_op_can_start) {
            // Busy loop and wait for spi_flash_op_block_func to disable cache
            // on the other CPU
        }
`

indeed if this takes too much time that crash is very reasonable.

if a poll on a bool  is the right thing to do (I don't even think a bool write is atomic enough) I suggest to either reset the watchdog or add a delay (hoping IDLE tak will call wdt_reset in time).

I am also using UART while updating, but I think the relation between those two things is a secondary side effect of a busy wait.