espressif/esp-idf #4986: MQTT 异常断开errno=Software caused connection abort

shenyong-iot01, 2020-03-25:
您好，IDF3.2.2SDK  MQTT直接使用IDF3.3.1内版本，经挂机测试12小时左右设备MQTT异常断开多次，请帮忙看下原因，谢谢！断开连接部分log如下   
[19:58:18.539] NTP send: ip = 119.28.183.184, port = 123
[19:58:18.539] Date & Time (CST): 2020/3/24-2-19:58:13
[19:58:18.570] Date & Time (UTC): 2020/3/24-2-11:58:13
[19:58:18.570] E (17191377) esp-tls: read error :-76:
[20:43:51.137] E (17191377) TRANS_SSL: esp_tls_conn_read error, errno=Software caused connection abort
[20:43:51.137] E (17191377) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=113
[20:43:51.137] E (17191387) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1
[20:43:51.137] [APP MQTT] MQTT_EVENT_DISCONNECTED
[20:43:51.137] I (17203587) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
[20:44:03.328] [APP MQTT] MQTT_EVENT_CONNECTED

david-cermak, 2020-03-25:
Hi @shenyong-iot01 

This means the connection was closed from the server side. Perhaps you can get some information what has happened from the server side log?

shenyong-iot01, 2020-03-26:
@david-cermak 谢谢您的回复，我将和服务器端沟通。另外这两个log打印相差将近1小时，系统时间戳却是一样的17191377，是否函数内阻塞...
[19:58:18.570] E (17191377) esp-tls: read error :-76:
[20:43:51.137] E (17191377) TRANS_SSL: esp_tls_conn_read error, errno=Software caused connection abort

david-cermak, 2020-04-01:
@shenyong-iot01 
Can you please share the exact version of the SDK you use? e.g. running 
```
git rev-parse HEAD
```

There was such an issue, that the socket read/write blocked for a long time if error comes. This has been fixed in https://github.com/espressif/esp-idf/commit/11995d808c7963e1692047e44af3f09c83d2c121
So if you're on 3.2.2 (https://github.com/espressif/esp-idf/commit/055943e29346e77c50589c61a8a26101a8b35d7b), seems like the fix is not included. Can you please check if the issue goes away if you switch to the 3.2 release branch or cherry-picking that commit?

shenyong-iot01, 2020-04-01:
之前使用ESP32 SDK version: v3.2.2-5-gf90a429-dirty，MQTT 使用IDF3.3.1内版本。我后续将sdk更新到ESP32 SDK version: v3.3.1-dirty后再复测这个问题，谢谢！

shenyong-iot01, 2020-04-11:
ESP32 SDK version: v3.3.1-dirty 内修复相应问题，我这边按照11995d8修复方法添加进去，目前未复测到长时间阻塞，这个问题暂时可以关闭，谢谢！

closed 2020-04-11 by shenyong-iot01