espressif/esp-idf #4988: Change DHCPD netmask does not work

AxelLin, 2020-03-25:
Module or chip used: ESP32-WROOM-32
IDF version: v4.2-dev-792-g6330b3345e87
Build System: CMake
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Operating System: Linux
Power Supply: USB

In components/esp_netif/esp_netif_defaults.c,
I change the _g_esp_netif_soft_ap_ip settings as below:
const esp_netif_ip_info_t _g_esp_netif_soft_ap_ip = {
        .ip = { .addr = IP4TOADDR( 10, 33, 1, 1) },
        .gw = { .addr = IP4TOADDR( 10, 33, 1, 1) },
        .netmask = { .addr = IP4TOADDR( 255, 255, 0, 0) },
};

Then I build and run wifi/getting_started/softAP example code.

The device boots fine, my laptop successfully join the softAP and got IP as expected.

I (1449) wifi softAP: wifi_init_softap finished. SSID:myssid password:mypassword
I (31499) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (31499) wifi:station: c8:ff:28:04:f8:09 join, AID=1, bgn, 40U
I (31519) wifi softAP: station c8:ff:28:04:f8:09 join, AID=1
I (31549) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.33.1.2

However, when I check ifconfig on my laptop, I found the netmask is *wrong*.
I expect 255.255.255.0, but it got 255.255.255.0.

wlp3s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.33.1.2  netmask 255.255.255.0  broadcast 10.33.1.255
        inet6 fe80::6b03:c65:130a:436a  prefixlen 64  scopeid 0x20<link>
        ether c8:ff:28:04:f8:09  txqueuelen 1000  (Ethernet)
        RX packets 16888104  bytes 13955856917 (13.9 GB)
        RX errors 0  dropped 83  overruns 0  frame 0
        TX packets 10424612  bytes 1892064823 (1.8 GB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

AxelLin, 2020-03-25:
I just checked the implementation and found:
The dhcps_start() only takes ip argument, so changing netmask does not work.
I think this needs fix.

AxelLin, 2020-04-06:
I expect below code should be able to change the dhcps ip and netmask.
But it only changes the ip, as a result the netmask and broadcast address are wrong.

esp_netif_ip_info_t ip_info;
esp_netif_set_ip4_addr(&ip_info.ip, 10, 33, 1, 1);
esp_netif_set_ip4_addr(&ip_info.gw, 10, 33, 1, 1);
esp_netif_set_ip4_addr(&ip_info.netmask, 255, 255 , 0, 0);

esp_netif_dhcps_stop(netif);
esp_netif_set_ip_info(netif, &ip_info);
esp_netif_dhcps_start(netif);

Alvin1Zhang, 2020-04-07:
@AxelLin Thanks for reporting and follow-up, we have a fix which is now under internal reviewing, will update once it is available on GitHub. Thanks

AxelLin, 2020-04-29:
@Alvin1Zhang 
The fix is still not available after waiting for 3 weeks, I'm wondering if you can post the fix so I can test it.

Alvin1Zhang, 2020-04-29:
@AxelLin Thanks for your patience waiting for the fix, @david-cermak please help. Thanks.

david-cermak, 2020-04-29:
@AxelLin 
here's the patch 
[0001-dhcpserver-make-subnet-mask-configurable.patch.txt](https://github.com/espressif/esp-idf/files/4550252/0001-dhcpserver-make-subnet-mask-configurable.patch.txt)

Thanks in advance for testing it and sorry for such a slow turnaround.

AxelLin, 2020-04-29:
Hi @david-cermak 
I just tested the patch and the result looks good.

Thank you.

closed 2020-05-05 by espressif-bot