espressif/esp-idf #4989: Unable to restart Advertising after NimBLE reset event occurred.

tmsalidor, 2020-03-25:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version: v3.3.1
- Build System:         [Make]
- Compiler version: 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

We are using NimBLE stack with esp-idf version 3.3.1. It runs BLE server and is connected from iOS or Android application. It starts advertising after power-up, stops when connected and restarts when disconnected. It works normally well but we noticed an issue. When NimBle stack signals reset event during Advertising, it's unable to restart adverting after it anyway.

### Expected Behavior
The handling of reset event is noted in link below.
http://mynewt.apache.org/v1_7_0/network/ble_setup/ble_sync_cb.html
When Nimble detects something wrong, it signals reset event callback and reset controller.
Nimble stack re-syncs with controller and signals sync event callback.
Our code can start advertising after reception of sync event.

### Actual Behavior
Nimble signals reset event callback and then signals sync event callback after a while.
Our code requests to start advertising, but it gets error response 2 (BLE_HS_EALREADY). It looks already advertising working, but not actually.

### Steps to reproduce

1. Power up and initialize NimBLE stack.
2. Our code receives sync event from NimBLE then starts advertising by ble_gap_adv_start() API.
3. Connected by iOS or Android application. Do some processes.
4. Our code calls esp_ota_begin() API, and it sometimes causes erroneous disconnection of BLE (known behavior in issue https://github.com/espressif/esp-idf/issues/3551). Our code receives GAP disconnection event therefore restart advertising by ble_gap_adv_start() API.
5. Sometimes NimBLE reset event appears. Sync event also appears after a while then our code starts advertising by ble_gap_adv_start() API, but it fails.

### Code to reproduce this issue
Almost derived from esp-idf/components/examples/bluetooth/nimble/bleprph/main/main.c.

<details>
<summary>Code</summary>

```
void nm_ble_on_sync(void)
{
    int rc;

    /* Figure out address to use while advertising (no privacy for now) */
    rc = ble_hs_id_infer_auto(0, &s_own_addr_type);
    if (rc != 0) {
        ESP_LOGE(TAG, "error determining address type; rc=%d", rc);
        assert(false);
        return;
    }

    /* Printing ADDR */
    uint8_t addr_val[6] = {0};
    rc = ble_hs_id_copy_addr(s_own_addr_type, addr_val, NULL);
    if (rc != 0) {
        ESP_LOGE(TAG, "error determining address type; rc=%d", rc);
        assert(false);
        return;
    }

    ESP_LOGI(TAG, "BLE mac address: %02X:%02X:%02X:%02X:%02X:%02X",
    addr_val[0], addr_val[1], addr_val[2], addr_val[3], addr_val[4], addr_val[5]);

    rc = bleprph_advertise();
    if (rc) {
        ESP_LOGE(TAG, "bleprph_advertise fail: %d", rc);

    }
[... 261 lines cut ...]
Command complete: cmd_pkts=5 ogf=0x8 ocf=0x3 status=0 
ble_hs_hci_cmd_send: ogf=0x04 ocf=0x0009 len=0
0x09 0x10 0x00 
Command complete: cmd_pkts=5 ogf=0x4 ocf=0x9 status=0 bd_addr=c8:2b:96:81:fa:86
looking up peer sec; 
I (54987) NIM_COMMON: nim_on_sync:43
I (54997) nm-bt: BLE mac address: 86:FA:81:96:2B:C8
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0007 len=0
0x07 0x20 0x00 
Command complete: cmd_pkts=5 ogf=0x8 ocf=0x7 status=0 
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0008 len=32
0x08 0x20 0x20 0x17 0x02 0x01 0x06 0x03 0x03 0xff 0xff 0x0c 0x09 0x52 0x65 0x6d 0x6f 0x2d 0x38 0x31 0x46 0x41 0x38 0x34 0x02 0x0a 0x03 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
Command complete: cmd_pkts=5 ogf=0x8 ocf=0x8 status=0 
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0006 len=15
0x06 0x20 0x0f 0x30 0x00 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x07 0x00 
Command complete: cmd_pkts=5 ogf=0x8 ocf=0x6 status=0 
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000a len=1
0x0a 0x20 0x01 0x01 
Command complete: cmd_pkts=5 ogf=0x8 ocf=0xa status=0 
```

</details>

With above results, we guess current esp-idf (or NimBLE stack) is missing some necessary clean-up process in NimBLE host layer at NimBLE controller reset event.

dhrishi, 2020-03-26:
@tmsalidor Thanks for reporting the issue. We are already working on making NimBLE stack behave correctly after reset and have raised an upstream request, see here: https://github.com/apache/mynewt-nimble/pull/777

Once that is reviewed and integrated, we will include the same in esp-nimble as well.

tmsalidor, 2020-03-27:
@dhrishi Thanks for your comment. I got the cause and fix is already in progress.

prasad-alatkar, 2020-05-21:
Hi @tmsalidor Can you please confirm if the issue is resolved with latest `release/v3.3` ?  Reference commit on IDF `release/v3.3`: aad04405.

tmsalidor, 2020-05-30:
Hello @prasad-alatkar, I checked the latest commit of release/v3.3 and confirmed it's resolved. Thanks a lot.

dhrishi, 2020-05-30:
@tmsalidor Great! Can you please close this issue?

closed 2020-05-30 by tmsalidor