espressif/esp-idf #4991: Remove PM_USE_RTC_TIMER_REF from kconfig

JosuGZ, 2020-03-25:
In `make menuconfig` we have an option to `Use RTC timer to prevent time drift (EXPERIMENTAL)`, but it seems to be [unused](https://github.com/espressif/esp-idf/search?q=PM_USE_RTC_TIMER_REF&unscoped_q=PM_USE_RTC_TIMER_REF).

I think It should probably be removed since it is misleading and does nothing.

VedantParanjape, 2020-03-31:
I would like to contribute to this.

igrr, 2020-03-31:
Thanks for pointing this out. Looks like this is a regression in IDF 4.0, the option was renamed in Kconfig but not in the code. I think we'll probably prefer to fix the option name in code, rather than removing the option...

JosuGZ, 2020-03-31:
> Thanks for pointing this out. Looks like this is a regression in IDF 4.0, the option was renamed in Kconfig but not in the code. I think we'll probably prefer to fix the option name in code, rather [...]

Keep in mind that some people may have this enabled so changing the name might change the behavior of their code. I don't know what this exactly does though.

closed 2020-09-16 by espressif-bot