espressif/esp-idf #4993: Compiling example in its default location - unable to remove folder.

LocoMatt, 2020-03-25:
I'm noticing same problem another time with W10. I compiled example code (examples\wifi\getting_started\station) code through mingw. Then I decided to update my esp-idf. Classic way, remove old esp-idf folder and just download using git. For some reason there is folder in such path
`examples\wifi\getting_started\station\build\include\config\smp\slave\con`
which cannot be removed by just delete. When I try to acces this folder I get error 

[image]

Translation is something like:
```
.....\s...\con
is not available.
Invaild acces.
```

As I know con is exclusive folder name in Windows, I'm bit unsure why it hapens, maybe it isn't typical bug, but maybe solution below will help someone in future.

Previously I had similar issue with esp-adf 
Only way to remove it is to use in cmd
`rd /s \\?\C:\folder_path `

dobairoland, 2020-03-26:
Hi @LocoMatt. Can you please tell us your IDF version (the output of `git describe --tags`)? Because I think we've already fixed this issue: https://github.com/espressif/esp-idf/issues/3299

LocoMatt, 2020-03-27:
I tried backuping this folder, but git says it's not a git repo. Probbably I did it manually from github? I can see that it was created early January this year. I guess it was 3.3 or 4.0 because it still used tcpip instead of esp-netif. Is there any other way to determine version by reffering to available files?

A bit off-topic, upgrading toolchain requirements for 4.2 still work with earlier verisons of IDF?

dobairoland, 2020-03-27:
>  Is there any other way to determine version by reffering to available files?

Version is not enough in this case.

You can take a look at `tools/kconfig/confdata.c` and see if you have a `conf_split_config` function there which was removed in https://github.com/espressif/esp-idf/commit/6cd6423092d1f6197fb2a438e2b6079365085401. 

> A bit off-topic, upgrading toolchain requirements for 4.2 still work with earlier verisons of IDF?

The same toolchain should work for all v4.x releases. You need an older one for v3.x.

LocoMatt, 2020-03-27:
[image]
Ain't there

dobairoland, 2020-03-27:
The folder could have been created by an earlier (unpatched) version of IDF.

If the function is not there anymore then it won't create those "con" directories.

dobairoland, 2020-04-02:
I checked out the latest v3.3 and v4.0 branches and did a build of the same example. No "con" dictionaries were created.

I close this issue under assumption that those directories were created by an IDF version which doesn't contain the fix.

closed 2020-04-02 by dobairoland