espressif/esp-idf #4994: UART echo example prints to serial monitor

weilbith, 2020-03-25:
## Environment

- Development Kit:      SBC-NodeMCU ESP32
- Module or chip used:  ESP32-WROOM-32
- IDF version:          v4.2-dev-630-gf7d1ed5f3
- Build System:         Make
- Compiler version      xtensa-esp32s2-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Using an IDE?:        No
- Power Supply:         USB

## Problem Description

I copied the [uart echo
example](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/uart/uart_echo/main/uart_echo_example_main.c), built it and flashed it to my controller. I only adjusted the TX and RX pins to `1` and `3` (according [to this](https://components101.com/sites/default/files/component_pin/ESP32-Pinout.png) pinout sheet). I connected a second microcontroller via cables to those pins. This controller sends in a loop the character `s` and then tries to read a response.

### Expected Behavior

The output I get via serial connection (`make monitor`) to the esp32 board
should be silent. Nothing gets printed. The other board should get responses of
the esp32.

### Actual Behavior

The esp32 sucessfully reads the `s` characters. But it prints them to the serial
USB connection (which `make monitor` connects to). But there is no response to
the other controller.

### Steps to reproduce

1. Download the above linked example project.
2. Connect seconds device via cable to the pins 1 and 3
3. Connect the board via USB to the PC
4. Build and flash the app to the board (`make flash`)
5. Watch serial output of controller (`make monitor`)

### Code to reproduce this issue

Checkout the example code
[here](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/uart/uart_echo/main/uart_echo_example_main.c)

## Debug Logs

No crash

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
- [x] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)

[Uploading sdkconfig_elf.zip…]()

negativekelvin, 2020-03-25:
Pins 1&3 are the default uart0 pins that are used for the serial monitor. Use different pins.

weilbith, 2020-03-26:
Good point. I wasn't aware about that and confused that this mix works. Thanks
for your help!

closed 2020-03-26 by weilbith