espressif/esp-idf #4995: Translate persisted peer lists from Bluedroid to Nimble.

zacwbond, 2020-03-25:
**Is your feature request related to a problem? Please describe.**

We're updating code on existing devices to use Nimble instead of Bluedroid.  It would be handy if, the first time the Nimble version boots, it could read the nonvolatile storage containing all of the bonded peers, then copy and re-save that information in the format desired by Nimble.

The idea would be to not lose all memory of previously-bonded devices when you do a firmware update transitioning from Bluedroid to Nimble.

I could probably work on this on my own, but there's a maze of dependencies in both stacks' NVS support and I'm just not sure where to begin.

dhrishi, 2020-04-01:
@zacwbond I don't really think we could consider this as a feature and implement it. Although, I can ask @GYC-Espressif to update about the Bluedroid parameters that are written in NVS for bonding and then @prasad-alatkar for the corresponding NimBLE parameters in NVS

zacwbond, 2020-04-13:
So I created a partial implementation of this.  I basically ported `btc_storage_get_bonded_ble_devices_list()` and all its dependencies into Nimble.  Now, from inside the Nimble stack, I can read and display all the data about remotes that were bonded to Bluedroid.  

My problem is that I don't know exactly how to translate the data into the Nimble data structures.  

Below is the code I created, but it does not quite work.  I am calling it in `ble_store_config_conf_init()`, just before `ble_nvs_restore_sec_keys()`. 

The code is below, but the algorithm is basically as follows:

1. Call `btc_storage_get_bonded_ble_devices_list()` to get the list of devices and security info.
2. For each entry in the list...
   1. Copy the peer device address.  
   2. Copy the LTK data, including rand_num, ediv and key_size
   3. Ignore the IRK since we aren't doing authentication and it's always zero. 
   4. Ignore the CSRK (it's always zero in the Bluedroid data)
   5. Call `ble_store_write_peer_sec(&sec)` to save the security data.

I see there is a `ble_store_write_our_sec()` but I don't know what that's supposed to do.  

Looking at `btc_storage_get_bonded_ble_devices_list()`, it reads `BTM_LE_KEY_PENC`, `BTM_LE_KEY_PCSRK`, and `BTM_LE_KEY_PID`.  

But there's also `BTM_LE_KEY_PLK`, `BTM_LE_KEY_LLK`, `BTM_LE_KEY_LENC`, `BTM_LE_KEY_LID` and `BTM_LE_KEY_LCSRK`.  I don't know what any of those are for.  

I also have seen that the peer address stored in the Bluedroid NVRAM is byte-reversed from the peer address stored in the Nimble NVRAM for the same device.  Is that true of all the data?

```
static int bluedroid_nvs_to_nimble_nvs(void)
{
    static const char *FILE_NAME = "bt_config.conf";
    config                       = config_new(FILE_NAME);
    if (!config)
    {
        config_free(config);
        return 0;
    }

    static esp_ble_bond_dev_t devices[10];
    int device_count = btc_storage_get_num_ble_bond_devices();
    btc_storage_get_bonded_ble_devices_list(devices, device_count);

    for (int i = 0; i < device_count; i++)
    {
        static struct ble_store_value_sec sec;

        memset(&sec, 0, sizeof(sec));

        sec.peer_addr.type = BLE_ADDR_PUBLIC;
        memcpy(sec.peer_addr.val,
               devices[i].bond_key.pid_key.static_addr,
               sizeof(sec.peer_addr.val));

        // devices[i].bond_key.pcsrk_key.csrk is all 0, so we don't have one (?)
        sec.csrk_present = 0;
        // memcpy(sec.csrk, devices[i].bond_key.pcsrk_key.csrk, sizeof(sec.csrk));

        sec.ltk_present = 1;
        sec.ediv        = devices[i].bond_key.penc_key.ediv;
        memcpy(sec.ltk, devices[i].bond_key.penc_key.ltk, sizeof(sec.ltk));
        sec.key_size = devices[i].bond_key.penc_key.key_size;
        memcpy(&sec.rand_num,
               devices[i].bond_key.penc_key.rand,
               sizeof(sec.rand_num));

        // devices[i].bond_key.pid_key.irk was all 0, so we don;t have one (?)
        sec.irk_present = 0;
        // memcpy(sec.irk, devices[i].bond_key.pid_key.irk, sizeof(sec.irk));

        // I don't know how to derive these values from the Bluedroid data, but I know 
        // our devices use encryption without authentication.
        sec.sc            = 1;
        sec.authenticated = 0;

        ble_store_write_peer_sec(&sec);
        // Do I need to use ble_store_write_our_sec() somewhere?
    }

    config_free(config);

    // Wipe the old config.
    config_t *config = config_new_empty();
    if (!config)
    {
        return 1;
    }

    config_save(config, FILE_NAME);

    config_free(config);
    return 0;
}
```