espressif/esp-idf #4997: Android 端最新版本出现无法配网的bug！

tangchuanyong, 2020-03-26:
Android代码和设备端都使用station模式！
Android端发送的数据：
gattWrite= [8, 8, 0, 1, 1]
gattWrite= [9, 0, 1, 10, 90, 101, 98, 114, 97, 95, 104, 111, 109, 101]
gattWrite= [13, 0, 2, 11, 90, 101, 98, 114, 97, 49, 50, 51, 52, 53, 54]
gattWrite= [12, 0, 3, 0]
Android接受的数据：
RM release without a matched acquire!
process_indicate, ignore HID ind/notificiation
E/response ==: [61, 4, 1, 23, 0, 0, 0, 1, 6, -68, -35, -62, -43, 78, 124, 2, 10, 90, 101, 98, 114, 97, 95, 104, 111, 109, 101]

最终结果：
Receive device status response:
op mode:NULL

能帮忙看下哪里出问题了吗？

tangchuanyong, 2020-03-26:
接收的数据：
E/response ==: [0, 4, 0, 1, 0]
E/response ==: [61, 4, 1, 23, 0, 0, 0, 1, 6, -68, -35, -62, -43, 78, 124, 2, 10, 90, 101, 98, 114, 97, 95, 104, 111, 109, 101]

tangchuanyong, 2020-03-26:
我在android的https://github.com/EspressifApp/EspBlufiForAndroid/issues/18  他说发送的数据没有问题！

tangchuanyong, 2020-03-26:
2020-03-26 08:43:24.067 802-1186/com.espressif.espblufi W/BlufiClientImpl: gattWrite= [8, 8, 0, 1, 1]
2020-03-26 08:43:24.093 802-952/com.espressif.espblufi W/BlufiClientImpl: response ==: [0, 4, 0, 1, 0]
2020-03-26 08:43:24.093 802-1186/com.espressif.espblufi W/BlufiClientImpl: gattWrite= [9, 0, 1, 10, 90, 101, 98, 114, 97, 95, 104, 111, 109, 101]
2020-03-26 08:43:24.132 802-1186/com.espressif.espblufi W/BlufiClientImpl: gattWrite= [13, 0, 2, 11, 90, 101, 98, 114, 97, 49, 50, 51, 52, 53, 54]
2020-03-26 08:43:24.178 802-1186/com.espressif.espblufi W/BlufiClientImpl: gattWrite= [12, 0, 3, 0]
2020-03-26 08:43:26.730 802-952/com.espressif.espblufi W/BlufiClientImpl: response ==: [61, 4, 1, 23, 0, 0, 0, 1, 6, -68, -35, -62, -43, 78, 124, 2, 10, 90, 101, 98, 114, 97, 95, 104, 111, 109, 101]

tangchuanyong, 2020-03-26:
2020-03-26 08:54:57.635 1491-5786/? D/ActivityTrigger: activityStartTrigger: Activity is Triggerred in full screen ApplicationInfo{4fbbb16 com.espressif.espblufi}
2020-03-26 08:54:57.635 1491-5786/? E/ActivityTrigger: activityStartTrigger: not whiteListedcom.espressif.espblufi/com.espressif.espblufi.ui.ConfigureOptionsActivity/24
2020-03-26 08:54:57.636 1491-5786/? D/ActivityTrigger: activityResumeTrigger: The activity in ApplicationInfo{4fbbb16 com.espressif.espblufi} is now in focus and seems to be in full-screen mode
2020-03-26 08:54:57.636 1491-5786/? E/ActivityTrigger: activityResumeTrigger: not whiteListedcom.espressif.espblufi/com.espressif.espblufi.ui.ConfigureOptionsActivity/24
2020-03-26 08:54:57.653 1491-5786/? D/ActivityTrigger: activityResumeTrigger: The activity in ApplicationInfo{4fbbb16 com.espressif.espblufi} is now in focus and seems to be in full-screen mode
2020-03-26 08:54:57.653 1491-5786/? E/ActivityTrigger: activityResumeTrigger: not whiteListedcom.espressif.espblufi/com.espressif.espblufi.ui.ConfigureOptionsActivity/24
2020-03-26 08:54:57.655 1491-5786/? D/RMS-PreloadedAppRecordMgr: setNeedKeepQuiet pkgName=com.espressif.espblufi,userId=0uid=0,keepQuiet=false
2020-03-26 08:54:57.667 6290-6290/com.espressif.espblufi W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@501c851
2020-03-26 08:54:57.701 6290-6290/com.espressif.espblufi W/ressif.espbluf: Accessing hidden method Landroid/widget/PopupWindow;->setEpicenterBounds(Landroid/graphics/Rect;)V (light greylist, linking)
2020-03-26 08:54:57.702 6290-6290/com.espressif.espblufi W/ressif.espbluf: Accessing hidden method Landroid/widget/PopupWindow;->setClipToScreenEnabled(Z)V (light greylist, reflection)
2020-03-26 08:54:57.751 1491-5785/? D/TansactManager: addProxyBpBinder BpTransactProxy{descriptor=IContentObserver pkg=com.espressif.espblufi proc=com.espressif.espblufi uid=10490 pid=6290 proxy=false binder=android.os.BinderProxy@40ac56e}
2020-03-26 08:54:57.777 6290-6290/com.espressif.espblufi I/TextInputLayout: EditText added is not a TextInputEditText. Please switch to using that class instead.
2020-03-26 08:54:57.795 6290-6290/com.espressif.espblufi I/TextInputLayout: EditText added is not a TextInputEditText. Please switch to using that class instead.
2020-03-26 08:54:57.798 6290-6295/com.espressif.espblufi I/ressif.espbluf: Compiler allocated 4MB to compile void android.widget.TextView.<init>(android.content.Context, android.util.AttributeSet, int, int)
2020-03-26 08:54:57.833 6290-6290/com.espressif.espblufi W/ressif.espbluf: Accessing hidden method Landroid/net/wifi/WifiInfo;->getWifiSsid()Landroid/net/wifi/WifiSsid; (light greylist, reflection)
2020-03-26 08:54:57.833 6290-6290/com.espressif.espblufi W/ressif.espbluf: Accessing hidden method Landroid/net/wifi/WifiSsid;->getOctets()[B (light greylist, reflection)
2020-03-26 08:54:57.868 1491-5785/? D/TansactManager: addProxyBpBinder BpTransactProxy{descriptor=IWindow pkg=com.espressif.espblufi proc=com.espressif.espblufi uid=10490 pid=6290 proxy=false binder=android.os.BinderProxy@1d1c40f}
2020-03-26 08:54:57.883 6290-6290/com.espressif.espblufi I/TouchMotionHandler: handler onAttached
2020-03-26 08:54:57.961 706-706/? E/ANDR-IOP: IOP HAL: Received pkg_name = com.espressif.espblufi pid = 0
2020-03-26 08:54:57.961 1491-1634/? I/ActivityManager: Displayed com.espressif.espblufi/.ui.ConfigureOptionsActivity: +301ms
2020-03-26 08:54:57.970 706-706/? E/ANDR-IOP: IOP HAL: Received pkg_name = com.espressif.espblufi pid = 0
2020-03-26 08:54:58.011 6290-6321/com.espressif.espblufi D/OpenGLRenderer: endAllActiveAnimators on 0x7251556600 (RippleDrawable) with handle 0x7251458c00
2020-03-26 08:54:59.101 6290-6290/com.espressif.espblufi E/BackgroundFallback: initVivoBgFallback callstackcom.android.internal.policy.DecorView.<init>:388 com.android.internal.policy.PhoneWindow.generateDecor:2441 com.android.internal.policy.PhoneWindow.installDecor:2820 com.android.internal.policy.PhoneWindow.getDecorView:2185 androidx.appcompat.app.AppCompatDelegateImpl.createSubDecor:717 androidx.appcompat.app.AppCompatDelegateImpl.ensureSubDecor:659 androidx.appcompat.app.AppCompatDelegateImpl.setContentView:552 androidx.appcompat.app.AppCompatDialog.setContentView:95 androidx.appcompat.app.AlertController.installContent:232 androidx.appcompat.app.AlertDialog.onCreate:279 
2020-03-26 08:54:59.142 1491-5785/? D/TansactManager: addProxyBpBinder BpTransactProxy{descriptor=IWindow pkg=com.espressif.espblufi proc=com.espressif.espblufi uid=10490 pid=6290 proxy=false binder=android.os.BinderProxy@90104cc}
2020-03-26 08:54:59.151 6290-6290/com.espressif.espblufi W/[ConfigureOptionsActivity]: Generate configure params null
2020-03-26 08:54:59.165 6290-6290/com.espressif.espblufi I/TouchMotionHandler: handler onAttached
2020-03-26 08:54:59.185 751-751/? D/SurfaceFlinger: duplicate layer name: changing com.espressif.espblufi/com.espressif.espblufi.ui.ConfigureOptionsActivity to com.espressif.espblufi/com.espressif.espblufi.ui.ConfigureOptionsActivity#1
2020-03-26 08:55:00.338 6290-6290/com.espressif.espblufi I/TouchMotionHandler: handler onDetached
2020-03-26 08:55:00.341 6290-6321/com.espressif.espblufi D/OpenGLRenderer: endAllActiveAnimators on 0x724f4a9e00 (RippleDrawable) with handle 0x724fe624e0
[... cut ...]
2020-03-26 08:55:03.283 6290-6344/com.espressif.espblufi W/BlufiClientImpl: response ==: [61, 4, 1, 23, 0, 0, 0, 1, 6, -68, -35, -62, -43, 78, 124, 2, 10, 90, 101, 98, 114, 97, 95, 104, 111, 109, 101]
2020-03-26 08:55:40.052 6290-6401/com.espressif.espblufi W/BlufiClientImpl: gattWrite= [20, 0, 4, 0]
2020-03-26 08:55:40.141 6290-6295/com.espressif.espblufi I/ressif.espbluf: Compiler allocated 4MB to compile void android.view.ViewRootImpl.performTraversals()
2020-03-26 08:55:40.187 6290-6344/com.espressif.espblufi W/BlufiClientImpl: response ==: [61, 4, 2, 23, 0, 0, 0, 1, 6, -68, -35, -62, -43, 78, 124, 2, 10, 90, 101, 98, 114, 97, 95, 104, 111, 109, 101]

tangchuanyong, 2020-03-26:
已解决，硬件端数据问题！

Alvin1Zhang, 2020-03-26:
@tangchuanyong Glad your issue has been resolved, could you help close the issue? Thanks.

Alvin1Zhang, 2020-03-27:
@tangchuanyong Thanks for reporting. Feel free to reopen if the issue reappears. Thanks.

closed 2020-03-27 by Alvin1Zhang