espressif/esp-idf #4998: BLE MESH sensor server/client example request

sheldon123z, 2020-03-26:
Hi, 
As the topic stated, any BLE mesh sensor server or client examples would be very helpful for developers to develop products with esp-idf framework.   

Thanks!

Alvin1Zhang, 2020-03-26:
@sheldon123z  Thanks for the feature request, we will look into and evaluate. Thanks.

Campou, 2020-03-27:
Hi @sheldon123z 

Thanks for the feedback, I will try to provide a sensor model example in the next few weeks.

Thanks.

sheldon123z, 2020-03-30:
Hi @Campou ,

Thanks for your reply. Since I am using multiple esp32 boards to develop a ble-mesh sensor network, I do want to know if there are specific documents/guidelines that I can use for programming sensor models? 

Much appreciated if you can provide.

Best.

nanaadum, 2020-04-15:
Hi @Campou, have you made some progress with the sensor model example?  Also, are there any example codes so far that can be edited to become a BLE mesh sensor model?

Campou, 2020-04-21:
Hi @nanaadum,

The examples about sensor models (client, server and setup server) are ready, waiting to be merged internally. And it will synchronized to Github soon.

Thanks for your patience.

nanaadum, 2020-04-21:
Hi @Campou,

That's awesome! Thanks for the update.

sheldon123z, 2020-04-22:
@Campou awesome! I will check it out

nanaadum, 2020-04-24:
Hi @Campou please can you help me out with where I can find the examples for the sensor models when they come out?
Also please do you have an idea of what day the sensor model code will be synchronized to github?

nanaadum, 2020-04-24:
Thanks a lot for helping @Campou

sheldon123z, 2020-04-28:
Hi @Campou , the examples are excellent and they are very helpful!!! Thank you very much! issue closed.

closed 2020-04-28 by sheldon123z

nanaadum, 2020-04-28:
Hi @Campou thank you so much. Very grateful!!!