espressif/esp-idf #4999: I2C driver bug

fpgamaster, 2020-03-26:
- IDF version: v4.2-dev-792-g6330b3345

components/driver/i2c.c:

esp_err_t i2c_master_cmd_begin(i2c_port_t i2c_num, i2c_cmd_handle_t cmd_handle, TickType_t ticks_to_wait)
{
....
       TickType_t wait_time = xTaskGetTickCount();
        if (wait_time - ticks_start > ticks_to_wait) { // out of time
            wait_time = I2C_CMD_ALIVE_INTERVAL_TICK;
        } else {
            wait_time = ticks_to_wait - (wait_time - ticks_start);
            if (wait_time < I2C_CMD_ALIVE_INTERVAL_TICK) {
                wait_time = I2C_CMD_ALIVE_INTERVAL_TICK;
            }
        }
....
}
This code results in wait_time = I2C_CMD_ALIVE_INTERVAL_TICK.
For situations when a timeout is needed it defaults to 1s (I2C_CMD_ALIVE_INTERVAL_TICK)!

A simple workaround is to replace this part of code with:
wait_time = ticks_to_wait;

koobest, 2020-03-27:
Hi, @fpgamaster
Thanks reporting this. but this timeout is not  for the whole transmission, but just to ensure that the FSM is not stuck.

thanks !!

fpgamaster, 2020-03-27:
Hi @koobest ,
Yes, but if a device will respond with a delay you need to set a timeout and to poll the device again.
In this case no matter what timeout you set below I2C_CMD_ALIVE_INTERVAL_TICK it defaults to I2C_CMD_ALIVE_INTERVAL_TICK.
I hit this issue playing with a SHT20 sensor.

Best Regards

koobest, 2020-04-02:
Hi, 
I refer to its data sheet. When using `Hold Master mode`, the sensor will pull down the SCL (SCL stretch), but due to the hardware limitation, the stretching time cannot exceed 1048576 * Cycle_apb (13ms), otherwise it will cause `timeout`. This timeout is not the timeout set by `i2c_master_cmd_begin`. it is seted by `i2c_set_timeout`. 
I also see that there is a `No Hold Master mode`, is it possible to use this mode?

thanks

fpgamaster, 2020-04-03:
Hi @koobest,
I forgot to mention that all test were in 'No hold' mode.
Best regards

DavidJRobertson, 2020-12-27:
I am encountering this problem as well @koobest  

The device I am using is the [LTC2499 ADC](https://www.analog.com/media/en/technical-documentation/data-sheets/2499fe.pdf). With this part, a conversion is initiated by addressing the device for a write (and optionally writing config data) - the stop bit of the write cycle initiates the conversion. Then, at a later time, a read can be performed to retrieve the result. However, while the conversion is in progress, the device does not respond to reads at all (i.e. it does not acknowledge the address with read flag).

If a sufficient amount of time is allowed before attempting the read, then everything works. But, if you try to poll the device, you will run into this bug. No matter what value you give for `ticks_to_wait` when calling `i2c_master_cmd_begin`, it will block the task for a minimum of one second. 

Probing the bus with a logic analyzer, I see the following:
[image]

tcbennun, 2021-01-08:
I'm suffering from this bug as well.

I have a system where multiple devices sit on the same I2C bus, and I poll each on a regular basis, at well over 100 Hz. The timing here needs to be precise.

Occasionally, one of the devices has an unavoidable failure and will not respond on the I2C bus for some time.

I don't understand why the timeout specified in `I2C_CMD_ALIVE_INTERVAL_TICK` is not configurable. It is **unacceptable** that I cannot use the I2C bus for 1 second if a device failure occurs. I need to communicate with every other device on the bus several times during that interval.

The `I2C_CMD_ALIVE_INTERVAL_TICK` parameter, in my case, should be no more than 10 ms in order to manage these timeouts acceptably.

andycarle, 2021-02-26:
To add a vote here: the [Moddable SDK](https://github.com/Moddable-OpenSource/moddable) team would also like to see a way to customize `I2C_CMD_ALIVE_INTERVAL_TICK`. 

We have a software example for developers that automatically scans the I2C bus for connected devices. Because we attempt a read from every possible I2C peripheral address, it encounters many expected device failures during normal operation. On other platforms we can scan every address in less than a second, but due to this issue it takes more than a minute to do the same on ESP32.

Because individual developers are building with our tools and their own ESP-IDF clones, we cannot easily simply modify `i2c.c`. It would be very helpful if there were a way to set `I2C_CMD_ALIVE_INTERVAL_TICK` in the sdkconfig or if `i2c_master_cmd_begin` respected the `ticks_to_wait` argument even if it was shorter than `I2C_CMD_ALIVE_INTERVAL_TICK`.

JesusFreke, 2021-09-15:
I just ran into this as well. The effect of this value is that it's impossible to use a timeout less than a full 1 second in `i2c_master_cmd_begin`, because anything less than this value will get replaced [here](https://github.com/espressif/esp-idf/blob/fbd25c3/components/driver/i2c.c#L1456).

dhalbert, 2022-02-03:
We are looking at this in https://github.com/adafruit/circuitpython/pull/5908.

I went back in time, and when `I2C_CMD_ALIVE_INTERVAL_TICK` was first introduced, it was a ceiling, not a floor:
https://github.com/espressif/esp-idf/blame/9075b507b54a99ce732fc2819ba17274d9a69ab5/components/driver/i2c.c#L1142
```c
        TickType_t wait_time = (ticks_to_wait < (I2C_CMD_ALIVE_INTERVAL_TICK) ? ticks_to_wait : (I2C_CMD_ALIVE_INTERVAL_TICK));

simondddd, 2022-02-11:
I thing this commit is where it get wrong
https://github.com/espressif/esp-idf/commit/d913fff6d70152d201c84c0026312c119bfe9871#diff-380dc90b7523e5ecbb9c3fb9799a4c3db81ef22a126b3ffac560c7750ab72258L1142-R1151

me-no-dev, 2022-04-28:
I can confirm this is an issue as well. Now that Arduino uses IDF's I2C driver, we are getting reports and I was able to reproduce it. `I2C_CMD_ALIVE_INTERVAL_TICK` should be configurable, though in reality the code should not wait longer than the provided `ticks_to_wait`. Please fix and backport this issue.

theotherjenkutler, 2022-12-06:
@fpgamaster yes its been a hopeful few years for reliable i2c communication.

chuyec, 2022-12-15:
I have this problem too. Hope for a speedy solution

fpgamaster, 2022-12-15:
Hi @chuyec 

Try my ugly workaround. I use it for a while :)
The diff against esp-idf v4.4.3:

```
diff --git a/components/driver/i2c.c b/components/driver/i2c.c
index f4beb3bcf1..d2c9da4448 100644
--- a/components/driver/i2c.c
+++ b/components/driver/i2c.c
@@ -1477,6 +1477,7 @@ esp_err_t i2c_master_cmd_begin(i2c_port_t i2c_num, i2c_cmd_handle_t cmd_handle,
     i2c_cmd_evt_t evt;
     while (1) {
         TickType_t wait_time = xTaskGetTickCount();
+#if 0
         if (wait_time - ticks_start > ticks_to_wait) { // out of time
             wait_time = I2C_CMD_ALIVE_INTERVAL_TICK;
         } else {
@@ -1485,6 +1486,9 @@ esp_err_t i2c_master_cmd_begin(i2c_port_t i2c_num, i2c_cmd_handle_t cmd_handle,
                 wait_time = I2C_CMD_ALIVE_INTERVAL_TICK;
             }
         }
+#else
+       wait_time = ticks_to_wait;
+#endif
         // In master mode, since we don't have an interrupt to detective bus error or FSM state, what we do here is to make
         // sure the interrupt mechanism for master mode is still working.
         // If the command sending is not finished and there is no interrupt any more, the bus is probably dead caused by external noise.

```
Best regards

chuyec, 2022-12-16:
@fpgamaster Thanks a lot for your suggestion!

I have already applied this fix from @KlausPopp: https://github.com/ci4rail/esp-idf/pull/5/files 
It seems to have helped.

chuyec, 2022-12-16:
> @KlaussPopp Would you be interested in proposing this as a PR to https://github.com/espressif/esp-idf ?

Hi @dhalbert!

You made a mistake in mentioning the author. Of course, if he hasn't changed his nickname.🤔  He may not have seen your question.
The correct mention is @KlausPopp
🙂

Best regards

AxelLin, 2022-12-17:
> Hi @chuyec
>
> Try my ugly workaround. I use it for a while :) The diff against esp-idf v4.4.3:
>
> ```
> diff --git a/components/driver/i2c.c b/components/driver/i2c.c
> index f4beb3bcf1..d2c9da4448 100644 [...]

@o-marshmallow @mythbuster5 
Seems this issue was reported by multiple users but there is no response from espressif.
Can you take a look at this issue? Or redirect this to someone who is working on I2C driver?

tracestep, 2023-04-03:
I am having the same issue... the thing is I am using an environment in which I can't simply patch the idf library. Espressif, please fix this.

theotherjenkutler, 2023-04-04:
@tracestep it’s been years that I’ve been waiting for i2c to work reliably with anything @espressif makes.

AxelLin, 2023-04-07:
> @tracestep it’s been years that I’ve been waiting for i2c to work reliably with anything @espressif makes.

@theotherjenkutler esp-idf version are you using now? (git describe --tags output).
@igrr This issue was reported by 2020 and people keep reporting similar issue. Any one can take a look at the issue?

luxflow, 2023-08-26:
for anyone having this problem when using platformio (espressif32 platform)
rollback to previous version (esp-idf v4.4.3)

```ini
#add platformio.ini
platform = espressif32@5.2.0
platform_packages =
    toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
```

not-my-real-name, 2023-11-09:
This is still an issue in 4.4.5. A quick glance at the master branch makes me think it will still be an issue there too. 

I2C_CMD_ALIVE_INTERVAL_TICK is internal to i2c.c and is only used for the modbus master in one place. It looks like all it is doing is making sure the i2c event queue doesn't wait less than a minimum amount of time. It looks like this has been done because there is a generic event in the interrupt saying some sort of master activity has occured.  If it doesn't have some sort of activity it does a low level reset on the i2c state machine. I don't see why it has to be so long though? If you know your I2C is slow, or going to clock stretch a long time, you would have to make your total timeout for that command longer than the maximum clock stretch period you can expect anyway?  If you are going fast you could quite reasonably consider the i2c hardware dead enough to need a reset after much less than 1 second.

I might be missing something, but it looks reducing I2C_CMD_ALIVE_INTERVAL_TICK to a lower value would work fine? Even 50ms seems generous? I have only done about 5 minutes testing with 50ms but it seems to work ok so far?

AxelLin, 2023-11-09:
> > @tracestep it’s been years that I’ve been waiting for i2c to work reliably with anything @espressif makes.
> 
> @igrr This issue was reported by 2020 and people keep reporting similar issue. Any one can take a look at the issue?

Hi @igrr 
The issue status becomes "Reviewing" since Aug 30 but I don't find anyone assigned to this issue.
Is it possible to post the fix here for testing?

AxelLin, 2023-11-10:
The same issue: https://github.com/espressif/esp-idf/issues/12079

AxelLin, 2023-11-16:
https://github.com/espressif/esp-idf/issues/11605

konwektor, 2023-11-18:
go this same bug. Transmission over i2c between esp32 and arduino leonardo for bluetooth controllers support. During gameplay stops reacting fo 1 sec. i dont have this problem with xbox one s/x series x/s gamnepads  but all others supported by blueretro, from time to time doesnt react 1 sec.
`#include <esp32/rom/ets_sys.h>
#include "adapter/adapter.h"
#include "adapter/config.h"
#include "driver/i2c.h"
#include <soc/i2c_periph.h>
#include "ogx360_i2c.h"
#include <string.h>

bool playerConnected[4] = {0};

void ogx360_check_connected_controllers()
{
    ets_printf("OGX360_I2C: Looking for attached OGX360 modules\n");
    for (int i=0;i<4;i++)
    {
        const char ping[] = { 0xAA };
        esp_err_t result = i2c_master_write_to_device(I2C_NUM_0, i + 1, (void*)ping, sizeof(ping), 150);
        playerConnected[i] = (result == ESP_OK);
        ets_printf("OGX360_I2C: Player %d %s\n", i, playerConnected[i] ? "Found" : "Not Found");
    }
}

void ogx360_i2c_init(void) {
    i2c_config_t conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = 22,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_io_num = 21,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = 400000,
        .clk_flags = 0,
    };
    ets_printf("OGX360_I2C: Init I2C\n");
    i2c_param_config(I2C_NUM_0, &conf);
    i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER, 0, 0, 0);
    I2C0.timeout.tout = 1048575;
    i2c_set_timeout(I2C_NUM_0, 1048575);
}

bool initialized = false;

void ogx360_process(uint8_t player)
{
    if (!initialized)
    {
        ogx360_i2c_init();
        ogx360_check_connected_controllers();
        initialized = true;
    }

//  while(1) // Hack
    {
        //for (int player=0;player<4;player++)
        {
            if (playerConnected[player])
            {
                esp_err_t result = i2c_master_write_to_device(I2C_NUM_0, player + 1, (void*)&wired_adapter.data[player].output, 21, 1);
                if (result != ESP_OK) ets_printf("OGX360 write result: %d Index: %d\n",result,player);
                if (result == 0)
                {    
                    memset((void*)&wired_adapter.data[player].output,0,7);
                    result = i2c_master_read_from_device(I2C_NUM_0,player + 1, (void*)&wired_adapter.data[player].output, 7, 1);
                    if (result != ESP_OK) ets_printf("OGX360 read result: %d Index: %d\n",result,player);
                }
            }
        }
    }
}`

not-my-real-name, 2023-12-04:
I had a look at the new API a little while ago and was disappointed to see it doesn't look like you can do partial transactions like the old one lets you do.

We have an I2C slave where the request changes part way through depending on the reply of the slave.  The read length is not known/fixed at compile time.

E.g. Slave replies with a number of bytes to follow.  On the current cmd list system I send two separate parts. One from start to the length byte, and then another to read the payload and send the stop etc.

In our case we could do a repeat start i guess at the expense of a little speed, but the new API, while simple, is more abstracted than the old one.   Actually, looking at it I don't think you can do that either. You can only do a repeat start on a single transmit/receive operation... 

The old API is a bit long winded, but it does give a lot more control.

I'm having good luck so far with just using I2C_CMD_ALIVE_INTERVAL_TICK at 50ms by the way.

konwektor, 2023-12-09:
> go this same bug. Transmission over i2c between esp32 and arduino leonardo for bluetooth controllers support. During gameplay stops reacting fo 1 sec. i dont have this problem with xbox one s/x series x/s gamnepads but all others supported by blueretro, from time to time doesnt react 1 sec.
> 
> [#ogx360_i2c.c](https://github.com/konwektor/BlueRetro/commit/cbe18f412a3ba41a8ea35c7feed5e65e532392c5)

Small update; I think problem is triggered when Bluetooth classic is used in the same time with I2c.
Mine Esp32 is working as a master <=> level (5V-3,3V)shifter<=> arduino pro micro (leonardo version 5V 16Mhz).
Gamepad (BLE, or classic) <=> Esp32 master I2C <=> Leonardo I2C Slave  <=> USB host (Xbox)
In my case I dont have 1sec. timeout when im using  Gamepads from Xbox One S/X or Xbox Series X/S. 
All those are working after software update some time ago only in BLE mode..

Mine PS4 Gamepad, and Wii U are BT classic only. When I using  those, timeout is triggered.

AxelLin, 2023-12-16:
> Hi @igrr The issue status becomes "Reviewing" since Aug 30 but I don't find anyone assigned to this issue.

Now I'm wondering if this will be fixed in 2023?

timoxd7, 2024-02-22:
> Now I'm wondering if this will be fixed in 2023?

@AxelLin Now its 2024, still waiting. Need the patch too... #12079

AxelLin, 2024-03-11:
> @AxelLin Now its 2024, still waiting. Need the patch too... #12079

I'm pretty sure the espressif's developers know current behavior and this issue.
Just no idea why there is no response from espressif.
@igrr

simondddd, 2024-03-13:
> I thing this commit is where it gets wrong [d913fff#diff-380dc90b7523e5ecbb9c3fb9799a4c3db81ef22a126b3ffac560c7750ab72258L1142-R1151](https://github.com/espressif/esp-idf/commit/d913fff6d70152d201c84c0026312c119bfe9871#diff-380dc90b7523e5ecbb9c3fb9799a4c3db81ef22a126b3ffac560c7750ab72258L1142-R1151)

I insist, I think in line 1148 should be >, not <

dhalbert, 2024-03-13:
@simonddd that was my conclusion above as well

closed 2024-03-16 by espressif-bot

AxelLin, 2024-04-21:
@o-marshmallow 
still needs to backport the fix for v4.4, v5.0 and v5.2.