espressif/esp-idf #5001: xTaskCreate(): do not "add it to the list of tasks that are ready to run"

keelung-yang, 2020-03-26:
Firstly, It's an issue of FreeRTOS: Why force my task ready to run? It's unreasonable since I've no choice to keep it quiet untill I told it run.

There are many tasks need to run immediatly. BUT not ALL of them.

Such as, some tasks created, but should not run until some event happens. If that event is not happened, the task function should never be executed.

Here is an example, _ftx_task()_ is called enen if _ftx->period == 0_
```c
int dmodbus_ftx(diag_ftx_t * ftx)
{
    static TaskHandle_t task_handle;
    static diag_ftx_t ftx_holded;
    if (task_handle == NULL)
    {
        xTaskCreate(ftx_task, "dm-ftx", 2048, &ftx_holded, 
            tskIDLE_PRIORITY + 1, &task_handle);
        vTaskSuspend(task_handle);
    }
    
    if (ftx->period == 0xFFFF)
    {
        send_ftx(ftx);
    }
    else if (ftx->period > 0)
    {
        memcpy(&ftx_holded, ftx, sizeof(diag_ftx_t));
        vTaskResume(task_handle);
    }
    else if (ftx->period == 0)
    {
        vTaskSuspend(task_handle);
    }

    return MB_DIAG_ERROR_OK;
}
```

negativekelvin, 2020-03-26:
>should not run until some event happens

Freertos has semaphores, queues, event groups, and notifications to solve this problem

keelung-yang, 2020-03-26:
> > should not run until some event happens
> 
> Freertos has semaphores, queues, event groups, and notifications to solve this problem

Yep, but need more code, not concise.

The issue is the API design, not the solution o bypassing it.
User have no choice, must add unnessary code to prevent it run immediatly.

igrr, 2020-03-26:
@keelung-yang As you have observed, FreeRTOS doesn't provide any flags or functions to create a task in a suspended state. There are workarounds, as @negativekelvin has mentioned, and also some more mentioned in the official reply on FreeRTOS forum: https://forums.freertos.org/t/how-to-create-task-in-suspended-state/774. Of all the mentioned options, I think the most light weight one is to use task notifications. It adds a couple of lines of code, so it can be considered negligible increase in complexity.

If you would like to get this addressed at FreeRTOS API level, I suggest opening an issue at https://github.com/aws/amazon-freertos/issues. Our plan is to update FreeRTOS to an upstream version, therefore going forward we would prefer keeping IDF specific changes to the minimum.

closed 2020-03-26 by igrr