espressif/esp-idf #5003: Timeline for std::filesystem support

xuio, 2020-03-26:
Do you plan to support `std::filesystem`? (-> #2449)
Is it possible to build `xtensa-esp32-elf-g++` with different flags to support it?

igrr, 2020-03-29:
@xuio We do plan to enable std::filesystem support once we upgrade to a GCC version where it is fully supported (It is considered experimental in GCC 8).

For now you can try building the toolchain yourself, adding `--enable-libstdcxx-filesystem-ts` to the list of GCC configuration flags (`CT_CC_GCC_EXTRA_CONFIG_ARRAY` option in crosstool-NG). If you meet issues along the way, please let us know.

Regarding support for C++17 / C++20 in IDF, see also the discussion in https://github.com/espressif/esp-idf/pull/4846.

xuio, 2020-04-05:
I tried adding the option while building the toolchain, but there is no library created in the corresponding folder.

Apart from `std::filesystem`, is it possible to build `libatomic` for xtensa? I am trying to build protobuf c++ and eventually gRPC, but it needs 64bit atomics. (it fails to find `__atomic_fetch_add_8` during linking)

igrr, 2020-04-09:
Thanks for the feedback, will look into std::filesystem support while preparing for the next toolchain release.

Regarding __atomic_fetch_add_8, please subscribe to https://github.com/espressif/esp-idf/issues/3163 — it will get updated and closed automatically when we implement this.

SebiTimeWaster, 2021-10-18:
any roadmap for filesystem?

chipweinberger, 2021-12-01:
It looks like ESP-IDF v3.3 uses  GCC v8.4, shouldn't std::filesystem be implemented there?

Why does it not work?

0xjakob, 2021-12-02:
@chipweinberger 

> The filesystem library facilities may be unavailable if a hierarchical file system is not accessible to the implementation, ...

https://en.cppreference.com/w/cpp/filesystem

nebkat, 2023-12-20:
Has there been any further consideration given to this?

I have found most `std::filesystem` functions actually work quite well, except for some unusual cases in path concatenation, and `std::filesystem::directory_iterator` does not work at all.

It appears this may be because `_GLIBCXX_HAVE_DIRENT_H` is not defined, even though we do have `dirent.h`in `newlib/platform_include`. Is this something that needs to be added to https://github.com/espressif/gcc/blob/esp-12_2_0/libstdc++-v3/config/os/newlib/os_defines.h ?

0xjakob, 2024-02-27:
@nebkat We will take a look to see if we can just add the correct definition.

Lapshin, 2024-05-02:
Should work after commit https://github.com/espressif/esp-idf/commit/994b4ed459269be1f6548137eac84ec3f9a8e37f

Please open another issue if something wrong is found