espressif/esp-idf #5004: tcp_connect does not respect the timeout parameter

X-Ryl669, 2020-03-26:
In current repository, `tcp_connect` in esp_transport has this signature:
```
static int tcp_connect(esp_transport_handle_t t, const char *host, int port, int timeout_ms)
```
Yet, inside the function, there are 2 blocking functions being called and none will respect the `timeout_ms` given.
The first one is `inet_pton`/`resolve_dns` and the second one is `connect`.
Please notice that lwip does not respect the socket option `SO_SNDTIMEO` for connect.

As a result, if a server takes forever to connect (for example, by being down), `tcp_connect` will block indefinitively.

The correct way to have a timeout here is to set the socket as non blocking (via `fcntl`), call `connect`, then `select` for socket writeability within the given timeout.

X-Ryl669, 2020-03-26:
Also, as a more general remark, the parameter `timeout_ms` in all those transport functions should be called `inactivity_time_ms` instead since the timeout option set on a socket is reset on socket activity. So if you're receiving a byte per second with a `timeout_ms` set to 3s, and you are receiving 5 bytes, it'll actually wait 5 seconds to complete.

anecdata, 2020-04-18:
We've noticed that TCP sockets seem to block if, for example, an unreachable numeric IPv6 address is supplied. Sounds like this may be the underlying issue.

closed 2020-05-07 by espressif-bot

X-Ryl669, 2020-05-08:
You've resolved one issue but not the other. `resolve_dns` is still blocking.