espressif/esp-idf #5005: Cannot read larger BLE characteristics from iPhone

sebirdman, 2020-03-26:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Module or chip used:  ESP32-WROOM-32
- IDF version: release/v3.3 @ c44a9f9
- Build System:        CMake

## Problem Description

With the latest commits in the v3.3 release branch. I cannot read larger data from my iPhone. This is NOT an issue on 3.3.1.

### Expected Behavior

The iPhone reads the characteristic and stays connected 100% of the time

### Actual Behavior

~90% of the time, the iPhone tries to read the data and instead disconnects with the error on the esp32 side being
#define GATT_CONN_TIMEOUT                   HCI_ERR_CONNECTION_TOUT

### Steps to reproduce

1) iPhone App negotiates largest mtu size (517)
2) iPhone App attempts to read a characteristic with a value of around 450 bytes

### Code to reproduce this issue

If desired, i can provide some, but will take some time to pull out of my app.

### Debugging

I've tried to shorten this characteristic's value and see where the error starts happening, it appears that somewhere between 200 and 255 bytes, i stop being able to reliably read.

### Workarounds:

1) Revert: I've done some digging and it appears that this issue occurs right at the time that this commit
https://github.com/espressif/esp32-bt-lib/commit/5b3a571bfc58b5f05a09046f3fe02c08141df517
went in. Reverting to before this seems to fix it.
2) Set lower maximum MTU

sebirdman, 2020-03-26:
I should also note: i've not tested this on an android device with an MTU of 517. It may be an issue there as well.

xiewenxiang, 2020-08-20:
@sebirdman 

I have test it, but can't reproduce this issue.

So we need you to provide code that can be reproduced.

IDF version: 

```
commit c44a9f9a4e80973b12e989640abc81bf77f2c6a8 (HEAD -> release/v3.3)
Merge: 6feb88f69b 649a6afeb9
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Wed Mar 25 11:15:52 2020 +0800

    Merge branch 'bugfix/send_ok_when_sta_disconnect_for_v3.3' into 'release/v3.3'
    
    lw-ip:bugfix for TCPv6 send ok when sta disconnect for 3.3
    
    See merge request espressif/esp-idf!8016
```

log:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6796
load:0x40078000,len:12072
load:0x40080400,len:6708
entry 0x40080778
I (71) boot: Chip Revision: 0
I (32) boot: ESP-IDF v3.3.2-8-gc44a9f9a4e-dirty 2nd stage bootloader
I (32) boot: compile time 16:59:26
I (32) boot: Enabling RNG early entropy source...
I (38) boot: SPI Speed      : 40MHz
I (42) boot: SPI Mode       : DIO
I (46) boot: SPI Flash Size : 4MB
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x20c44 (134212) map
I (144) esp_image: segment 1: paddr=0x00030c6c vaddr=0x3ffbdb60 size=0x03164 ( 12644) load
I (149) esp_image: segment 2: paddr=0x00033dd8 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/xiewenxiang/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (151) esp_image: segment 3: paddr=0x000341e0 vaddr=0x40080400 size=0x0be30 ( 48688) load
I (180) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x729fc (469500) map
0x400d0018: _flash_cache_start at ??:?

I (345) esp_image: segment 5: paddr=0x000b2a1c vaddr=0x4008c230 size=0x06058 ( 24664) load
0x4008c230: timer_armed at /Users/xiewenxiang/esp/esp-idf/components/esp32/esp_timer.c:435

I (366) boot: Loaded app from partition at offset 0x10000
I (366) boot: Disabling RNG early entropy source...
I (367) cpu_start: Pro cpu up.
I (370) cpu_start: Application information:
I (375) cpu_start: Project name:     gatt_server_service_table_demo
I (382) cpu_start: App version:      v3.3.2-8-gc44a9f9a4e-dirty
I (389) cpu_start: Compile time:     Aug 20 2020 16:59:28
I (395) cpu_start: ELF file SHA256:  4a189b656d2077b4...
I (401) cpu_start: ESP-IDF:          v3.3.2-8-gc44a9f9a4e-dirty
I (407) cpu_start: Starting app cpu, entry point is 0x400810fc
0x400810fc: call_start_cpu1 at /Users/xiewenxiang/esp/esp-idf/components/esp32/cpu_start.c:268

I (0) cpu_start: App cpu up.
I (418) heap_init: Initializing. RAM available for dynamic allocation:
I (425) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (431) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (437) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (443) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (449) heap_init: At 3FFC9400 len 00016C00 (91 KiB): DRAM
I (455) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (462) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (468) heap_init: At 40092288 len 0000DD78 (55 KiB): IRAM
I (474) cpu_start: Pro cpu start user code
I (157) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (182) BTDM_INIT: BT controller compile version [e989f20]
I (182) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (312) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (692) GATTS_TABLE_DEMO: create attribute table successfully, the number handle = 8

I (692) GATTS_TABLE_DEMO: SERVICE_START_EVT, status 0, service_handle 40
I (702) GATTS_TABLE_DEMO: advertising start successfully
I (34412) GATTS_TABLE_DEMO: ESP_GATTS_CONNECT_EVT, conn_id = 0
I (34412) GATTS_TABLE_DEMO: 67 23 c5 6e a0 da 
I (34622) GATTS_TABLE_DEMO: ESP_GATTS_MTU_EVT, MTU 517
I (35012) GATTS_TABLE_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 24,latency = 0, timeout = 400
I (35492) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (38492) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (40742) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (46292) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (48362) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (49982) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (50492) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
```

test code(change characteristic value length to 450 bytes and support 517 MTU size):

[gatt_server_service_table.zip](https://github.com/espressif/esp-idf/files/5101977/gatt_server_service_table.zip)

capture screenshot:

[image]

BLE debug APP screen shoot:

[image]

chegewara, 2020-08-20:
iPhone does not support MTU over 185 IIRC.

xiewenxiang, 2020-08-20:
In recent years, the iPhone should be supported to 527 MTU size, only many years ago it was 188. I don't have such a phone at hand for the time being, So I need to find it and then give you a reply ASAP.

xiewenxiang, 2020-08-20:
@chegewara 

I tested the problem on an older iPhone with an MTU of 185.

First a Read Request is sent, followed by two packets of READ BLOB Requests. There is no problem reading 450 bytes of data

xiewenxiang, 2020-08-20:
[image]
[image]
[image]
[image]

sebirdman, 2020-09-02:
This was resolved for me as of idf 4.0

closed 2020-09-14 by sebirdman