espressif/esp-idf #5006: FOTA in 2MB?  Has Anyone Done This With Reasonable Success?  (ublox NINA-W102)

jwestmoreland, 2020-03-26:
Hello Everyone,

This is more of a general question - I was wondering how successful some have been getting FOTA working with one of the modules that only has 2MB - I can see it could be possible to change the partition table - but 2MB is small for this task.

Does anyone out there have it reliably working and willing to comment?

I've seen some u-toob videos where some have rolled their own so to speak - but would like to hear from some of you that have been working on this issue for a while and specifically dealing with squeezing things into 2MB.

Thanks In Advance For Your Time,
John W.

iosix, 2020-03-27:
Yes, we have this in production in the field.  We have two banks of size
0xF0000, and switch between the two for OTA.  We use the opposite bank as a
buffer for data when we aren't updating.

Robert Vogt IV
CEO
IOSiX, LLC
1300 Tefft Ct #1
Saline, MI  48176
Reception: (855) OBD-1939
Direct/Cell: +1-734-730-9690
robert@iosix.com

jwestmoreland, 2020-03-27:
Hello Robert,

That answers one of my questions - my current app is ~0xF0000 in bytes - so I was wondering how feasible this really was.

Good news to know some are going it successfully, and yes, I know I can pare down my app - but I really like the AT Command set - I bet some here don't even recognize the term 'Hayes'.  Anyway, always interesting to here this stuff.

Thanks Again,
John

iosix, 2020-03-27:
Turning off built-in ESP debug is key to reducing size.  We are doing WiFi,
BTLE, PPP, microSD/FAT32, NMEA, and CAN+other protocols in around 0xD0000.

Robert Vogt IV
CEO
IOSiX, LLC
1300 Tefft Ct #1
Saline, MI  48176
Reception: (855) OBD-1939
Direct/Cell: +1-734-730-9690
robert@iosix.com

jwestmoreland, 2020-03-27:
Robert,

Thanks for the great info - but it's too much fun running JTAG on these devices!  Ha!

But, yes, I bet it took a little work to get that squeezed down - but that's not a bad footprint
for all of that capability.

I haven't tried CAN yet but have been curious about that on these ESP32's.  Lot of capability in a small package.

Thanks Again,
John

Alvin1Zhang, 2020-04-01:
@jwestmoreland Thanks for the question, for how questions or third party issues, suggest you could raise a post on our forum https://esp32.com, thanks.

closed 2020-04-01 by Alvin1Zhang