espressif/esp-idf #5007: esp-idf/examples/peripherals/rmt/ir_protocols  not getting output, After running the above code I am  not able to get the NEC device working, Please guide me what should parameters do I need to change for get it working

sara-patel, 2020-03-27:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2020-03-27:
@sara-patel Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

sara-patel, 2020-03-27:
Hi  Alvin, Thanks for responding. I am using ESP32 and programming it using ESP-IDF environment.I tried using the ir_protocol example it is compiling without errors, and also I am able to flash on the ESP32 . I wanted to check NEC protocol so i just changed command and ID value but my NEC device is not responding so my concern was does the example work as it is or need to change any other parameters?

koobest, 2020-03-30:
Hi, @sara-patel

Does ESP32 act as a receiver or transmitter?
Can you check these tow things of the:
1. On esp32,  the transmitter can carrier modulated, but the receiver can not carrier demodulation.
2. The idle level. Some devices are high when idle, some are low. you can use the oscilloscope to capture the waveform of the correct communication.

thanks !!

sara-patel, 2020-03-30:
hello Koobest,
ESP32 acts as both transmitter and receiver but I am now just checking the transmitter part have commented the receiver part in the code. And as  I am working from home I do not have oscilloscope to capture the waveform. It would be helpful of you if you can suggest me any other alternate wave to check it.

sara-patel, 2020-03-31:
one more question I have why does receive task does not work alone? I need to always start the tx task  in order to receive the raw data from the rx task. Can someone help me with the receive task  on how to make it working independently?

koobest, 2020-04-01:
Hi,
I don’t know if your transmitting device has carrier enabled, and the IDLE level is high or low. If you want to use ESP32 for docking test, you should connect TX_PIN and RX_PIN, and modify the example:
[image]

[image]

sara-patel, 2020-04-01:
Hi Koobest,
Thanks for your reply and my tx has stared working now. Now I want to use the example_ir_rx_task(),
to receive any IR raw data irrespective of any protocol then what change do I need to make? As of now when I comment the tx task and receive ir data from an non-NEC  protocol remote then I am not able to get the full data correctly ? Can you help me with it ?

koobest, 2020-04-01:
The receiving task can currently only parse the protocols of NEC and RC5, and with out carrier.
If you need to parse other protocols, you need to implement it yourself.  the raw data is in there:

[image]

sara-patel, 2020-04-01:
ok, Thanks. Will try it out . And is there any limitation of the data it receives in rx task at once?

koobest, 2020-04-01:
Bit length cannot exceed 63 bits, unless more than 1 memory block is configured for the receiving module, see `rmt_config_t` and the TRM.

sara-patel, 2020-04-01:
Sorry I did not get what is TRM?

sara-patel, 2020-04-01:
Thanks Koobest, I am able to receive full data by changing the mem_block_num.
I have one more question, is there any difference between IR remote of Air Conditioners and IR remote of TV? As after being able to operate AC's by receiving their raw data in rx task and sending the received data in tx task   am bale to operate two different types of AC's but not TV

koobest, 2020-04-01:
May be different in IDLE level or/and carrier.

sara-patel, 2020-04-01:
ok will have to go in depth I guess.

sara-patel, 2020-04-02:
I have noticed in this example that whenever RX task runs it only scans and waits for raw data once, after that whatever tx task sends raw data, then next time the rx task considers t as its new scanned raw data? why  and how does it happen? As I want to make rx task to read new data every time and then send the same in tx task  so can anyone guide me with it ?

sara-patel, 2020-04-09:
can anyone please guide me on how to stop rx task from receiving the raw data once it has read and transmitted it using tx task? As I want rx task to every time read a new data

koobest, 2020-04-11:
Hi, @sara-patel 
what't you TX and RX configuration?  TX chenne number TX channel mem_block_num, RX channel number and RX mem_block_num ?

the example uses channel0 for TX and channel1 for RX. if the TX mem_block_num > 1, you should change the RX channel number

sara-patel, 2020-04-13:
Thanku Koobest for your reply.After changing the RX channel number and the adding delay after the transmission task I am able to get the example working the way I needed.

Alvin1Zhang, 2020-04-29:
@sara-patel Glad your issue gets resolved. Feel free to reopen if you still have the issue, thanks.

closed 2020-04-29 by Alvin1Zhang

Imdad72, 2023-02-07:
@sara-patel kindly tell me how u controlled non-NEC types remotes.
this is very urgent.

Imdad72, 2023-02-07:
@sara-patel u can reach me personally on imdad475f@gmail.com
*this is v.v urgent