espressif/esp-idf #5008: File Serving Example - HTTP Error Code Not Sent/Received Before Socket Close

mriksman, 2020-03-27:
I loaded the example `file_serving`. In the `upload_script.html` file, I removed the lines;
```
    } else if (fileInput[0].size > 200*1024) {
        alert("File size must be less than 200KB!");
```
Because I wanted to see this code work properly in `file_server.c`;
```
    /* File cannot be larger than a limit */
    if (req->content_len > MAX_FILE_SIZE) {
        ESP_LOGE(TAG, "File too large : %d bytes", req->content_len);
        /* Respond with 400 Bad Request */
        httpd_resp_send_err(req, HTTPD_400_BAD_REQUEST,
                            "File size must be less than "
                            MAX_FILE_SIZE_STR "!");
        /* Return failure to close underlying connection else the
         * incoming file content will keep the socket busy */
        return ESP_FAIL;
    }
```
However, I just receive a `"Server closed the connection abruptly!"` error - not the `HTTPD_400_BAD_REQUEST`. If I set the return to be `ESP_OK`, the correct response is received, but the socket isn't closed so the file continues to be sent to the server.

This issue should have been resolved with TCP_NODELAY here https://github.com/espressif/esp-idf/commit/28412d8cb699b4f2fa34930fc7895444982af312. 
It seems it hasn't.

shubhamkulkarni97, 2020-06-04:
@mriksman,

While debugging, I observed that this issue could be reproduced with some browsers. For example, `upload_script.html` works well with Google Chrome but there is some issue with Safari Web Browser. Can you try using Google Chrome? Please let us know if you face the issue even with Google Chrome.

I also verified Wireshark capture for the response from the server, it indicates `400 Bad Request` error.

> However, I just receive a `"Server closed the connection abruptly!"` error - not the `HTTPD_400_BAD_REQUEST`. If I set the return to be `ESP_OK`, the correct response is received, but the socket isn't closed so the file continues to be sent to the server.

Returning ESP_OK in case of error will not close underlying connection and incoming file will keep the socket busy. Hence, ESP_FAIL should be returned in case of any errors.

mriksman, 2020-06-04:
My tests are all performed in Google Chrome; I use the Developer Tools.

I did some more research, and it doesn't seem to be an ESP specific issue - it's a HTTP standard/browser related issue. HTTP is a synchronous protocol, so the client will send all of its request before waiting for a response.

You could send a Expect: 100-continue from the client first, let the ESP do checks, and then send a 100 Continue in it's response if the client should proceed to upload.

I'd like to see the examples updated to show this - as the current examples do not work as expected.

shubhamkulkarni97, 2022-07-14:
@mriksman I am closing this issue as it is difficult to be handled on device side. Please feel free to reopen if you have any concerns.

closed 2022-07-14 by shubhamkulkarni97