espressif/esp-idf #5009: A stack overflow in task sys_evt has been detected

rebeliousconformist, 2020-03-27:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|
- Kit version (for WroverKit/): [v4]
- Module or chip used:  [|ESP32-WROVER-B|
- IDF version :: v4.2-dev-701-g0ae960f2f
- Build System:         [|CMake & idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
  xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [|Linux|]

- Using an IDE?: [|Yes  eclipse & idf.py from command line
- Power Supply:         [USB|

## Problem Description
//Detailed problem description goes here.

### Expected Behavior

The list of all the running tasks are shown

Ive using the existing event loops 
Im publishing some test data to a mqtt broker
when the WIFI connection disconnets then the MQTT broker should shutdown and restart
also the task which is created should also shutdown. 

The program crashes

### Actual Behavior
<pre>Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6884
load:0x40078000,len:13720
load:0x40080400,len:3920
entry 0x40080680
I (29) boot: ESP-IDF v4.2-dev-701-g0ae960f2f 2nd stage bootloader
I (29) boot: compile time 13:29:11
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1e134 (123188) map
I (160) esp_image: segment 1: paddr=0x0002e15c vaddr=0x3ffb0000 size=0x01ebc (  7868) load
I (163) esp_image: segment 2: paddr=0x00030020 vaddr=0x400d0020 size=0x8c0ec (573676) map
I (383) esp_image: segment 3: paddr=0x000bc114 vaddr=0x3ffb1ebc size=0x01a50 (  6736) load
I (386) esp_image: segment 4: paddr=0x000bdb6c vaddr=0x40080000 size=0x00400 (  1024) load
I (390) esp_image: segment 5: paddr=0x000bdf74 vaddr=0x40080400 size=0x14f5c ( 85852) load
I (447) boot: Loaded app from partition at offset 0x10000
I (447) boot: Disabling RNG early entropy source...
I (448) cpu_start: Pro cpu up.
I (451) cpu_start: Application information:
I (456) cpu_start: Project name:     wifi_station
I (461) cpu_start: App version:      1
I (466) cpu_start: Compile time:     Mar 27 2020 13:29:10
I (472) cpu_start: ELF file SHA256:  43cbe0faf4facc14...
I (478) cpu_start: ESP-IDF:          v4.2-dev-701-g0ae960f2f
I (484) cpu_start: Starting app cpu, entry point is 0x400811b0
I (477) cpu_start: App cpu up.
I (495) heap_init: Initializing. RAM available for dynamic allocation:
I (502) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (508) heap_init: At 3FFB94E0 len 00026B20 (154 KiB): DRAM
I (514) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (520) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (527) heap_init: At 4009535C len 0000ACA4 (43 KiB): IRAM
I (533) cpu_start: Pro cpu start user code
I (551) spi_flash: detected chip: generic
I (552) spi_flash: flash io: dio
I (552) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (560) wifi station: [APP] Startup..
I (560) wifi station: [APP] Free memory: 261344 bytes
I (570) wifi station: [APP] IDF version: v4.2-dev-701-g0ae960f2f
I (610) wifi station: ESP_WIFI_MODE_STA
I (620) wifi:wifi driver task: 3ffd7300, prio:23, stack:3584, core=0
I (620) system_api: Base MAC address is not set
I (620) system_api: read default base MAC address from EFUSE
I (640) wifi:wifi firmware version: d76a2dd
I (640) wifi:config NVS flash: enabled
I (640) wifi:config nano formating: disabled
I (650) wifi:Init dynamic tx buffer num: 32
I (650) wifi:Init data frame dynamic rx buffer num: 32
I (650) wifi:Init management frame dynamic rx buffer num: 32
I (660) wifi:Init management short buffer num: 32
I (660) wifi:Init static rx buffer size: 1600
I (670) wifi:Init static rx buffer num: 10
I (670) wifi:Init dynamic rx buffer num: 32
I (770) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (770) wifi:mode : sta (bc:dd:c2:d1:c0:88)
I (770) wifi station: wifi_init_sta finished***ERROR*** A stack overflow in task sys_evt has been detected.
abort() was called at PC 0x4008fbb4 on core 0

ELF file SHA256: 43cbe0faf4facc1452c5f2ac9a26d6d0ceb5423569e8689f77bed0299aa6fc62

Backtrace: 0x4008f809:0x3ffd1120 0x4008fb9d:0x3ffd1140 0x4008fbb4:0x3ffd1160 0x400933aa:0x3ffd1180 0x40094e94:0x3ffd11a0 0x40094e4a:0xe0a718ac |&lt;-CORRUPTED

Rebooting...
ets Jun  8 2016 00:22:57
</pre>

### Steps to reproduce

Run the code in the zip 
[station.zip](https://github.com/espressif/esp-idf/files/4393527/station.zip)

Alvin1Zhang, 2020-03-30:
@rebeliousconformist Thanks for reporting, we will look into.

david-cermak, 2020-03-31:
Hi @rebeliousconformist 

Please note that the handlers registered for WiFi and network (`WIFI_EVENT`, `IP_EVENT`) are called from system_event task, which has a limited stack size. Seems like your code trying to use most of that stack for local variables:
``` cpp
static void event_handler(void* arg, esp_event_base_t event_base,
                                int32_t event_id, void* event_data)
{
       ...
    	char stats_buffer[16384];
    	vTaskList(stats_buffer);
    	printf("%s\n", stats_buffer);
     ...
}
```

while the sdkconfig in your project says `CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=16384` .
In that setup the stack overflow in task sys_evt is expected.

You can use either static or dynamic allocation to address the issue, possibly prefixing the buffer definition with `static ` should fix the issue.

Alvin1Zhang, 2020-05-21:
@rebeliousconformist Thanks for reporting. Would you please help share if any updates or details for this issue? Thanks.

rebeliousconformist, 2020-06-17:
@david-cermak understood makes sense. would like to close it

closed 2020-06-17 by rebeliousconformist