espressif/esp-idf #5010: Program Crash after vTaskDelete

rebeliousconformist, 2020-03-27:
Development Kit: [ESP32-Wrover-Kit|

    Kit version (for WroverKit/): [v4]

    Module or chip used: [|ESP32-WROVER-B|

    IDF version :: v4.2-dev-701-g0ae960f2f

    Build System: [|CMake & idf.py]

    Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0

    Operating System: [|Linux|]

    Using an IDE?: [|Yes eclipse & idf.py from command line

    Power Supply: [USB|

## Problem Description

Im trying to achieve the following:
1. connect to a wifi network
2. After the MQTT loop starts then, im starting another parallel task which shall publish MQTT topics.
3. IF the Wifi disconnects, then MQTT client is stopped and the task created should be deleted. 
4. Upon reconnection the  MQTT client shall be restarted and along with the task.

### Expected Behavior
1. Irrespective of disconnection the task shall be delted and recreated without errors.
2. The MQTT event loop events shall be triggered only when the specific event occurs, but they are triggered every instance. 

### Actual Behavior
1. Even though the task handle in not NULL. 
```
vTaskDelete(taskhandle);
```
2. Strange behaviour of the event loop. It gets triggered automatically not for a particular event but for all the events mentioned in the event loop enum.

 crashes the program

### Steps to reproduce
see attached project zip file please. 
[station_issue.zip](https://github.com/espressif/esp-idf/files/4393788/station_issue.zip)

chegewara, 2020-03-27:
Maybe you are trying to delete task that is already deleted and `taskhandle` is no longer valid, or it is not valid at all?

rebeliousconformist, 2020-03-27:
@chegewara if you look at this line
```
task !NULL *** 1073607572 
```
 and my code in the .zip file .
Im checking if the taskhandle is not null. when the task is not created then the task handle points= NULL.

See the .zip file

chegewara, 2020-03-27:
Just look at this logs:
```
WIFI_EVENT task !NULL *** 1073607572 
I (647003) wifi station: retry to connect to the AP
I (647003) wifi station: connect to the AP fail
WIFI_EVENT EVENT WIFI_EVENT_STA_DISCONNECTED : the event id is 5 
WIFI_EVENT EVENT eTaskGetState(hello_task) : the state is 1 
WIFI_EVENT task !NULL *** 1073607572 
```
It looks like you are trying to delete the same task twice:
`WIFI_EVENT task !NULL *** 1073607572 `

bitmakk, 2020-03-27:
The answer is very simple, I done it multiple times when i was more beginner:
You call vTaskDelete(loopHandle), and you assume, that it will be NULL-pointer.
Unfortunately it will stay on the address where the task was, and this is the cause it crashes.
Well, the task will be deleted, but that does nothing to the incoming parameter (the pointer).
After you delete the task, you have to set the loopHandle variable to NULL manually.

rebeliousconformist, 2020-03-28:
@bitmakk thanks for the tip. Yes the code works now. 
Questoin : is there a cleaner way to do it. Monitoring to check if the task is deleted is not working either because, in getting a stack overflow in sys event. Increasing the stack size did not solve the problem

chegewara, 2020-03-28:
> @bitmakk thanks for the tip. Yes the code works now.
> Questoin : is there a cleaner way to do it. Monitoring to check if the task is deleted is not working either because, in getting a stack overflow in sys event. Increasing the stack size did not solve the problem

It is clean, what you have to do is:
```
if(taskhandle != NULL){
vTaskDelete(taskhandle);
taskhandle = NULL;
}
```

rebeliousconformist, 2020-03-29:
@chegewara Ok. Cool.

closed 2020-03-29 by rebeliousconformist