espressif/esp-idf #5011: MQTT loop[ strange event triggering

rebeliousconformist, 2020-03-27:
Development Kit: [ESP32-Wrover-Kit|

    Kit version (for WroverKit/): [v4]

    Module or chip used: [|ESP32-WROVER-B|

    IDF version :: v4.2-dev-701-g0ae960f2f

    Build System: [|CMake & idf.py]

    Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0

    Operating System: [|Linux|]

    Using an IDE?: [|Yes eclipse & idf.py from command line

    Power Supply: [USB|

## Problem Description

Im trying to achieve the following:
1. connect to a wifi network
2. After the MQTT loop starts then, im starting another parallel task which shall publish MQTT topics.
3. IF the Wifi disconnects, then MQTT client is stopped and the task created should be deleted. 
4. Upon reconnection the  MQTT client shall be restarted and along with the task.

### Expected Behavior
1. Irrespective of disconnection the task shall be delted and recreated without errors.
2. The MQTT event loop events shall be triggered only when the specific event occurs, but they are triggered every instance. 

### Actual Behavior
1. Even though the task handle in not NULL. 
```
vTaskDelete(taskhandle);
```
2. Strange behaviour of the event loop. It gets triggered automatically not for a particular event but for all the events mentioned in the event loop enum.

 crashes the program

### Steps to reproduce
see attached project zip file please. 
[station_issue.zip](https://github.com/espressif/esp-idf/files/4393803/station_issue.zip)

Alvin1Zhang, 2020-03-30:
@rebeliousconformist Thanks for reporting, we will look into.

david-cermak, 2020-03-31:
Hi @rebeliousconformist 

seen your code and it looks like you're calling:
```
    esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt_event_handler, client);
```
to register the event handler for `ESP_EVENT_ANY_ID` (==all events from `MQTT_EVENTS` base).
You might want to register a separate handler just for the specific event id using that event number, for example:
``` cpp
    esp_mqtt_client_register_event(client, MQTT_EVENT_CONNECTED, mqtt_connect_handler, client);
```
For registering a handler only for connection events.

Alvin1Zhang, 2020-05-21:
@rebeliousconformist Thanks for reporting. Would you please help share if any updates or details for this issue? Thanks.

rebeliousconformist, 2020-06-17:
@Alvin1Zhang would like to close it as after the new update the events make sense.

closed 2020-06-17 by rebeliousconformist

Alvin1Zhang, 2020-06-18:
@rebeliousconformist Thanks for sharing the updates.