espressif/esp-idf #5012: Soft AP mode  with  transfer data

varunsavita, 2020-03-27:
I am able to connect in soft AP mode but not able to transfer data.
SDK latest version

Alvin1Zhang, 2020-03-30:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks. @varunsavita

varunsavita, 2020-03-30:
@Alvin1Zhang  Thanks for the response.
(https://docs.espressif.com/projects/esp-idf/en/latest
IDF branch master
- Development Kit:     ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
- Operating System:     Windows
- (Windows only) environment type: MSYS2 mingw32
- Power Supply:         USB

-----------Code -------------
 

#include <sys/param.h>
#include "esp_netif.h"
#include "esp_eth.h"

#include <esp_http_server.h>

#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sys.h"

/* The examples use WiFi configuration that you can set via project configuration menu.

   If you'd rather not, just change the below entries to strings with
   the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
*/
#define EXAMPLE_ESP_WIFI_SSID      CONFIG_ESP_WIFI_SSID
#define EXAMPLE_ESP_WIFI_PASS      CONFIG_ESP_WIFI_PASSWORD
#define EXAMPLE_MAX_STA_CONN       CONFIG_ESP_MAX_STA_CONN

/* A simple example that demonstrates how to create GET and POST
 * handlers for the web server.
 */

static const char *TAG = "wifi_AP_WEBserver";

//------------------------------------------------------------------------------
/* An HTTP GET handler */
static esp_err_t hello_get_handler(httpd_req_t *req)
{
    char*  buf;
    size_t buf_len;

    /* Get header value string length and allocate memory for length + 1,
     * extra byte for null termination */
    buf_len = httpd_req_get_hdr_value_len(req, "Host") + 1;
    if (buf_len > 1)
    {
        buf = malloc(buf_len);
        /* Copy null terminated value string into buffer */
        if (httpd_req_get_hdr_value_str(req, "Host", buf, buf_len) == ESP_OK)
        {
            ESP_LOGI(TAG, "Found header => Host: %s", buf);
        }
        free(buf);
    }

    buf_len = httpd_req_get_hdr_value_len(req, "Test-Header-2") + 1;
    if (buf_len > 1)
[... 265 lines cut ...]
     */
    //ESP_ERROR_CHECK(example_connect());
    ESP_LOGI(TAG, "init softAP");
    ESP_ERROR_CHECK(wifi_init_softap());

    /* Register event handlers to stop the server when Wi-Fi or Ethernet is disconnected,
     * and re-start it upon connection.
     */
#ifdef CONFIG_EXAMPLE_CONNECT_WIFI
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &connect_handler, &server));
    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_STA_DISCONNECTED, &disconnect_handler, &server));
#endif // CONFIG_EXAMPLE_CONNECT_WIFI
#ifdef CONFIG_EXAMPLE_CONNECT_ETHERNET
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &connect_handler, &server));
    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ETHERNET_EVENT_DISCONNECTED, &disconnect_handler, &server));
#endif // CONFIG_EXAMPLE_CONNECT_ETHERNET

    /* Start the server for the first time */
    server = start_webserver();
}
[image]

[image]

Please guide me where is wrong ..

Alvin1Zhang, 2020-03-31:
@varunsavita Thanks for providing the details, we will look into. Thanks.

varunsavita, 2020-04-10:
@Alvin1Zhang 
Please guide me on how to solve it.

Alvin1Zhang, 2020-04-10:
@varunsavita Thanks for the follow up and sorry for slow turnaround. @mahavirj Please help, thanks.

mahavirj, 2020-04-10:
@varunsavita 

Per title of issue it seems you could connect to SoftAP interface but could not access `uri` handlers registered with HTTP server.

 - Can you please confirm that you can access `http://<soft-ap-ip>/hello` or `http://<soft-ap-ip>/echo` when you flash default web server example from browser or equivalent HTTP client?
 - If answer to above is yes, maybe explain your problem in more elaborate way, e.g. whether you are trying to register some custom handler, what do you expect from that handler etc. (or maybe post specific code snippet where you think you are facing problem), this will allow for much faster help.

Thanks.

varunsavita, 2020-04-10:
Hi @mahavirj 
First, I create SoftAP mode with HTTP server(simple code example).
When code uploaded then I have a mobile app,
First< I connected WiFi in SoftAP mode, then send **get request - 192.168.4.1/23** 
I replaced hello by 23
I want to send header response "Access-Control-Allow-Origin:"

mahavirj, 2020-04-12:
@varunsavita Did you try using `httpd_resp_set_hdr(req, "Access-Control-Allow-Origin",  "<value_string>")`?

varunsavita, 2020-04-14:
thanks @mahavirj 
I will try 
I got in device 
 I (191756) wifi_AP_WEBserver: Request headers lost
I (193296) wifi_AP_WEBserver: Request headers lost

varunsavita, 2020-04-17:
@mahavirj 
Thank you for supporting,
Actually, I want to read unregister uri in code but not able.
able to do register uri and taken action accordingly.
I want to do unregister uri to register in code 
Ex- 192.168.4.1\SSID:anything

varunsavita, 2020-04-20:
@mahavirj @Alvin1Zhang 
Please reply on this topic

Alvin1Zhang, 2020-04-20:
@varunsavita Thanks for the follow up, but according to your updates, we are still not clear what you would like to achieve and the details are not enough for us to move forward. Would you please help share more details suggested by @mahavirj ? Thanks.

varunsavita, 2020-04-20:
Thank you for replying,
I have a project - 
**The device(esp32) reads real-time from the sensor and sends it to MQTT**.
Require Part -- WiFi credential change frequently as per client 
I have a mobile app to change the credential 
Ready - Mobile APP
Mobile APP send to 4 API
1:- static IP(AP mode)/23  -- start the config
2:- static IP(AP mode)/SSID: variable string  ---  set SSID 
3:- static IP(AP mode)/password: variable string --- set a password
4:- static IP(AP mode)/0 -- Done

static IP(AP mode)/23  and static IP(AP mode)/0 are fixed API

I used to code for API fixed  

**httpd_register_uri_handler(server, &API);

static const httpd_uri_t API= {
    .uri       = "/23",
    .method    = HTTP_GET,
    .handler   = hello_get_handler,
    .user_ctx  = "Got response"
};**

/*-----  not fixed API ------*/
2:- static IP(AP mode)/SSID: variable string  ---  set SSID 
3:- static IP(AP mode)/password: variable string --- set a password

Ex- /SSID: variable string
/SSID:  -- fixed part
variable string  -- vary part i have to store it.

Error
a:- (186995) httpd_uri: httpd_uri: URI '/PASSWORD:vakd' not found  
b:- httpd_resp_send_err: 404 Not Found - This URI does not exist

**Error-  a** from from   function **esp_err_t httpd_uri(struct httpd_data *hd)**  and file name httpd_uri.c
Error - b from from function **esp_err_t httpd_resp_send_err(httpd_req_t *req, httpd_err_code_t error, const char *usr_msg)** file name httpd_txrx.c

**I want to do unregister uri to register in the main code**
Require -SSID and PASSWORD

Please guide me on how to implement it let me know you require more info.
Please give me another option for a fast conversion ex - skype, WeChat, Whatsapp.

shubhamkulkarni97, 2020-04-20:
@varunsavita 
>I want to do unregister uri to register in the main code

Why do you need to unregister an uri? Instead you can register multiple URIs.

For sending SSID and password, you can make a simple POST request by registering uri as demonstrated here: https://github.com/espressif/esp-idf/blob/d85d3d969ff4b42e2616fd40973d637ff337fae6/examples/protocols/http_server/simple/main/main.c#L145

Open terminal and run `curl -d "SSID=<your_ssid>" -X POST http://<your_ip>:80/<post_uri>` to send post request.

 
> Please guide me on how to implement it let me know you require more info.
> Please give me another option for a fast conversion ex - skype, WeChat, Whatsapp.

For faster communication you may post the question on ESP32 forum (https://www.esp32.com) instead of GitHub.

varunsavita, 2020-04-20:
@shubhamkulkarni97 Thank you so much 
> For sending SSID and password, you can make a simple POST request by registering uri as demonstrated here:
Mobile APP is read, and my bad situation, I will have to receive from all get methods.
No choice to change architecture.

shubhamkulkarni97, 2020-04-21:
@varunsavita 

Can you try `http://<soft_ap_ip>/hello?<ssid>=<your_ssid>` and check if it helps?

varunsavita, 2020-04-21:
@shubhamkulkarni97 

> http://<soft_ap_ip>/hello?<ssid>=<your_ssid>
where  can I use it in code (device -esp32)

You mean
**static const httpd_uri_t API= {
.uri = "/23?SSID",
.method = HTTP_GET,
.handler = hello_get_handler,
.user_ctx = "Got response"
};** 

> http://<soft_ap_ip>/hello?<ssid>=<**your_ssid**>
Your_ssid came from mobile app.

shubhamkulkarni97, 2020-04-21:
Can you try following steps:

* Run simple http_server example in ESP-IDF
* Make a GET request from browser/app using `http://<soft_ap_ip>/hello?<ssid>=<your_ssid>` format.

I tried at my side and pasting output here:
```
I (32178) example: Found header => Host: 192.168.43.59
I (32178) example: Found URL query => ssid=Test
```

varunsavita, 2020-04-21:
[image]

 **GET request from browser/app using http://<soft_ap_ip>/hello?<ssid>=<your_ssid> format.**
@shubhamkulkarni97 Thank you for the useful info.
I added Screenshot.
My bad situation, Mobile App is ready, 
Get fixed -- static IP/ssid:my_ssid
I will have to work on only Get fixed -- static IP/ssid:my_ssid

shubhamkulkarni97, 2020-04-22:
> Can we change something like in" Found URL query"
> **? replace by :**
> Then the problem will solve it

@varunsavita This cannot be changed as format is defined by protocol (HTTP)

varunsavita, 2020-04-22:
@shubhamkulkarni97 
Thank you for the answer,
Please guide me on another way to solve it in the device code.

varunsavita, 2020-04-23:
Thank you @shubhamkulkarni97  @mahavirj 
Could I read the message (GET request from browser **http://<soft_ap_ip>/SSID:varun.**) in main.c 
Then I filter it from message.
It is not possible then , please guide me on how to solve this problem.

varunsavita, 2020-04-27:
@Alvin1Zhang @shubhamkulkarni97 @mahavirj 
Please guys, Is it possible to solve it or not?

shubhamkulkarni97, 2020-05-18:
@varunsavita This change specific to this application cannot be done. Thanks.

closed 2020-05-26 by Alvin1Zhang