espressif/esp-idf #5013: Can't use esp_wifi_set_country without getting error message

brian90013, 2020-03-27:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROVER-B
- IDF version: v4.0-265-gfb14ab08c
- Build System:         make
- Compiler version: (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     linux
- Using an IDE?: no
- Power Supply:         USB

## Problem Description

I would like to use the esp_wifi_set_country() call to set Wi-Fi AP limits based on the country of usage. However, a recent IDF change (fb14ab0 I believe) now has new logging messages coming out marked as error (but not colored red like most error messages) for a number of operations including set_country(). (It also looks like the INFO message from set_country() has an extraneous newline at the end.)

I have a test file below showing that in AP mode I can only avoid an error message by calling set_country() after esp_wifi_start(). That seems unfortunate if I can't configure allowed channels and power levels until after the transceiver starts? That could immediately put the device in violation of regulations. However, attempting to call set_country() at any earlier point prints:

```
E (666) wifi:External AP's channel is out of channel range, disconnect
```

A second, smaller but related issue appears if I try to use a device in Japan on channel 14. The only way to avoid an error message is to wait to call set_country() after esp_wifi_start(). However, if the channel configured in the esp_wifi_set_config() call isn't allowed by the default country configuration, then the esp_wifi_set_config() call will return ERROR for "channel number is out of range". If I call set_country() before set_config(), then I get the error message in the set_country() call.

Issue #4223 has a similar issue with set_country() and esp_wifi_set_max_tx_power where the power level can't be set before starting Wi-Fi, at which point the device might be transmitting above the limits. Allowing set_country() to be called before esp_wifi_start() could solve both problems.

### Code to reproduce this issue

```cpp
#include "esp_log.h"
#include "esp_wifi.h"
#include "nvs_flash.h"
#include <string.h>

static const char * TAG = "main";

void app_main(void)
{
    nvs_flash_init();
    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    wifi_country_t country = {
        .cc = "CN",
        .schan = 1,
        .nchan = 13, 
        .max_tx_power = 20, 
        .policy = WIFI_COUNTRY_POLICY_AUTO,
    };
    ESP_LOGI(TAG, "Calling set_country 1");
    ESP_ERROR_CHECK(esp_wifi_set_country(&country));
    ESP_LOGI(TAG, "Done with set_country 1");

    wifi_config_t wifi_config = {0};
    strlcpy((char *)wifi_config.ap.ssid, "ap", sizeof(wifi_config.ap.ssid));
    wifi_config.ap.ssid_len = strlen((char *)wifi_config.ap.ssid);
    wifi_config.ap.channel = 13;
    wifi_config.ap.authmode = WIFI_AUTH_OPEN;
    wifi_config.ap.max_connection = 4;

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));

    ESP_LOGI(TAG, "Calling set_country 2");
    ESP_ERROR_CHECK(esp_wifi_set_country(&country));
    ESP_LOGI(TAG, "Done with set_country 2");
    
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));

    ESP_LOGI(TAG, "Calling set_country 3");
    ESP_ERROR_CHECK(esp_wifi_set_country(&country));
    ESP_LOGI(TAG, "Done with set_country 3");
    
    ESP_ERROR_CHECK(esp_wifi_start());
    
    ESP_LOGI(TAG, "Calling set_country 4");
    ESP_ERROR_CHECK(esp_wifi_set_country(&country));
    ESP_LOGI(TAG, "Done with set_country 4");
    while(1) { vTaskDelay(1000 / portTICK_PERIOD_MS); }
}
```

## Debug Logs

```
I (556) wifi:wifi driver task: 3ffc11d0, prio:23, stack:3584, core=0
I (556) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (556) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (586) wifi:wifi firmware version: 101cee8
I (586) wifi:config NVS flash: enabled
I (586) wifi:config nano formating: disabled
I (586) wifi:Init dynamic tx buffer num: 32
I (586) wifi:Init data frame dynamic rx buffer num: 32
I (596) wifi:Init management frame dynamic rx buffer num: 32
I (596) wifi:Init management short buffer num: 32
I (606) wifi:Init static rx buffer size: 1600
I (606) wifi:Init static rx buffer num: 10
I (606) wifi:Init dynamic rx buffer num: 32
I (616) main: Calling set_country 1
I (616) wifi:set country: cc=CN schan=1 nchan=13 policy=0

E (626) wifi:External AP's channel is out of channel range, disconnect
I (636) main: Done with set_country 1
I (636) main: Calling set_country 2
I (636) wifi:set country: cc=CN schan=1 nchan=13 policy=0

E (646) wifi:External AP's channel is out of channel range, disconnect
I (656) main: Done with set_country 2
I (656) main: Calling set_country 3
I (666) wifi:set country: cc=CN schan=1 nchan=13 policy=0

E (666) wifi:External AP's channel is out of channel range, disconnect
I (676) main: Done with set_country 3
I (776) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (776) wifi:mode : sta (aa:bb:cc:dd:ee:ff) + softAP (aa:bb:cc:dd:ee:ff)
I (776) wifi:Total power save buffer number: 16
I (776) wifi:Init max length of beacon: 752/752
I (786) wifi:Init max length of beacon: 752/752
I (786) main: Calling set_country 4
I (796) wifi:set country: cc=CN schan=1 nchan=13 policy=0

I (796) main: Done with set_country 4
```

zhangyanjiaoesp, 2020-03-31:
@brian90013 
1. you should call `esp_wifi_set_country()` after `esp_wifi_start()`
2. If you devide is in Japan country code on 14 channel , then you call `esp_wifi_set_country()` to set the country to 1~13,CN. This mesage will print also. Beacuse 14 is out of range 1~13.
We print this log that to warning that your channel has changed.

brian90013, 2020-03-31:
@zhangyanjiaoesp , thank you for your response. I will place the set_country() call after wifi_start().

Sorry I was confusing about part 2 using channel 14. The code below shows why I can't use set_country() and channel 14. The set_config() function returns esp_err_t 0x2 (ERROR) which causes ESP_ERROR_CHECK() to abort. This is before calling wifi_start() or set_channel(). This is why I assumed I could call set_country() before calling set_config() or wifi_start().

```cpp
#include "esp_log.h"
#include "esp_wifi.h"
#include "nvs_flash.h"
#include <string.h>

static const char * TAG = "main";

void app_main(void)
{
    nvs_flash_init();
    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));

    wifi_config_t wifi_config = {0};
    strlcpy((char *)wifi_config.ap.ssid, "ap", sizeof(wifi_config.ap.ssid));
    wifi_config.ap.ssid_len = strlen((char *)wifi_config.ap.ssid);
    wifi_config.ap.channel = 14;
    wifi_config.ap.authmode = WIFI_AUTH_OPEN;
    wifi_config.ap.max_connection = 4;
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));

    ESP_ERROR_CHECK(esp_wifi_start());
    
    ESP_LOGI(TAG, "Calling set_country");
    wifi_country_t country = {
        .cc = "JP",
        .schan = 1,
        .nchan = 14, 
        .max_tx_power = 20, 
        .policy = WIFI_COUNTRY_POLICY_AUTO,
    };
    ESP_ERROR_CHECK(esp_wifi_set_country(&country));
    ESP_LOGI(TAG, "Done with set_country");

    while(1) { vTaskDelay(1000 / portTICK_PERIOD_MS); }
}
```

Produces this looping crash log:
```
I (558) wifi:wifi driver task: 3ffc11d0, prio:23, stack:3584, core=0
I (558) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (558) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (588) wifi:wifi firmware version: 101cee8
I (588) wifi:config NVS flash: enabled
I (588) wifi:config nano formating: disabled
I (588) wifi:Init dynamic tx buffer num: 32
I (588) wifi:Init data frame dynamic rx buffer num: 32
I (598) wifi:Init management frame dynamic rx buffer num: 32
I (598) wifi:Init management short buffer num: 32
I (608) wifi:Init static rx buffer size: 1600
I (608) wifi:Init static rx buffer num: 10
I (618) wifi:Init dynamic rx buffer num: 32
E (618) wifi:channel number is out of range
ESP_ERROR_CHECK failed: esp_err_t 0x2 (ERROR) at 0x40088904
0x40088904: _esp_error_check_failed at esp-idf-v4.0/components/esp32/panic.c:726

file: "set-country/main/main.c" line 24
func: app_main
expression: esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config)

ELF file SHA256: e2012c982e4f9852e7da84ac132229c49be5583198c898a7e7e14b88a3797536

Backtrace: 0x40088361:0x3ffbb4f0 0x40088907:0x3ffbb510 0x400d2ad5:0x3ffbb530 0x400d10de:0x3ffbb680 0x4008d229:0x3ffbb6a0
0x40088361: invoke_abort at esp-idf-v4.0/components/esp32/panic.c:155
0x40088907: _esp_error_check_failed at esp-idf-v4.0/components/esp32/panic.c:727
0x400d2ad5: app_main at set-country/main/main.c:24 (discriminator 1)
0x400d10de: main_task at esp-idf-v4.0/components/esp32/cpu_start.c:553
0x4008d229: vPortTaskWrapper at esp-idf-v4.0/components/freertos/port.c:143
Rebooting...
```

zhangyanjiaoesp, 2020-04-01:
@brian90013 
After calling `esp_wifi_init()`, the country is set to [1,13,CN]. Then you call `esp_wifi_set_config()` to set AP to 14 channel, because 14 isn't in the range, so the  `esp_wifi_set_config()` return error, and print log `E (618) wifi:channel number is out of range`.

You can write the code as the following, and I have tested locally, it works well.

> 

   void app_main(void)
   {
    nvs_flash_init();
    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
    ESP_ERROR_CHECK(esp_wifi_start());
    
    ESP_LOGI(TAG, "Calling set_country");
    wifi_country_t country = {
        .cc = "JP",
        .schan = 1,
        .nchan = 14, 
        .max_tx_power = 20, 
        .policy = WIFI_COUNTRY_POLICY_AUTO,
    };
    ESP_ERROR_CHECK(esp_wifi_set_country(&country));
    ESP_LOGI(TAG, "Done with set_country");

    wifi_config_t wifi_config = {0};
    strlcpy((char *)wifi_config.ap.ssid, "ap", sizeof(wifi_config.ap.ssid));
    wifi_config.ap.ssid_len = strlen((char *)wifi_config.ap.ssid);
    wifi_config.ap.channel = 14;
    wifi_config.ap.authmode = WIFI_AUTH_OPEN;
    wifi_config.ap.max_connection = 4;
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));
    while(1) { vTaskDelay(1000 / portTICK_PERIOD_MS); }
    }

>

brian90013, 2020-04-01:
Thank you for your solution. While it is helpful that you can avoid the error conditions, I wonder if putting the commands out of order is really the proper solution?

Below are three IDF networking examples that all follow the ordering of set_config() then start(). This appears to be a pattern in the IDF and it makes sense to configure a module before it begins to run. Please consider changing set_country() so it can be called before wifi_set_config() and wifi_start(). Note the first example is softAP so your code above is flipping the order of the demo project.

[softAP](https://github.com/espressif/esp-idf/blob/release/v4.0/examples/wifi/getting_started/softAP/main/softap_example_main.c)
```cpp
ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));
ESP_ERROR_CHECK(esp_wifi_start());
```

[internal mesh](https://github.com/espressif/esp-idf/blob/release/v4.0/examples/mesh/internal_communication/main/mesh_main.c)
```cpp
ESP_ERROR_CHECK(esp_mesh_set_config(&cfg));
/* mesh start */
ESP_ERROR_CHECK(esp_mesh_start());
```

[mqtt](https://github.com/espressif/esp-idf/blob/release/v4.0/examples/protocols/mqtt/tcp/main/app_main.c)
```cpp
esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqtt_cfg);
esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt_event_handler, client);
esp_mqtt_client_start(client);
```

zhangyanjiaoesp, 2020-04-01:
@brian90013 
What I have done is not just to avoid the error condition. If you must set AP's info with the order:

> ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
> ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP,` &wifi_config));
> ESP_ERROR_CHECK(esp_wifi_start());

then you should set the AP's channel in the range [1,13], channel 14 is out of rang, of course it will return error. 
If you want to  set it to channel 14,  you must set the country to which supports [1,14] channel. 
The error is meant to remind you that you have setting wrong !

Alvin1Zhang, 2020-04-08:
@zhangyanjiaoesp Thanks for supporting. @brian90013 Thanks for sharing the updates. Feel free to reopen if you still have the issue. Thanks.

closed 2020-04-08 by Alvin1Zhang