espressif/esp-idf #5015: BLufi Example on ESP-IDF3?

DasVoelkel, 2020-03-28:
Just wanted to ask if the example is still on V3, i assume as much as it wasn't updated since the V4 drop.
Or is V4 backwards compatible with this code?

Alvin1Zhang, 2020-03-30:
@DasVoelkel Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

DasVoelkel, 2020-03-30:
Nvm, I think it's just outdated, but it's easy to get to the new version or search for analog functionality

closed 2020-03-30 by DasVoelkel