espressif/esp-idf #5016: Convert_to_comake:  RuntimeError: PROJECT_NAME does not appear

marrog86, 2020-03-28:
I downloaded the Project [ka-Radio](https://github.com/karawin/Ka-Radio32)  and tried to convert it. 
Without any changes i called the converting tool:
 ./convert_to_cmake.py ~........./Ka-Radio32-master/

error returned: 
```

Converting /home/....../Ka-Radio32-master/...
Traceback (most recent call last):
  File "./convert_to_cmake.py", line 205, in <module>
    main()
  File "./convert_to_cmake.py", line 201, in main
    convert_project(args.project)
  File "./convert_to_cmake.py", line 115, in convert_project
    raise RuntimeError("PROJECT_NAME does not appear to be defined in IDF project Makefile at %s" % project_path)
RuntimeError: PROJECT_NAME does not appear to be defined in IDF project Makefile at /........../ESP/Ka-Radio32-master/

```

The Makefile: 
```
#
# This is a project Makefile. It is assumed the directory this Makefile resides in is a
# project subdirectory.
#

PROJECT_NAME := KaRadio32

COMPONENT_ADD_INCLUDEDIRS := components/include

include $(IDF_PATH)/make/project.mk

# Copy some defaults into the sdkconfig by default
# so BT stack is enabled
sdkconfig: sdkconfig.defaults
	$(Q) cp $< $@

menuconfig: sdkconfig
defconfig: sdkconfig
```

Alvin1Zhang, 2020-03-30:
@marrog86 Thanks for reporting, we will look into.

dobairoland, 2020-03-30:
@marrog86 I cannot reproduce the issue. It works for me in every setup I've tried. Please provide more information about your system: version of your Python, version of IDF, OS and version, any information you can give about the Makefile (encoding, line endings).

dobairoland, 2020-04-08:
@marrog86 Do you have any update regarding this issue?

dobairoland, 2020-04-17:
I'm closing this for inactivity. Feel free to reopen it if the issue still exits.

closed 2020-04-17 by dobairoland