espressif/esp-idf #5018: BLE related panic (rwble.c) still exists

fpgamaster, 2020-03-29:
Hi All,
Sorry for the duplicate issue report but this panic makes me crazy.

BT and WiFi are enabled. WiFi is connected only no data traffic. With WiFi disabled this panic gone, at least not seen.

Any ideas how to fix or a workaround this are welcome.

The crash and some sdkconfig are below:

## Environment
- Development Kit:     ESP32-PICO-Kit
- Kit version PicoKit v4
- Module or chip used: ESP32-PICO-D4
- IDF version v4.2-dev-910-g4fe04f115
- Build System:  Make
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

ASSERT_PARAM(512 0), in rwble.c at line 238

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:

PC      : 0x40088486  PS      : 0x00060034  A0      : 0x8008c67c  A1      : 0x3ffbf6e0  

0x40088486: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffbdbe4  A7      : 0x60008050  A8      : 0x80088481  A9      : 0x3ffbf6c0  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  

EXCVADDR: 0x00000000  LBEG    : 0x400883b9  LEND    : 0x400883c0  LCOUNT  : 0x00000000  

0x400883b9: r_assert_param at ??:?
0x400883c0: r_assert_param at ??:?

Core  0 was running in ISR context:

EPC1    : 0x401d4237  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40088486

0x401d4237: uart_ll_write_txfifo at /home/xxxx/esp/esp-idf/components/soc/src/esp32/include/hal/uart_ll.h:188
 (inlined by) uart_hal_write_txfifo at /home/xxxx/esp/esp-idf/components/soc/src/hal/uart_hal_iram.c:40

0x40088486: r_assert_param at ??:?

Backtrace:0x40088483:0x3ffbf6e0 0x4008c679:0x3ffbf700 0x4008d2a7:0x3ffbf720 0x4008604d:0x3ffbf740 0x4000bfed:0x3ffde4a0 0x4009b8fd:0x3ffde4b0 0x40099d60:0x3ffde4d0 0x400d2ef1:0x3ffde510 0x4008350d:0x3ffde530 0x40083870:0x3ffde570 0x400f80ee:0x3ffde5b0 0x400d2d8d:0x3ffde5d0 0x400f032d:0x3ffde5f0 0x400f41c9:0x3ffde630 0x400f423d:0x3ffde650 0x400f3785:0x3ffde670 0x4009c445:0x3ffde6a0

0x40088483: r_assert_param at ??:?
0x4008c679: r_rwble_isr at ??:?
0x4008d2a7: r_rwbtdm_isr_wrapper at intc.c:?
0x4008604d: _xt_lowint1 at /home/xxxx/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105
0x4009b8fd: xTaskGetSchedulerState at /home/xxxx/esp/esp-idf/components/freertos/tasks.c:4016
0x40099d60: xQueueGenericReceive at /home/xxxx/esp/esp-idf/components/freertos/queue.c:1451 (discriminator 4)
0x400d2ef1: queue_recv_wrapper at /home/xxxx/esp/esp-idf/components/esp_wifi/esp32/esp_adapter.c:349
0x4008350d: coex_bb_reset_unlock at ??:?
0x40083870: coex_core_request at ??:?
0x400f80ee: esp_coex_adapter_register at ??:?
0x400d2d8d: coex_wifi_release_wrapper at /home/xxxx/esp/esp-idf/components/esp_wifi/esp32/esp_adapter.c:485
0x400f032d: pm_tbtt_process at ??:?
0x400f41c9: dbg_lmac_init at ??:?
0x400f423d: pp_timer_do_process at ??:?
0x400f3785: ppTask at ??:?
0x4009c445: vPortTaskWrapper at /home/xxxx/esp/esp-idf/components/freertos/xtensa/port.c:143

[... 308 lines cut ...]
# CONFIG_BTM_TRACE_LEVEL_VERBOSE is not set
CONFIG_BTM_INITIAL_TRACE_LEVEL=2
CONFIG_BTH_LOG_SDP_INITIAL_TRACE_LEVEL=2
# CONFIG_BTIF_TRACE_LEVEL_NONE is not set
# CONFIG_BTIF_TRACE_LEVEL_ERROR is not set
CONFIG_BTIF_TRACE_LEVEL_WARNING=y
# CONFIG_BTIF_TRACE_LEVEL_API is not set
# CONFIG_BTIF_TRACE_LEVEL_EVENT is not set
# CONFIG_BTIF_TRACE_LEVEL_DEBUG is not set
# CONFIG_BTIF_TRACE_LEVEL_VERBOSE is not set
CONFIG_BTIF_INITIAL_TRACE_LEVEL=2
# CONFIG_BTC_TRACE_LEVEL_NONE is not set
# CONFIG_BTC_TRACE_LEVEL_ERROR is not set
CONFIG_BTC_TRACE_LEVEL_WARNING=y
# CONFIG_BTC_TRACE_LEVEL_API is not set
# CONFIG_BTC_TRACE_LEVEL_EVENT is not set
# CONFIG_BTC_TRACE_LEVEL_DEBUG is not set
# CONFIG_BTC_TRACE_LEVEL_VERBOSE is not set
CONFIG_BTC_INITIAL_TRACE_LEVEL=2

CONFIG_BTDM_COEX_BT_OPTIONS=y
CONFIG_BTDM_COEX_BLE_ADV_HIGH_PRIORITY=y
CONFIG_ESP32_WIFI_SW_COEXIST_ENABLE=y
CONFIG_SW_COEXIST_ENABLE=y
"

Alvin1Zhang, 2020-03-30:
@fpgamaster Sorry for the issue brought to you and thanks for reporting, we will look into. Thanks.

dedvalson, 2020-06-30:
I am seeing this also - ASSERT_PARAM(1024 0), in rwble.c at line 238

chhajedji, 2020-07-01:
Hi @fpgamaster Please run your application by rebasing to latest commits. `ASSERT_PARAM(512, 0)` should be resolved in recent commits.

@dedvalson we will look into `ASSERT_PARAM(1024, 0)` and get back to you shortly. Meanwhile I would like to know the steps to reproduce this issue. Also how is assert (512, 0) is linked with assert (1024, 0).

AxelLin, 2020-07-01:
> Hi @fpgamaster Please run your application by rebasing to latest commits. `ASSERT_PARAM(512, 0)` should be resolved in recent commits.

Not sure which commit do you mean by latest commit?
The current master tree v4.2-dev-1905-g625bd5eb1806 does not have new update for a few weeks.

fpgamaster, 2020-07-01:
@chhajedji I am already on the latest commit and can confirm that this issue is still not fixed.

chhajedji, 2020-07-01:
@fpgamaster sorry for the inconvenience. `ASSERT_PARAM(512, 0)` occurs due to various reasons. I will look into the issue again. Please tell me which example are you running in which you are seeing this type of behavior.

dedvalson, 2020-07-01:
I have no specific steps to reproduce it, that is what is strange. It very occasionally shows up on customer systems that have been running for days. There is nothing we can determine that changes when it happens. If you could tell us what it indicates perhaps I could give a better idea. We are running BLE simultaneously with WiFi (station mode). However there is no BT activity at all prior to the failure.

fpgamaster, 2020-07-01:
Hi @dedvalson,
It is same here :) The system is running for days and I can't manage to find the exact situation when this happens.
May be you can add some debug messages before the assert in order to track the issue?
BR

chhajedji, 2020-07-09:
Hello @fpgamaster @dedvalson 

Although we were not able to reproduce assert 512 and 1024, we figured out that assert (1024, 0) and assert (512,0) arises due to timing mismatch in communicating with the hardware in Bluetooth controller. Requests for fixing both the issues has been raised and will be soon available.

This issue could be rare as when both WiFi and Bluetooth are running and as both are using same RF, some cases could be present where collision happens between them and crash occurs.

sicanins, 2020-07-17:
Hi, 

I ran into the same issue now with v4.2-dev-1547-g5b5a8c276.

Our app was working before we first time enabled the BLE.

I am not using Wifi at all, only BLE. The BLE code is mostly the ble-spp-server example. No significant changes.

What is special about the app is, that we are running our own 900MHz radio protocol on a level 3 irq. It makes heavy use of this IRQ and also blocks sometimes for several 100 us.
The bt is hooked up to a level1 IRQ here as I can see from the crash log.

While porting this protocol to the ESP we found that there are regularly critical sections that take even up to 700us. A promiment one is the malloc. Maybe because we are using the SPIRAM?

Maybe thats why you are not seeing this issue and I see it very reproducable.

We have the crash seen several times, not sure how long it took exactly. More than a few minutes and less than 1.5h.

Any idea how long this will take to fix? We would like to roll out our otherwise working product...

Thanks!

fpgamaster, 2020-07-18:
Hi @sicanins,

It looks like a race condition.
This issue is a showstopper for us too :(

BR

chhajedji, 2020-07-20:
Hi, commit which fixes `ASSERT_PARAM (512, 0)` shall be merged into master soon. You can track the commit by it's message which closes this issue.

sicanins, 2020-07-20:
Hi,
We are suffering the (1024, 0). Any idea when this will be fixed?

Thanks,

chhajedji, 2020-07-20:
Hi @sicanins

assert (1024, 0) is also in pipeline and a merge request for it has also been raised. But this request is not raised by me and by some other colleague. But you can be relieved as it will also be merged into Github soon.

sicanins, 2020-07-20:
Hi @chhajedji ,

great, thanks for the update, will happily wait for the fix!

chhajedji, 2020-07-30:
Hi everyone,

MR for fixing assert (1024, 0) and assert (512, 0) has been merged into master and soon will be updated on Github. You can track this fix by `43f1e432151677b004b49de5697c0318a84aefae` this commit ID. Thanks.

AxelLin, 2020-08-04:
> Hi everyone,
> 
> MR for fixing assert (1024, 0) and assert (512, 0) has been merged into master and soon will be updated on Github. You can track this fix by `43f1e432151677b004b49de5697c0318a84aefae` this commit ID. Thanks.

Err, recent master tree update does not include this fix.
It's a panic issue, I hope don't need to wait too long to get the fix.

chhajedji, 2020-08-08:
Unluckily, the master got synced with a commit just before the fixing commit. It will be shown in the next sync! Sorry for the inconvenience

AxelLin, 2020-08-10:
@chhajedji 

I'm not sure if it's just *unlucky*.
I even cannot find the fix in esp32-bt-lib git tree, how can it merge to idf master tree without merging to esp32-bt-lib first?
Or is there something wrong?

PS. This is really a gating issue for BT users because the users cannot workaround the panic issue.
I'm appreciate if someone can help to release the fix asap.
@Alvin1Zhang

chhajedji, 2020-08-10:
Hi @AxelLin, [here](https://github.com/espressif/esp32-bt-lib/commit/93eeca8b535dd609da8b1cdac44a11707e03c333) is the fix which resolves this issue. IFS_UNDERRUN is the issue causing assert(512, 0) and ADV_UNDERRUN is causing assert(1024, 0). This commit resolves both the issues.

dedvalson, 2020-08-25:
> Please check with the latest master. Fix for ASSERT(1024, 0) and
> ASSERT(512, 0) have been synced. If the issue is resolved, please close
> this issue. Thanks.

Is there a way that this can be used with V4.1 Beta 2? I tried just
updating it in that tree but got a bunch of link errors.

chhajedji, 2020-08-25:
If you can use release/v4.1 branch then this fix will be present on it also as release/v4.1, 4.2 and master use same commit from bt-lib. Backporting it to major releases is still to be done.

chhajedji, 2020-08-25:
Hi @fpgamaster Please close this issue as it is fixed.

dedvalson, 2020-08-27:
Hi,

It does not look like this fix is present in Release/4.1 or Release/4.2
yet. I cloned a fresh copy of Release/4.1 and the fix is not present.

I tried to update just this component in Release/4.1 to this commit but
unfortunately I then get the following two undefined references

...
esp-idf/v4.1a/components/bt/controller/lib\libbtdm_app.a(llc_llcp.o):(.text+0x0):
undefined reference to `llcp_channel_map_req_handler'
... esp-idf/v4.1a/components/bt/controller/lib\libbtdm_app.a(llc_llcp.o):
in function `llcp_channel_map_req_pre_handler':

Unfortunately I cannot use master branch.

Any suggestions, this is totally blocking my product release.

Don

chhajedji, 2020-08-27:
Hi @dedvalson 

The issue you have quoted had been a blocker for many further commits in branch `release/v4.1` and `release/v4.2`. It has been resolved now and and slowly further commits are getting added to these 2 releases.

Although `master` and `release/v4.1` uses same bt-lib branch, this submodule has not been updated in v4.1 branch. Request to merge has been raised and will be merged soon. But it may take a while to reflect here on Github. Sorry that I missed this part in my previous comment.

For a workaround, temporarily you can try below steps to overcome this issue on `release/v4.1`. Once the fixing commit is present on Github, you can use the default branch.

- `cd $IDF_PATH/components/bt/controller/lib`
- `git checkout master`

Note that if you do `git submodule update --init --recursive`, then this fix will be undone.

This could work as a temporary hackish way to overcome the issue.

dedvalson, 2020-08-27:
Hi,

Unfortunately when I attempt the workaround you mentioned I get two
unresolved symbols:

...
esp-idf/v4.1a/components/bt/controller/lib\libbtdm_app.a(llc_llcp.o):(.text+0x0):
undefined reference to `llcp_channel_map_req_handler'
... esp-idf/v4.1a/components/bt/controller/lib\libbtdm_app.a(llc_llcp.o):
in function `llcp_channel_map_req_pre_handler':

Thanks,

Don

chhajedji, 2020-08-29:
Then you will have to wait for the next sync of release/v4.1. Proper fix could take some time to reflect but after next sync, checking out components/bt/controller/lib submodule to master as suggested [here](https://github.com/espressif/esp-idf/issues/5018#issuecomment-681888418) should help.

dedvalson, 2020-08-31:
Is there any guess about when the next sync will take place? As far as I
know, There currently isn't a release (other than Master) that this works
in.

dedvalson, 2020-09-03:
Has there been any progress on this? I could use 4.1 or 4.2 but I don't
think I can move a production product onto the master branch. This is
really harming our product. If there is some way that I can build a version
from bits and pieces let me know. I tried just updating this item but then
I get unresolved externals (see above).

Please advise,

don

chhajedji, 2020-09-04:
Hello @dedvalson,

I am providing a patch and a library for `release/v4.1`. Please apply the patch in your `IDF_PATH` and replace the library at `$IDF_PATH/components/bt/controller/lib/` with the `libbtdm_app.a`. This library contains fix for IFS UNDERRUN and ADV UNDERRUN (assert 1024 and 512). This is a temporary fix, soon when Github will be synced, you can find the commits fixing these issues on Github.

[changes.tar.gz](https://github.com/espressif/esp-idf/files/5174867/changes.tar.gz)

enricop, 2020-09-04:
hello, we are facing this issue also in v4.0.1:

```
ASSERT_PARAM(64 0), in rwble.c at line 238
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4008481d  PS      : 0x00060034  A0      : 0x80088a14  A1      : 0x3ffbee00
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054
A6      : 0x3ffbe620  A7      : 0x60008050  A8      : 0x8008481d  A9      : 0x3ffbede0
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40084755  LEND    : 0x4008475c  LCOUNT  : 0x00000000
Core 0 was running in ISR context:
EPC1    : 0x4008c30e  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008481d
```

is it possibile (before October) to backport the fix also to v4.0 branch?

dedvalson, 2020-09-05:
Chinmay,

That appears to have worked. I wanted to tell you how much I appreciate
someone going out of their way to help like that. I really appreciate it.

Thanks,

Don

dedvalson, 2020-09-06:
Chinmay,

I think I spoke too soon. My app links fine with the new patch, but it
doesn't work at all. The call to btdm_controller_init fails with a return
value of -4. My config is just the default one (esp_bt_controller_config_t
btCfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();)

I will see if I can get the samples to fail also.

Thanks,

Don

chhajedji, 2020-09-07:
Hi @enricop 

Please open a separate issue for assert 64, 0 and we will help you with it. This issue is already tackling with assert 512, 0 and 1024, 0. It is easier to manage different crashes in different issues.

chhajedji, 2020-09-07:
Hi @dedvalson 

Sorry for the patch not working as expected. Although this is independent of any host, I would still like to know which host are you using and which example in esp-idf I should try to reach closest to your use-case?

chhajedji, 2020-09-07:
Hi @dedvalson 

Please test the given patch and let me know if it works. I tried running `nimble/bleprph` and `nimble/blecent` examples and they were running fine.

[changes.tar.gz](https://github.com/espressif/esp-idf/files/5183229/changes.tar.gz)

dedvalson, 2020-09-08:
Hi @chinmay

It works!! Thanks for your help.

Don

closed 2020-09-15 by Alvin1Zhang