espressif/esp-idf #5019: On the fifth operation on SSL_Connect error on esp32

CS-Alwadi, 2020-03-29:
Every connection is working successfully until the fifth one.

when we turn on mdbed debug, esp32 prints ( I (225782) openssl: mbedtls_ssl_handshake() return -0x4e )
why on the fifth one?

chegewara, 2020-03-29:
If its always 5th connection then check if you dont have memory leak and if you have enough free heap to perform mbedtls/ssl operations. Its about 70kB required IIRC.

CS-Alwadi, 2020-03-29:
i've check the head size after the fifth connection it's (134244) bytes

CS-Alwadi, 2020-03-29:
Before evry connection there is ~ 135000 bytes of heap

chegewara, 2020-03-29:
https://forums.mbed.com/t/tls-handshake-issue/3583

AdityaHPatwardhan, 2020-04-14:
@CS-Alwadi, can you provide more details, maybe full error log ! Thank you !

CS-Alwadi, 2020-04-16:
Actually this is the only error that showed up ( I (225782) openssl: mbedtls_ssl_handshake() return -0x4e ). on the fifth connection

AdityaHPatwardhan, 2020-04-29:
Can you please share some code or something with which I am able to reproduce the issue, I do not face this issue. For verifying please remove call `esp_tls_conn_delete` on line 123 in [https_request_example_main](https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_request/main/https_request_example_main.c), new connection are created successfully even after 5th connection ( until the heap runs out)

Alvin1Zhang, 2020-05-07:
@CS-Alwadi Thanks for reporting, would you please help share if any updates for the issue? Thanks.

CS-Alwadi, 2020-05-17:
we found that there was bug in our program, we had an indicator that counts up successfully created sockets, but when shutdown socket we didn't decrease it. we can close this issue. 
thank you all

closed 2020-05-18 by mahavirj